home-assistant/operating-system #3313: WiFi has disappeared after upgrading OS from 12.1 to 12.2

bearhntr, 2024-04-16:
### Describe the issue you are experiencing

I was prompted there was an OS Upgrade in HA 11-APR-2024 from 12.1 to 12.2. I have had WiFi working for many months - after the upgrade I noticed that under Settings > System > Network it was no longer showing both interfaces. I rebooted the hardware and still not there.

I am running HAOS installation on an HP T620 ThinClient (as I have for nearly 2 years).

This is after the 12.2 upgrade.
[DMESG_Output-12APR2024.txt](https://github.com/home-assistant/operating-system/files/14999122/DMESG_Output-12APR2024.txt)

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

6.6.25-haos

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.  I was prompted to update the OS (11-Apr-2024)
2.  I performed the update/upgrade
3.  Was not asked to make backup
4.   I performed the update, system rebooted (hardware reboot)
5.   WiFi gone.

I have just performed the `ha os update --version 12.1`   and the WiFi is back.

[image]
...

### Anything in the Supervisor logs that might be useful for us?

```txt
2024-04-16 11:26:25.782 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/amd64-addon-configurator with version 5.8.0
2024-04-16 11:26:25.793 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/log-viewer/amd64 with version 0.17.0
2024-04-16 11:26:25.802 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/amd64-addon-mariadb with version 2.6.1
2024-04-16 11:26:25.810 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/amd64-addon-samba with version 12.3.1
2024-04-16 11:26:25.821 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/zwave-js-ui/amd64 with version 3.4.2
2024-04-16 11:26:25.832 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/vscode/amd64 with version 5.15.0
2024-04-16 11:26:25.869 INFO (MainThread) [supervisor.backups.manager] Found 14 backup files
2024-04-16 11:26:25.914 INFO (MainThread) [supervisor.discovery] Loaded 1 messages
2024-04-16 11:26:25.914 INFO (MainThread) [supervisor.ingress] Loaded 2 ingress sessions
2024-04-16 11:26:25.915 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state setup
2024-04-16 11:26:25.915 INFO (MainThread) [supervisor.resolution.check] System checks complete
2024-04-16 11:26:25.915 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state setup
2024-04-16 11:26:25.916 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
2024-04-16 11:26:25.917 INFO (MainThread) [supervisor.jobs] 'ResolutionFixup.run_autofix' blocked from execution, system is not running - setup
2024-04-16 11:26:25.917 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state setup
2024-04-16 11:26:25.918 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
2024-04-16 11:26:25.919 INFO (MainThread) [__main__] Running Supervisor
2024-04-16 11:26:25.935 INFO (MainThread) [supervisor.os.manager] Rauc: A - marked slot kernel.0 as good
2024-04-16 11:26:25.961 INFO (MainThread) [supervisor.addons.manager] Phase 'initialize' starting 0 add-ons
2024-04-16 11:26:25.976 INFO (MainThread) [supervisor.addons.manager] Phase 'system' starting 2 add-ons
2024-04-16 11:26:26.009 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mariadb_pass
2024-04-16 11:26:26.009 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mariadb_user
2024-04-16 11:26:26.010 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mariadb_user
2024-04-16 11:26:26.018 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mariadb_pass
2024-04-16 11:26:26.018 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mariadb_user
2024-04-16 11:26:26.019 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mariadb_user
2024-04-16 11:26:26.592 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on homeassistant/amd64-addon-mariadb with version 2.6.1
2024-04-16 11:26:27.268 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on ghcr.io/hassio-addons/zwave-js-ui/amd64 with version 3.4.2
2024-04-16 11:26:28.274 WARNING (MainThread) [supervisor.services.modules.mysql] Can't remove not exists services
2024-04-16 11:26:28.889 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mariadb_pass
2024-04-16 11:26:28.890 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mariadb_user
2024-04-16 11:26:28.890 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mariadb_user
2024-04-16 11:26:30.583 INFO (MainThread) [supervisor.services.modules.mysql] Set core_mariadb as service provider for MySQL
[... 232 lines cut ...]
version_api | ok
installed_addons | Advanced SSH & Web Terminal (17.2.0), File editor (5.8.0), Log Viewer (0.17.0), MariaDB (2.6.1), Samba share (12.3.1), Studio Code Server (5.15.0), Z-Wave JS UI (3.4.2)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 3
views | 6
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | April 11, 2024 at 14:10
-- | --
current_recorder_run | April 15, 2024 at 18:47
estimated_db_size | 92.22 MiB
database_engine | mysql
database_version | 10.6.12

</details>

bearhntr, 2024-04-16:
I pulled another DMESG after dropping the OS back to 12.1 - attached here:  

[DMESG_Output-16APR2024-dropped to OS 12.1.txt](https://github.com/home-assistant/operating-system/files/14999476/DMESG_Output-16APR2024-dropped.to.OS.12.1.txt)

bearhntr, 2024-04-16:
dmesg | grep wl
[    0.000000] DMI: Hewlett-Packard HP t620 Quad Core TC/21B4, BIOS L40 v02.19 02/19/2019
[    2.898961] iwlwifi 0000:01:00.0: enabling device (0100 -> 0102)
[    2.917889] iwlwifi 0000:01:00.0: Detected crf-id 0x400410, cnv-id 0x400410 wfpm id 0x80000000
[    2.927619] iwlwifi 0000:01:00.0: PCI dev 2725/0024, rev=0x420, rfid=0x10d000
[    2.955435] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-83.ucode failed with error -2
[    2.958049] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-82.ucode failed with error -2
[    2.960570] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-81.ucode failed with error -2
[    2.963135] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-80.ucode failed with error -2
[    2.965748] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-79.ucode failed with error -2
[    2.968411] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-78.ucode failed with error -2
[    2.971197] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-77.ucode failed with error -2
[    2.974008] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-76.ucode failed with error -2
[    2.976781] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-75.ucode failed with error -2
[    2.979765] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-74.ucode failed with error -2
[    2.982686] iwlwifi 0000:01:00.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-73.ucode failed with error -2
[    3.005451] iwlwifi 0000:01:00.0: api flags index 2 larger than supported by driver
[    3.009646] iwlwifi 0000:01:00.0: TLV_FW_FSEQ_VERSION: FSEQ Version: 0.0.2.36
[    3.012689] iwlwifi 0000:01:00.0: loaded firmware version 72.a764baac.0 ty-a0-gf-a0-72.ucode op_mode iwlmvm
[    3.222975] iwlwifi 0000:01:00.0: Detected Intel(R) Wi-Fi 6 AX210 160MHz, REV=0x420
[    3.455490] iwlwifi 0000:01:00.0: WFPM_UMAC_PD_NOTIFICATION: 0x1f
[    3.457157] iwlwifi 0000:01:00.0: WFPM_LMAC2_PD_NOTIFICATION: 0x1f
[    3.457172] iwlwifi 0000:01:00.0: WFPM_AUTH_KEY_0: 0x80
[    3.457185] iwlwifi 0000:01:00.0: CNVI_SCU_SEQ_DATA_DW9: 0x0
[    3.459796] iwlwifi 0000:01:00.0: loaded PNVM version e28bb9d7
[    3.476901] iwlwifi 0000:01:00.0: Detected RF GF, rfid=0x10d000
[    3.545477] iwlwifi 0000:01:00.0: base HW address: 60:e3:2b:fb:b1:b6
[    3.587551] iwlwifi 0000:01:00.0 wlp1s0: renamed from wlan0
[    7.621116] iwlwifi 0000:01:00.0: WFPM_UMAC_PD_NOTIFICATION: 0x1f
[    7.621157] iwlwifi 0000:01:00.0: WFPM_LMAC2_PD_NOTIFICATION: 0x1f
[    7.621175] iwlwifi 0000:01:00.0: WFPM_AUTH_KEY_0: 0x80
[    7.621194] iwlwifi 0000:01:00.0: CNVI_SCU_SEQ_DATA_DW9: 0x0
[    7.709127] iwlwifi 0000:01:00.0: Registered PHC clock: iwlwifi-PTP, with index: 0
[   10.881043] wlp1s0: authenticate with 78:d2:94:36:3d:f0
[   10.884240] wlp1s0: send auth to 78:d2:94:36:3d:f0 (try 1/3)
[   10.915902] wlp1s0: authenticated
[   10.916626] wlp1s0: associate with 78:d2:94:36:3d:f0 (try 1/3)
[   10.967679] wlp1s0: RX AssocResp from 78:d2:94:36:3d:f0 (capab=0x1511 status=0 aid=7)
[   10.970910] wlp1s0: associated
[   20.021077] wlp1s0: deauthenticating from 78:d2:94:36:3d:f0 by local choice (Reason: 3=DEAUTH_LEAVING)
[   27.055381] wlp1s0: authenticate with 78:d2:94:36:3d:f0
[   27.060653] wlp1s0: send auth to 78:d2:94:36:3d:f0 (try 1/3)
[   27.093019] wlp1s0: authenticated
[   27.094927] wlp1s0: associate with 78:d2:94:36:3d:f0 (try 1/3)
[   27.142556] wlp1s0: RX AssocResp from 78:d2:94:36:3d:f0 (capab=0x1511 status=0 aid=7)
[   27.146232] wlp1s0: associated

sairon, 2024-04-17:
This seems to be either a driver kernel regression or AX210 doesn't like the new firmware. I think I have this card somewhere, I will check if I can reproduce.

dndlion, 2024-04-18:
Having a related issue, in that bluetooth on my AX201 card also fails here's relevant boot log output:

Apr 18 08:15:32 homeassistant kernel: Bluetooth: Core ver 2.22
Apr 18 08:15:32 homeassistant kernel: NET: Registered PF_BLUETOOTH protocol family
Apr 18 08:15:32 homeassistant kernel: Bluetooth: HCI device and connection manager initialized
Apr 18 08:15:32 homeassistant kernel: Bluetooth: HCI socket layer initialized
Apr 18 08:15:32 homeassistant kernel: Bluetooth: L2CAP socket layer initialized
Apr 18 08:15:32 homeassistant kernel: Bluetooth: SCO socket layer initialized
Apr 18 08:15:32 homeassistant kernel: usbcore: registered new interface driver btusb
Apr 18 08:15:32 homeassistant kernel: Bluetooth: hci0: Firmware timestamp 2023.42 buildtype 1 build 73111
Apr 18 08:15:32 homeassistant kernel: Bluetooth: hci0: Found device firmware: intel/ibt-0040-1050.sfi
Apr 18 08:15:32 homeassistant kernel: Bluetooth: hci0: Boot Address: 0x100800
Apr 18 08:15:32 homeassistant kernel: Bluetooth: hci0: Firmware Version: 60-48.23
Apr 18 08:15:32 homeassistant kernel: usb 2-1: USB disconnect, device number 2
Apr 18 08:15:32 homeassistant kernel: Bluetooth: hci0: FW download error recovery failed (-19)
Apr 18 08:15:32 homeassistant kernel: Bluetooth: hci0: sending frame failed (-19)
Apr 18 08:15:32 homeassistant kernel: Bluetooth: hci0: Reading supported features failed (-19)
Apr 18 08:15:32 homeassistant kernel: Bluetooth: hci0: Error reading debug features

Luckily I had old firmware backed up in /mnt/boot/firmware/intel that I've switched back to for the time being that option was discovered in a previous issue: https://github.com/home-assistant/core/issues/98523

bearhntr, 2024-04-19:
I hope they fix this in the 12.3 or 12.4 OS update - I hate seeing that Orange dot telling me I need to do an update.

bearhntr, 2024-05-04:
Any updates on this?  When can I upgrade the OS again?  I now have multiple updates needed.  I really hate running in this mode.  The one circled is the one which broke WiFi and BT when I installed it.

[image]

42Cosmic42, 2024-05-08:
I'm using the same card and run into the same issues.

Intel(R) Wi-Fi 6 AX210

[    1.691899] iwlwifi 0000:02:00.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-83.ucode failed with error -2
[    1.691928] iwlwifi 0000:02:00.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-82.ucode failed with error -2
[    1.691956] iwlwifi 0000:02:00.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-81.ucode failed with error -2
[    1.691983] iwlwifi 0000:02:00.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-80.ucode failed with error -2
[    1.692006] iwlwifi 0000:02:00.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-79.ucode failed with error -2
[    1.692035] iwlwifi 0000:02:00.0: Direct firmware load for iwlwifi-ty-a0-gf-a0-78.ucode failed with error -2
[    1.698141] iwlwifi 0000:02:00.0: api flags index 2 larger than supported by driver

bearhntr, 2024-05-08:
I installed 12.3 today....my WiFi is GONE again.

[image]

[image]

bearhntr, 2024-05-08:
@dndlion 

RE-installed  OS 12.1 -- and it comes back

[image]

closed 2024-08-14 by github-actions[bot]