home-assistant/operating-system #3318: When I install home assistant operating system 12.2 update the oracle vm randomly abort

WillemGoosen, 2024-04-18:
### Describe the issue you are experiencing

When I install home assistant operating system 12.2 update the oracle vm randomly abort and when I revert to an earlier VM export the VM does not abort.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

12.2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. When I install OS 12.2 update
2.The Oracle VM aborts randomly normally over night.
3.Roll back to VM export prior to OS 12.2 update the problem is gone.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
This is the only warning I have at this stage.
2024-04-18 16:41:28.148 WARNING (MainThread) [supervisor.addons.options] Option 'interface' does not exist in the schema for Samba share (core_samba)
```

### Anything in the Host logs that might be useful for us?

```txt
2024-04-18 16:42:05.509 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration start_time which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-04-18 16:42:05.511 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration sonoff which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-04-18 16:42:05.514 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration watchman which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-04-18 16:42:05.517 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-04-18 16:42:05.519 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration alarmo which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-04-18 16:42:05.522 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration eskom_loadshedding which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-04-18 16:42:19.731 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'hacs' accesses hass.components.frontend. This is deprecated and will stop working in Home Assistant 2024.9, it should be updated to import functions used from frontend directly at custom_components/hacs/frontend.py, line 68: hass.components.frontend.async_register_built_in_panel(, please create a bug report at https://github.com/hacs/integration/issues
2024-04-18 16:42:19.795 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'eskom_loadshedding' calls `async_add_job`, which is deprecated and will be removed in Home Assistant 2025.4; Please review https://developers.home-assistant.io/blog/2024/03/13/deprecate_add_run_job for replacement options at custom_components/eskom_loadshedding/__init__.py, line 62: hass.async_add_job(, please create a bug report at https://github.com/swartjean/ha-eskom-loadshedding/issues
2024-04-18 16:42:22.715 WARNING (zeroconf-ServiceBrowser-_googlecast._tcp-118) [pychromecast.dial] Failed to determine cast type for host <unknown> (HTTP Error 403: Forbidden) (services:{MDNSServiceInfo(name='Smart-TV-cdb6a98df088e5e31277d70bed7a09cf._googlecast._tcp.local.')})
2024-04-18 16:42:25.975 WARNING (MainThread) [homeassistant.components.light] sonoff.sonoff_100092e1bc_1 (<class 'custom_components.sonoff.light.XFanLight'>) does not set supported color modes, this will stop working in Home Assistant Core 2025.3, please create a bug report at https://github.com/AlexxIT/SonoffLAN/issues
2024-04-18 16:42:25.977 WARNING (MainThread) [homeassistant.components.light] sonoff.sonoff_100166e6a4_1 (<class 'custom_components.sonoff.light.XFanLight'>) does not set supported color modes, this will stop working in Home Assistant Core 2025.3, please create a bug report at https://github.com/AlexxIT/SonoffLAN/issues
2024-04-18 16:42:25.978 WARNING (MainThread) [homeassistant.components.light] sonoff.sonoff_1001589dd2_1 (<class 'custom_components.sonoff.light.XFanLight'>) does not set supported color modes, this will stop working in Home Assistant Core 2025.3, please create a bug report at https://github.com/AlexxIT/SonoffLAN/issues
2024-04-18 16:42:25.978 WARNING (MainThread) [homeassistant.components.light] sonoff.sonoff_1001589bdb_1 (<class 'custom_components.sonoff.light.XFanLight'>) does not set supported color modes, this will stop working in Home Assistant Core 2025.3, please create a bug report at https://github.com/AlexxIT/SonoffLAN/issues
2024-04-18 16:42:25.980 WARNING (MainThread) [homeassistant.components.light] sonoff.sonoff_100054a715_1 (<class 'custom_components.sonoff.light.XFanLight'>) does not set supported color modes, this will stop working in Home Assistant Core 2025.3, please create a bug report at https://github.com/AlexxIT/SonoffLAN/issues
2024-04-18 16:42:25.980 WARNING (MainThread) [homeassistant.components.light] sonoff.sonoff_1000f6da1b_1 (<class 'custom_components.sonoff.light.XFanLight'>) does not set supported color modes, this will stop working in Home Assistant Core 2025.3, please create a bug report at https://github.com/AlexxIT/SonoffLAN/issues
2024-04-18 16:42:32.056 WARNING (MainThread) [zigpy.application] Zigbee channel 15 utilization is 96.20%!
2024-04-18 16:42:32.056 WARNING (MainThread) [zigpy.application] If you are having problems joining new devices, are missing sensor updates, or have issues keeping devices joined, ensure your coordinator is away from interference sources such as USB 3.0 devices, SSDs, WiFi routers, etc.
2024-04-18 16:48:42.343 ERROR (MainThread) [homeassistant.components.script.1713127889658] Master bedroom AC cool 21 Turbo: Error executing script. Error for call_service at pos 4: Swing mode fixed is not valid. Valid swing modes are: 
2024-04-18 16:48:42.345 ERROR (MainThread) [homeassistant.components.automation.r5_button_5_single] R5 Button 5 Single - AC Cool 21 turbo: Error executing script. Error for call_service at pos 1: Swing mode fixed is not valid. Valid swing modes are: 
2024-04-18 16:48:42.353 ERROR (MainThread) [homeassistant.components.automation.r5_button_5_single] Error while executing automation automation.r5_button_5_single: Swing mode fixed is not valid. Valid swing modes are:
```

### System information

## System Information

version | core-2024.4.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.2
os_name | Linux
os_version | 6.6.25-haos
[... 68 lines cut ...]

oldest_recorder_run | April 15, 2024 at 2:12 AM
-- | --
current_recorder_run | April 18, 2024 at 4:42 PM
estimated_db_size | 154.36 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

<details><summary>Sonoff</summary>

version | 3.6.0 (8dd8af9)
-- | --
cloud_online | 74 / 78
local_online | 71 / 71

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

agners, 2024-04-22:
What do you mean by "abort" exactly? The virtual machine is just disappearing/turning off?

By Orcale VM you mean Oracle's VirtualBox? What version are you using? Anything interesting in the virtual machine log? 

To rule out any other components which might got updated, can you try just downgrade the OS using `ha os update --version 12.1`?

WillemGoosen, 2024-04-23:
Hi

The server turns off with abort next to the server name.

Yes Oracle's VirtualBox.

Sadly I do not know how to downgrade, I normally just roll back to the previous backup prior to upgrade.

I will reapply 12.2 and see what I get from the virtual machine log.

Thanks much appreciated.

Willem Goosen
078 423 4192

WillemGoosen, 2024-04-23:
Hi

I have installed the appropriate VirtualBox Extension Pack and is monitoring.

Will get back with result.

I followed procedures in this:
https://www.wintips.org/fix-virtualbox-aborted-error-status-in-windows-10/

Thanks

Willem Goosen
078 423 4192

WillemGoosen, 2024-04-25:
Hi All

Fixed

Sorry it took so long as the abort does not happen immediately and normally happens overnight.

The reinstallation of the extension pack did not work.

So, I opted to uninstall oracle, reboot the pc, and deleted "C:\Program Files\Oracle folder" with another reboot thereafter.

The reason for this was that the oracle folder still had an ExtensionPack folder and on my Oracle VM VirtualBox Manager under preferences I do not have the extensions menu.

Anyhow I then installed VirtualBox 7.0.14 with its Extension Pack and it has been 24 hours and still going.

Thanks for the help, much appreciated.

Willem Goosen
078 423 4192

https://www.wintips.org/fix-virtualbox-aborted-error-status-in-windows-10/

agners, 2024-04-25:
Thanks for the update! Glad you were able to resolve the problem :tada: !

closed 2024-04-25 by agners