home-assistant/operating-system #3320: /dev/shm missing in 12.2

glyph-se, 2024-04-18:
### Describe the issue you are experiencing

I have been using /dev/shm to store the Home Assistant recorder database.

While I know that using SQLite in memory is not supported it seems to me like using /dev/shm as a memory disk should work like a regular file handle.

I have been using it for few years without any issue. I want to have some history, both for some automations but also for some dashboards, having it cleared upon restart is no problem for me.

There are also other users on Home Assistant forum doing this, [1](https://community.home-assistant.io/t/how-to-keep-your-recorder-database-size-under-control/295795/93) [2](https://community.home-assistant.io/t/recorder-db-in-ram/715912)

Updating Home Assistant Operating System to 12.2 has removed /dev/shm, I see no mentions of it in the release notes so it seems like a regression. If I am wrong and this was intended please enlighten me and close this :)

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

12.2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Add this to config
 ```
recorder:
  db_url: "sqlite:////dev/shm/ha-recorder-v2.db"
```
2. restart Home Assistant

### Anything in the Supervisor logs that might be useful for us?

```txt
Not that I could find
```

### Anything in the Host logs that might be useful for us?

```txt
Not that I could find
```

### System information

## System Information

version | core-2024.4.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.2
os_name | Linux
os_version | 6.1.73-haos-raspi
arch | aarch64
timezone | Europe/Stockholm
config_dir | /config

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.1
-- | --
update_channel | stable
supervisor_version | supervisor-2024.04.0
agent_version | 1.6.0
docker_version | 24.0.7
disk_total | 28.5 GB
disk_used | 8.7 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | Log Viewer (0.17.0), Advanced SSH & Web Terminal (17.2.0), Home Assistant Google Drive Backup (0.112.1), Z-Wave JS (0.4.5), Studio Code Server (5.15.0), UniFi Network Application (3.0.4)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 0
views | 4
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | April 13, 2024 at 20:42
-- | --
current_recorder_run | April 13, 2024 at 22:42
estimated_db_size | 15.15 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

agners, 2024-04-22:
> Updating Home Assistant Operating System to 12.2 has removed /dev/shm, I see no mentions of it in the release notes so it seems like a regression. If I am wrong and this was intended please enlighten me and close this :)

It shouldn't be removed, it might be read-only though due to a behavior change in Docker 25. This got fixed in Supervisor (see #4997). However, it seems that when the container has been created with the old Supervisor version, then Docker 25 gets updated (HAOS 12.2 installed), the fix we did to work around the behavior change doesn't get applied.

Long story short: Use `ha core rebuild` to recreate the Core container. This should bring back shm access.

glyph-se, 2024-04-23:
Hmm, when I check `/dev/shm` using the "Advanced SSH & Web Terminal" addon (without safe mode) it looks writeable. I tested and created an empty file in there just fine. I have limited knowledge about both Home Assistant and docker, maybe something related to startup order?

Though I did try your workaround, `ha core rebuild`, and it works. Recorder now starts properly. Thanks! Is this something that would be needed after each upgrade or just once because of the upgrade to Docker 25?

agners, 2024-04-23:
> Hmm, when I check /dev/shm using the "Advanced SSH & Web Terminal"

That is another container, so not the same world as Core sees.

> I did try your workaround, `ha core rebuild`, and it works. Recorder now starts properly. Thanks! Is this something that would be needed after each upgrade or just once because of the upgrade to Docker 25?

Just because of Docker 25.

Thanks for confirming. Closing then.