home-assistant/operating-system #3323: HAOS freeze randomly

axlFThn, 2024-04-23:
### Describe the issue you are experiencing

Every now and then, HAOS freezes, with HA not responding. After a hard reset, the system is as normal again, though history graphs lack data for a certain amount of time (assumable since the system got frozen). Access through developer SSH not possible (refused connection). Same problem existed on older installation with raspbian+dockers; SSD did not reveal any issue (hdparm)

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

6.1.73-haos-raspi

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Normal operation
2. HA mobile app lacks some data, browser interface cannot update dashboards and functions
3. Power reset Pi
4. System functions normally, but history graph misses data

### Anything in the Supervisor logs that might be useful for us?

```txt
[32m2024-04-23 07:02:47.576 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/45df7312 repository[0m
[32m2024-04-23 07:02:47.581 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/core repository[0m
[32m2024-04-23 07:02:48.319 INFO (MainThread) [supervisor.store] Loading add-ons from store: 81 all - 81 new - 0 remove[0m
[32m2024-04-23 07:02:48.417 INFO (MainThread) [supervisor.addons.manager] Found 8 installed add-ons[0m
[32m2024-04-23 07:02:48.451 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/vscode/aarch64 with version 5.15.0[0m
[32m2024-04-23 07:02:48.453 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-mosquitto with version 6.4.0[0m
[32m2024-04-23 07:02:48.466 INFO (MainThread) [supervisor.docker.interface] Attaching to zigbee2mqtt/zigbee2mqtt-aarch64 with version 1.36.1-1[0m
[32m2024-04-23 07:02:48.468 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-duckdns with version 1.16.0[0m
[32m2024-04-23 07:02:48.475 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-nginx_proxy with version 3.9.0[0m
[32m2024-04-23 07:02:48.484 INFO (MainThread) [supervisor.docker.interface] Attaching to dac3d81b/aarch64-addon-hdd_tools with version 1.1.0[0m
[32m2024-04-23 07:02:48.485 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-ssh with version 9.13.0[0m
[32m2024-04-23 07:02:48.486 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/esphome/esphome-hassio with version 2024.4.0[0m
[32m2024-04-23 07:02:48.540 INFO (MainThread) [supervisor.backups.manager] Found 7 backup files[0m
[32m2024-04-23 07:02:48.559 INFO (MainThread) [supervisor.discovery] Loaded 2 messages[0m
[32m2024-04-23 07:02:48.559 INFO (MainThread) [supervisor.ingress] Loaded 1 ingress sessions[0m
[32m2024-04-23 07:02:48.559 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state setup[0m
[32m2024-04-23 07:02:48.560 INFO (MainThread) [supervisor.resolution.check] System checks complete[0m
[32m2024-04-23 07:02:48.560 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state setup[0m
[32m2024-04-23 07:02:48.561 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete[0m
[32m2024-04-23 07:02:48.561 INFO (MainThread) [supervisor.jobs] 'ResolutionFixup.run_autofix' blocked from execution, system is not running - setup[0m
[32m2024-04-23 07:02:48.567 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state setup[0m
[32m2024-04-23 07:02:48.567 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete[0m
[32m2024-04-23 07:02:48.569 INFO (MainThread) [__main__] Running Supervisor[0m
[32m2024-04-23 07:02:48.582 INFO (MainThread) [supervisor.os.manager] Rauc: B - marked slot kernel.1 as good[0m
[32m2024-04-23 07:02:48.605 INFO (MainThread) [supervisor.addons.manager] Phase 'initialize' starting 0 add-ons[0m
[32m2024-04-23 07:02:48.618 INFO (MainThread) [supervisor.addons.manager] Phase 'system' starting 1 add-ons[0m
[32m2024-04-23 07:02:48.673 INFO (SyncWorker_4) [supervisor.docker.manager] Cleaning addon_core_mosquitto application[0m
[32m2024-04-23 07:02:49.515 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on homeassistant/aarch64-addon-mosquitto with version 6.4.0[0m
[32m2024-04-23 07:02:54.521 INFO (MainThread) [supervisor.addons.manager] Phase 'services' starting 4 add-ons[0m
[32m2024-04-23 07:02:54.581 INFO (SyncWorker_7) [supervisor.docker.manager] Cleaning addon_5c53de3b_esphome application[0m
[32m2024-04-23 07:02:54.994 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on ghcr.io/esphome/esphome-hassio with version 2024.4.0[0m
[32m2024-04-23 07:02:55.053 INFO (SyncWorker_5) [supervisor.docker.manager] Cleaning addon_a0d7b954_vscode application[0m
[32m2024-04-23 07:02:56.015 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on ghcr.io/hassio-addons/vscode/aarch64 with version 5.15.0[0m
[32m2024-04-23 07:02:56.122 INFO (SyncWorker_6) [supervisor.docker.manager] Cleaning addon_core_ssh application[0m
[32m2024-04-23 07:02:57.709 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on homeassistant/aarch64-addon-ssh with version 9.13.0[0m
[32m2024-04-23 07:02:57.828 INFO (SyncWorker_1) [supervisor.docker.manager] Cleaning addon_core_duckdns application[0m
[32m2024-04-23 07:02:59.622 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on homeassistant/aarch64-addon-duckdns with version 1.16.0[0m
[32m2024-04-23 07:03:04.019 INFO (MainThread) [supervisor.services.modules.mqtt] Set core_mosquitto as service provider for mqtt[0m
[32m2024-04-23 07:03:26.163 INFO (MainThread) [supervisor.core] Start Home Assistant Core[0m
[32m2024-04-23 07:03:26.203 INFO (SyncWorker_5) [supervisor.docker.manager] Starting homeassistant[0m
[32m2024-04-23 07:03:26.463 INFO (MainThread) [supervisor.homeassistant.core] Wait until Home Assistant is ready[0m
[32m2024-04-23 07:03:33.298 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state startup[0m
[32m2024-04-23 07:03:33.298 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete[0m
[32m2024-04-23 07:03:36.496 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token[0m
[... 221 lines cut ...]
version_api | ok
installed_addons | Mosquitto broker (6.4.0), ESPHome (2024.4.0), Studio Code Server (5.15.0), Terminal & SSH (9.13.0), NGINX Home Assistant SSL proxy (3.9.0), Duck DNS (1.16.0), Zigbee2MQTT (1.36.1-1), HDD Tools (1.1.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 3
views | 4
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 12 april 2024 kl. 18:35
-- | --
current_recorder_run | 23 april 2024 kl. 07:03
estimated_db_size | 52.67 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

agners, 2024-04-23:
Hm, that does sound like a hardware related issue, it might be the SSD/controller/USB bus locking up.

What is the state of the LED when the system locks up?

What power supply do you use?

If you have a different USB SSD, I'd try that.

axlFThn, 2024-04-23:
The LED for the SSD is blinking like under normal read/write access, but in a more stringent/repetitive way. Power supply is the official Pi-power-supply. I have to purchase another SSD if nothing else helps. I hoped that hdparm would give some information. Is there any way the system logging could be made more verbose to analyse this issue further?

agners, 2024-04-23:
What you can try is connecting a screen and wait for the crash to happen, it might have some indication.

I doubt it is the SSD itself, it is probably more something inbetween, which does not really show up with hdparm.

axlFThn, 2024-04-23:
Ok, I gonna check that.

axlFThn, 2024-05-14:
Tested with another disk (NVMe), together with a usb-hub. But crashes still persist (frequent). Went back to SSD, but changed USB port, which seemed to lower the frequency of crashes. Connected a screen, and finally was able to capture some logs on the CLI. 

Question - if it is not the SSD, where else could the fault be coming from?
[image]

havardthom, 2024-07-26:
Might be this issue relating to Zigbee2MQTT: https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/issues/609

You can try disabling the addon to see if it helps

closed 2024-11-02 by github-actions[bot]

RubenKelevra, 2024-11-15:
@axlFThn have you found a solution? I think I'm maybe having the same issue: https://github.com/home-assistant/operating-system/issues/3674