home-assistant/operating-system #3333: RPI5 Ethernet port LEDs not turned off

hollow-frenk, 2024-04-28:
### Describe the issue you are experiencing

As per the detailed description included in the list of steps to follow to reproduce the problem, although I entered the correct parameters in the config.txt file the leds do not turn off.

### What operating system image do you use?

rpi5-64  (Raspberry Pi 5 64-bit OS)

### What version of Home Assistant Operating System is installed?

6.1.73-haos-raspi

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Type login in the home assistant shell then press enter
2. Type vi /mnt/boot/config.txt then press enter
3. Edit the file by entering  <br>`dtparam=eth_led0=4<br>dtparam=eth_led1=4`
4. Save your changes and restart the system
5. The LEDs on the Ethernet port should be turned off but they are not.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
No
```

### Anything in the Host logs that might be useful for us?

```txt
No
```

### System information

## System Information

version | core-2024.4.4
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.2
os_name | Linux
os_version | 6.1.73-haos-raspi
arch | aarch64
timezone | Europe/Rome
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.34.0
Stage | running
Available Repositories | 1401
Downloaded Repositories | 9
HACS Data | ok

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.2
-- | --
update_channel | stable
supervisor_version | supervisor-2024.04.0
agent_version | 1.6.0
docker_version | 25.0.5
disk_total | 228.5 GB
disk_used | 6.4 GB
healthy | true
supported | true
board | rpi5-64
supervisor_api | ok
version_api | ok
installed_addons | Home Assistant Google Drive Backup (0.112.1), Samba share (12.3.1), Tailscale (0.18.0), Advanced SSH & Web Terminal (17.2.0), Filebrowser (2.23.0_14)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 2
views | 11
mode | storage

</details>

<details><summary>NextDNS</summary>

can_reach_server | ok
-- | --

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 20 aprile 2024 alle ore 22:51
-- | --
current_recorder_run | 28 aprile 2024 alle ore 19:54
estimated_db_size | 218.75 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

fkbidi, 2024-06-04:
Hi, I have the same problem. It seems to be due to the firmware. The parameters have not been implemented and a firmware update is required. However, I don't know if this is possible without updating the whole HA.

ArthFink, 2024-07-14:
This is the code that needs to be added to the config.txt when using rasbian-64bit. All LEDs are off.

``` bash
# Turn off Power LED
dtparam=pwr_led_trigger=default-on
dtparam=pwr_led_activelow=off
# Turn off Activity LED
dtparam=act_led_trigger=none
dtparam=act_led_activelow=off
# Turn off Ethernet ACT LED
dtparam=eth_led0=45
# Turn off Ethernet LNK LED
dtparam=eth_led1=44 
```

When moving this configuration, only the pw led is off, the eth_leds stay on. I have tried locating the device tree to see how the pins are mapped, but I could not find it in the repo.

beeh5, 2024-10-13:
Issue not resolved yet...

HOCHH4U5JUMP3R, 2024-11-29:
This issue still persists. 
Core 2024.12.0b2
Supervisor 2024.11.4
Operating System 14.0.rc2
Frontend 20241127.1

Only the parameters for the ethernet LEDs are not working.
# Turn off Ethernet ACT LED
dtparam=eth_led0=4
# Turn off Ethernet LNK LED
dtparam=eth_led1=4

beeh5, 2025-02-28:
Issue not resolved yet. 
Core 2025.2.5
Supervisor 2025.02.4
Operating System 14.2
Frontend 20250214.0

danez, 2025-04-18:
I have the same issue. AFAIK the params were introduced in the [device tree](https://github.com/raspberrypi/linux/pull/5826) over a year ago. So maybe something needs to be updated?

PR #4137 (2025-07-02): Make all Raspberry Pi LAN drivers built-in
Make sure that all LAN drivers used on Raspberry Pi boards are built-in. Although they are defined as such in the base defconfig, we change them to modules in device support includes. For simplicity and keeping kernel config close to the RPi OS config, change them all to built-in in the main RPi include for all RPi targets.

This is not only a formal change - at least one regression is known if the PHY driver on RPi 5 is not built-in and MAC driver is - in that case the PHY hooked up to the RP1 isn't initialized properly, and it is reported as "Generi [... cut]
commit a338b671 (2025-07-02): Make all Raspberry Pi LAN drivers built-in (#4137)

Make sure that all LAN drivers used on Raspberry Pi boards are built-in.
Although they are defined as such in the base defconfig, we change them to
modules in device support includes. For simplicity and keeping kernel config
close to the RPi OS config, change them all to built-in in the main RPi include
for all RPi targets.

This is not only a formal change - at least one regression is known if the PHY
driver on RPi 5 is not built-in and MAC driver is - in that case the PHY hooked
up to the RP1 isn't initialized properly, and it is reported a [... cut]
closed 2025-07-02 by sairon