home-assistant/operating-system #3344: System errors - USB stack crashing?

alfwro13, 2024-05-07:
### Describe the issue you are experiencing

I think I started having this problem after recent Home Assistant OS upgrade. Before that it was rock solid. I am running my home assistant as a proxmox VM on Intel NUC hardware.
The problem is that every couple of days/weeks the randomly USB devices or bluetooth devices drop off from HA. Looking at the home assistant os screen I can see the following errors:
Bluetooth: hci0: Opcode 0x2042 failed: -110
Bluetooth: hci0: command 0x2042 tx timeout
Bluetooth: Resetting usb device
usb 2-10: device descriptor read/64, error 110
usb usb2-port10: unable to enumerate USB device

The only fix so far is to restart Home assistant OS.
I have 3 USB devices connected (passthrough to my HA OS VM) - zwave USB dongle, coral USB accelerator and arduino running fireangel alarm monitor. I have been using those 3 devices for the past 3-4 years without any problem until 2-3 months ago when I upgraded my HA OS.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

6.6.25-haos

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Keep normaly using HA
2. Wait a couple of days/weeks
3. When USB connected devices stop working - check HA OS console screen for errors
4.  Restart HA OS to fix the problem
...

### Anything in the Supervisor logs that might be useful for us?

```txt
nothing interesting in supervisor logs
```

### Anything in the Host logs that might be useful for us?

```txt
from home-assistant.log
Traceback (most recent call last):
  File "src/habluetooth/scanner.py", line 306, in habluetooth.scanner.HaScanner._async_start_attempt
  File "src/habluetooth/scanner.py", line 306, in habluetooth.scanner.HaScanner._async_start_attempt
  File "src/habluetooth/scanner.py", line 309, in habluetooth.scanner.HaScanner._async_start_attempt
  File "/usr/local/lib/python3.12/site-packages/bleak/__init__.py", line 198, in start
    await self._backend.start()
  File "/usr/local/lib/python3.12/site-packages/bleak/backends/bluezdbus/scanner.py", line 178, in start
    self._stop = await manager.passive_scan(
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/bleak/backends/bluezdbus/manager.py", line 527, in passive_scan
    assert_reply(reply)
  File "/usr/local/lib/python3.12/site-packages/bleak/backends/bluezdbus/utils.py", line 22, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.freedesktop.DBus.Error.UnknownObject] Method "RegisterMonitor" with signature "o" on interface "org.bluez.AdvertisementMonitorManager1" doesn't exist

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "src/habluetooth/scanner.py", line 505, in habluetooth.scanner.HaScanner._async_restart_scanner
  File "src/habluetooth/scanner.py", line 263, in _async_start
  File "src/habluetooth/scanner.py", line 350, in _async_start_attempt
habluetooth.scanner.ScannerStartError: hci0 (F8:AC:65:C6:3C:DD): Failed to start Bluetooth: [org.freedesktop.DBus.Error.UnknownObject] Method "RegisterMonitor" with signature "o" on interface "org.bluez.AdvertisementMonitorManager1" doesn't exist
; Try power cycling the Bluetooth hardware.
2024-05-07 00:33:43.887 ERROR (MainThread) [custom_components.frigate.api] Error fetching information from http://ccab4aaf-frigate-fa:5000/api/stats: Cannot connect to host ccab4aaf-frigate-fa:5000 ssl:default [Connect call failed ('172.30.33.7', 5000)]
2024-05-07 00:34:05.408 ERROR (MainThread) [bluetooth_auto_recovery.recover] hciNone [F8:AC:65:C6:3C:DD] seems not to exist (anymore), check BT interface mac address in your settings; Available adapters: {} 
2024-05-07 00:34:08.889 ERROR (MainThread) [custom_components.frigate.api] Error fetching information from http://ccab4aaf-frigate-fa:5000/api/stats: Cannot connect to host ccab4aaf-frigate-fa:5000 ssl:default [Connect call failed ('172.30.33.7', 5000)]
2024-05-07 00:34:08.889 ERROR (MainThread) [custom_components.frigate] Error fetching frigate data: 
2024-05-07 00:34:09.916 ERROR (MainThread) [habluetooth.scanner] hci0 (F8:AC:65:C6:3C:DD): Failed to restart Bluetooth scanner: hci0 (F8:AC:65:C6:3C:DD): Failed to start Bluetooth: [org.freedesktop.DBus.Error.UnknownObject] Method "RegisterMonitor" with signature "o" on interface "org.bluez.AdvertisementMonitorManager1" doesn't exist
; Try power cycling the Bluetooth hardware.
Traceback (most recent call last):
  File "src/habluetooth/scanner.py", line 306, in habluetooth.scanner.HaScanner._async_start_attempt
  File "src/habluetooth/scanner.py", line 306, in habluetooth.scanner.HaScanner._async_start_attempt
  File "src/habluetooth/scanner.py", line 309, in habluetooth.scanner.HaScanner._async_start_attempt
  File "/usr/local/lib/python3.12/site-packages/bleak/__init__.py", line 198, in start
    await self._backend.start()
  File "/usr/local/lib/python3.12/site-packages/bleak/backends/bluezdbus/scanner.py", line 178, in start
    self._stop = await manager.passive_scan(
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/bleak/backends/bluezdbus/manager.py", line 527, in passive_scan
    assert_reply(reply)
  File "/usr/local/lib/python3.12/site-packages/bleak/backends/bluezdbus/utils.py", line 22, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.freedesktop.DBus.Error.UnknownObject] Method "RegisterMonitor" with signature "o" on interface "org.bluez.AdvertisementMonitorManager1" doesn't exist
```

### System information

## System Information

version | core-2024.5.0
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.2
os_name | Linux
os_version | 6.6.25-haos
arch | x86_64
timezone | Europe/London
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.34.0
Stage | running
Available Repositories | 1486
Downloaded Repositories | 66

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.2
-- | --
update_channel | stable
supervisor_version | supervisor-2024.04.4
agent_version | 1.6.0
docker_version | 25.0.5
disk_total | 146.9 GB
disk_used | 105.2 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.8.0), InfluxDB (5.0.0), Mosquitto broker (6.4.0), chrony (3.0.1), AdGuard Home (5.0.7), Advanced SSH & Web Terminal (17.2.0), Samba share (12.3.1), Vaultwarden (Bitwarden) (0.21.2), ESPHome (2024.4.1), SQLite Web (4.1.2), Log Viewer (0.17.0), Grafana (9.2.2), Z-Wave JS UI (3.6.0), Frigate (Full Access) (0.13.2), MQTT Explorer (browser-1.0.1), Signal Messenger (6bc2eb0), Portainer (2.20.1), AirCast (4.2.1), Piper (1.5.0), Whisper (2.0.0), GivTCP (2.4.3), Lovelace Kindle Screensaver (1.0.10), openWakeWord (1.10.0), Firefly iii (6.1.15), Firefly iii Data Importer (20240414.1), Signal Messenger (0.83.0), AppDaemon (0.16.5), Music Assistant BETA (2.0.0b145)

</details>

<details><summary>Dashboards</summary>

dashboards | 6
-- | --
resources | 43
views | 51
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 6 April 2024 at 12:25
-- | --
current_recorder_run | 5 May 2024 at 18:28
estimated_db_size | 16086.53 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

<details><summary>Solcast PV Forecast</summary>

can_reach_server | ok
-- | --
used_requests | 4
rooftop_site_count | 2

</details>

alfwro13, 2024-05-07:
and a screenshot of the home assistant os console screen:
[image]

Eskander, 2024-05-10:
I have the same issue running HA on bare metal NUC. It also started around 3 months ago, before that my system was running fine for a year.

Do you happen to also have an Intel AX201 CNVi Wi-Fi/Bluetooth card?

alfwro13, 2024-05-10:
yep - same one - I pass-through the whole USB controller in proxmox:

00:14.0 USB controller: Intel Corporation Comet Lake PCH-LP USB 3.1 xHCI Host Controller
00:14.2 RAM memory: Intel Corporation Comet Lake PCH-LP Shared SRAM
**00:14.3 Network controller: Intel Corporation Comet Lake PCH-LP CNVi WiFi**

My proxmox is running on:  Intel NUC i7-10710U

Eskander, 2024-05-10:
It seems there's a firmware and/or driver issue with these cards causing them to crash. I've been trying different combinations for a while with no success. [See](https://forum.manjaro.org/t/linux-firmware-and-intel-ax20x-wifi-bluetooth-cards/104743)

alfwro13, 2024-05-13:
Upgraded to 12.3 and unfortunately this is still happening.

alfwro13, 2024-05-22:
> It seems there's a firmware and/or driver issue with these cards causing them to crash. I've been trying different combinations for a while with no success. [See](https://forum.manjaro.org/t/linux-firmware-and-intel-ax20x-wifi-bluetooth-cards/104743)

I was looking at that link but I cannot figure out how to run any normal linux commands on home assistant OS.

closed 2024-08-28 by github-actions[bot]