home-assistant/operating-system #3351: Upgrade to 12.3,, now unable to boot.

csimonosian, 2024-05-08:
### Describe the issue you are experiencing

Upgraded my Odioriud to 2.13, now HA is not accessible.

### What operating system image do you use?

odroid-n2 (Hardkernel ODROID-N2/N2+)

### What version of Home Assistant Operating System is installed?

12.3

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Accept the installation from HA's prompt that a new release is available.
2. That's it. The instance becomes unreachable.

### Anything in the Supervisor logs that might be useful for us?

```txt
Cannot get to supervisor logs, the system is down.
```

### Anything in the Host logs that might be useful for us?

```txt
No.
```

### System information

CVannot get to System Information because the system is unreachable.

csimonosian, 2024-05-08:
Booting with a keyboard and monitor attached produces this
[image]

agners, 2024-05-08:
Ugh :cry: 

What eMMC size/type do you have?

tengels, 2024-05-08:
I have the "standard" [64GB eMCC from Hard Kernel](https://www.hardkernel.com/shop/64gb-emmc-module-n2-linux/)

csimonosian, 2024-05-08:
Mine was the 32GB eMMC from the UK distributor. https://www.odroid.co.uk/odroid-accessories/odroid-os?product_id=1118.  I have been preparing myself for a clean install of 12.2!

aumuelle, 2024-05-08:
very same problem here on ODROID N2+ 4GB and eMMC 5.1, 64 GB from ALLNET (https://www.reichelt.at/at/de/entwicklerboards-emmc-5-1-64-gb-erweiterungsspeicher-debo-emmc5-0-64-p248578.html?search=emmc&&r=1)

macrouch, 2024-05-09:
I'm having the same problem. I don't understand much of this. How do I undo what happened in #3319? All I have is an `HAOS>` prompt and I can't control anything in my house now 😠

stleng08, 2024-05-09:
Same issue.  Wife mad.  Awaiting a fix or way back to get out of the dog house.

aumuelle, 2024-05-09:
For those who need a "way back now":  If you don't have a recent backup (?!?) .... 1 out of 10 or 20 boot attempts seems to be successfully here on my side. So perhaps you are lucky too. 
--> make a full backup; download it somewhere; get https://github.com/home-assistant/operating-system/releases/download/12.2/haos_odroid-n2-12.2.img.xz; flash your emmc with it; restore from your backup during onboarding

dw01113, 2024-05-09:
Having same issue, ODROID N2+ 4GB and eMMC 64GB. As prevoius poster said it seems to boot 12.2 sucessfully occasionally, so if your lucky then make sure you get a backup downloaded when it does. Is there a way to set it to always boot from the "good" 12.2 slot?

sairon, 2024-05-09:
Could you please all check what eMMC manufacturer is used on the module you are using? Either send a close-up picture of the module (ideally with the sticker removed) or check for one of the following:
- QR code in top left corner = Samsung
- "M" with an orbit around it = Micron
- "K" or "Ki" of the Kingston logo in top left corner = Kingston
- Anything else? (e.g. the third-party eMMC @aumuelle posted has FORESEE module on the product images)

We tried to test all possible eMMC cards with Hardkernel devs but there might still be some nuances we missed :cold_sweat:

dw01113, 2024-05-09:
@sairon Here is a picture of the emmc, it had a large "64" sticker on it, when removed it has taken the detail with it 

[image]

sairon, 2024-05-09:
@dw01113 Thank you, it's a bit obscured by the glue residue but the QR code and marking placement corresponds with the Samsung eMMC chips. I wonder if we find any pattern with other reports :eyes:

dw01113, 2024-05-09:
@sairon Doh! Yes it was the glue, here's a cleaned up picture

[image]

gaco79, 2024-05-09:
Same problem here. Re-flashed 12.2 and still refusing to boot from eMMC 😟
[image]

gaco79, 2024-05-09:
As a quick update to above, I've reflashed 12.1 and now seems to be booting. Not sure if that was my fault with the flashing process for 12.2 or something else.

bkvargyas, 2024-05-09:
I have a home assistant blue, and 12.3 flashed and works fine for me.     I have a couple more laying around somewhere I can try if it would help.

csimonosian, 2024-05-09:
Mine still with sticker inside the case...but it looks like a Samsung with the barcode in the corner.
[image]

csimonosian, 2024-05-09:
Well...I've just been getting it ready to flash 12.2 to it, but I couldn't get it to boot into SPI mode, however it's now booting seemingly correctly into HA! I'm not getting the error messages I first posted above, there are other messages about checksum errors on backup slot but then moves switflty on to booting into HA command line. And I can now access it. I just don't know what's going on and whether it's going to be stable. It wouldn't boot at all last night, now it's OK but with those checksum errors.

smarthomefamilyverrips, 2024-05-09:
> Well...I've just been getting it ready to flash 12.2 to it, but I couldn't get it to boot into SPI mode, however it's now booting seemingly correctly into HA! I'm not getting the error messages I first posted above, there are other messages about "invalid checksum on backup superblock" but then moves switflty on to booting into HA command line. And I can now access it. I just don't know what's going on and whether it's going to be stable. It wouldn't boot at all last night, now it's OK but with those checksum errors.
> 
> [EDIT : ah, found out the problem, it's not booting with my USB Z-Wave dongle plugged in, with the errors I first posted above. If I remove it, it boots. I had taken it out in order to open the case. So is it not an eMMC problem but a USB problem,, as noted in other reports here about 12.3 not booting?]

@csimonosian but you are on 12.2 now not on 12.3 .... your original problem occurred after update from 12.2 to 12.3 .... so not think is related to USB, the USB dongle issue posted also is not on a Odroid N2 platform btw

csimonosian, 2024-05-09:
@smarthomefamilyverrips No, I'm still on 12.3. I didn't flash it back to 12.2 because I couldn't get it to boot into SPI mode, the switch acted as though it was still in eMMC mode and booted from eMMC.

smarthomefamilyverrips, 2024-05-09:
> @smarthomefamilyverrips No, I'm still on 12.3. I didn't flash it back to 12.2 because I couldn't get it to boot into SPI mode, the switch acted as though it was still in eMMC mode and booted from eMMC.

@csimonosian Ha OK now I see I did read to quick and did miss read the "getting ready" as flashed 12.2 ... sorry my mistake 🫣

gaco79, 2024-05-09:
I have USB Sonoff Zigbee coordinator and Edge TPU Coral USB. Still wouldn't boot (wasn't even posting) when I unplugged these to go to a monitor. But then, wouldn't post when I thought I'd flashed 12.2 either...

stleng08, 2024-05-09:
> @sairon Doh! Yes it was the glue, here's a cleaned up picture
> 
> ![emmc](https://private-user-images.githubusercontent.com/94217352/329208311-7d528e91-9159-4d50-9038-01e994a6031a.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTUyNzY0MzMsIm5iZiI6MTcxNTI3NjEzMywicGF0aCI6Ii85NDIxNzM1Mi8zMjkyMDgzMTEtN2Q1MjhlOTEtOTE1OS00ZDUwLTkwMzgtMDFlOTk0YTYwMzFhLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTA5VDE3MzUzM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTEzZWI3OGI4YTllOWFmMzEzNTlkNDQzYmRlYmU4Y2I4NTg5YjkxYzJmNDdiOThiMzQ4ZDA3ODFlMDUzMmI1ODQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.NBtC2Qz1oO4C3RN-tQKBV5D_1-WzuPaWtO-iLGuZgjg)

I have this too.

stleng08, 2024-05-10:
> For those who need a "way back now": If you don't have a recent backup (?!?) .... 1 out of 10 or 20 boot attempts seems to be successfully here on my side. So perhaps you are lucky too. --> make a [...]

Thank you for this.  Back up and running on 12.2!  I am out of the dog house!  I would say I got lucky between 10 and 20 boots it actually went on 12.3 got back up, reflashed 12.2, restored backup, Bazinga!

csimonosian, 2024-05-12:
I've tested this further, as mentioend I can boot into 12.3 by unplugging the USB Z-Wave dongle. Obviously, all my Z-wave devices are then Unavailable. But if I plug the dongle back in, it works and looks like nothing is wrong.

Was there any change related to USB made in 12.3? As also mentioned, USB seems to be a frequent factor in the many reports of boot issues with 12.3.

dw01113, 2024-05-13:
Just wanted to add that my situation does not seem to be affected by USB devices, after performing 12.3 upgrade and it failed, it will boot into 12.2 occasioanly (I didn't downgrade it must have not upgraded correctly) and work. So have left it running until hopefully a new version is available.

Fernando-A-Rocha, 2024-05-18:
Having the same problem. Won't boot. Says serverip not set and Config file not found. Cannot autoload with TFTPGET

Fernando-A-Rocha, 2024-05-18:
> Booting with a keyboard and monitor attached produces this ![IMG_20240508_213120448_HDR](https://private-user-images.githubusercontent.com/72629997/329040903-e426e001-fb06-41d1-810d-2677040d1afe.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTYwNDI2MDYsIm5iZiI6MTcxNjA0MjMwNiwicGF0aCI6Ii83MjYyOTk5Ny8zMjkwNDA5MDMtZTQyNmUwMDEtZmIwNi00MWQxLTgxMGQtMjY3NzA0MGQxYWZlLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTE4VDE0MjUwNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQ4MTdjMDNhZTc0NDI0ZWUzMDVmOGI1ZWIxN2UzNGEyMDBiZmFkMDUyZjMwMWExMDNiNWVjMDk5NGRiYTQ3OWEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.xM2V6lNGSd9gcmLc91c1JA0sigWn979L5SLwtt0lL2U)

I have the exact same issue! Any solution please?

sairon, 2024-06-18:
Closing this in favor of #3227 which was probably never resolved fully.

Currently the 12.4 release will contain a revert that will use a U-Boot version that was proven working until 11.5, however the U-Boot update will inevitably have to be done in the future, so we need to come up with a better fix. If you had issues with the ODROID-N2 booting with any of the 11.5+ releases with `Card did not respond to voltage select! : -110` error on the HDMI console, please see https://github.com/home-assistant/operating-system/issues/3227#issuecomment-2175983018 and get in touch with me :pray:

closed 2024-06-18 by sairon