home-assistant/operating-system #3352: Home Assistant OS 12.3 breaks ZHA on raspberry pi 5

BeardElk, 2024-05-08:
### Describe the issue you are experiencing

After upgrade my Home Assistant SkyConnect/ZHA is no longer working. I've tried every possible reboot and restore from config and nothing works, changed ports, etc. 

### What operating system image do you use?

rpi5-64  (Raspberry Pi 5 64-bit OS)

### What version of Home Assistant Operating System is installed?

12.3

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Have Home Assistant SkyConnect installed on a Raspberry Pi 5 running Home Assistant OS 12.2
2.Upgrade Home Assistant OS to 12.3
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
No.
```

### Anything in the Host logs that might be useful for us?

```txt
No.
```

### System information

<html>
<body>
<div>
Version | core-2024.5.2
-- | --
Installation Type | Home Assistant OS
Development | false
Supervisor | true
Docker | true
User | root
Virtual Environment | false
Python Version | 3.12.2
Operating System Family | Linux
Operating System Version | 6.1.73-haos-raspi
CPU Architecture | aarch64
Timezone | Europe/Stockholm
Configuration Directory | /config

</div><div><h3>Core metrics</h3><ha-metric><ha-settings-row style="display: flex; padding: 0px; align-content: normal; align-self: auto; align-items: center; height: 54px; width: 535px;"><div class="prefix-wrap" style="display: var(--settings-row-prefix-display);"><slot name="prefix"></slot><div class="body" two-line="" style="padding-top: 8px; padding-bottom: 8px; padding-left: 0px; padding-inline: 0px 16px; overflow: hidden; display: var(--layout-vertical_-_display); flex-direction: var(--layout-vertical_-_flex-direction); justify-content: var(--layout-center-justified_-_justify-content); flex: 1 1 0%; min-height: calc(var(--paper-item-body-two-line-min-height,72px) - 16px);"><slot name="heading"><span slot="heading">Processor usage</span></slot><div class="secondary" style="white-space: normal; display: block; padding-top: 4px; font-family: var(
          --mdc-typography-body2-font-family,
          var(--mdc-typography-font-family, Roboto, sans-serif)
        ); -webkit-font-smoothing: antialiased; font-size: var(--mdc-typography-body2-font-size, .875rem); font-weight: var(--mdc-typography-body2-font-weight,400); line-height: normal; color: var(--secondary-text-color);"><slot name="description"><div slot="description" title="" style="white-space: normal; color: var(--secondary-text-color); display: flex; justify-content: space-between;"><span class="value" style="width: 48px; padding-right: 4px; padding-inline-end: 4px; padding-inline-start: initial; flex-shrink: 0;"><div style="direction: ltr; text-align: var(--float-start);">0.9<span> </span>%</div></span><ha-bar class="  " style="--ha-bar-primary-color: var(
          --metric-bar-ok-color,
          var(--success-color)
        );"><svg><g><rect></rect><rect width="0.9%"></rect></g></svg></ha-bar></div></slot></div></div></div><div class="content" style="display: var(--settings-row-content-display,flex); justify-content: flex-end; flex: 1 1 0%; padding: 16px 0px;"><slot></slot></div></ha-settings-row></ha-metric><ha-metric><ha-settings-row style="display: flex; padding: 0px; align-content: normal; align-self: auto; align-items: center; height: 54px; width: 535px;"><div class="prefix-wrap" style="display: var(--settings-row-prefix-display);"><slot name="prefix"></slot><div class="body" two-line="" style="padding-top: 8px; padding-bottom: 8px; padding-left: 0px; padding-inline: 0px 16px; overflow: hidden; display: var(--layout-vertical_-_display); flex-direction: var(--layout-vertical_-_flex-direction); justify-content: var(--layout-center-justified_-_justify-content); flex: 1 1 0%; min-height: calc(var(--paper-item-body-two-line-min-height,72px) - 16px);"><slot name="heading"><span slot="heading">Memory usage</span></slot><div class="secondary" style="white-space: normal; display: block; padding-top: 4px; font-family: var(
          --mdc-typography-body2-font-family,
          var(--mdc-typography-font-family, Roboto, sans-serif)
        ); -webkit-font-smoothing: antialiased; font-size: var(--mdc-typography-body2-font-size, .875rem); font-weight: var(--mdc-typography-body2-font-weight,400); line-height: normal; color: var(--secondary-text-color);"><slot name="description"><div slot="description" title="" style="white-space: normal; color: var(--secondary-text-color); display: flex; justify-content: space-between;"><span class="value" style="width: 48px; padding-right: 4px; padding-inline-end: 4px; padding-inline-start: initial; flex-shrink: 0;"><div style="direction: ltr; text-align: var(--float-start);">12.2<span> </span>%</div></span><ha-bar class="  " style="--ha-bar-primary-color: var(
          --metric-bar-ok-color,
          var(--success-color)
        );"><svg><g><rect></rect><rect width="12.2%"></rect></g></svg></ha-bar></div></slot></div></div></div><div class="content" style="display: var(--settings-row-content-display,flex); justify-content: flex-end; flex: 1 1 0%; padding: 16px 0px;"><slot></slot></div></ha-settings-row></ha-metric><h3>Supervisor metrics</h3><ha-metric><ha-settings-row style="display: flex; padding: 0px; align-content: normal; align-self: auto; align-items: center; height: 54px; width: 535px;"><div class="prefix-wrap" style="display: var(--settings-row-prefix-display);"><slot name="prefix"></slot><div class="body" two-line="" style="padding-top: 8px; padding-bottom: 8px; padding-left: 0px; padding-inline: 0px 16px; overflow: hidden; display: var(--layout-vertical_-_display); flex-direction: var(--layout-vertical_-_flex-direction); justify-content: var(--layout-center-justified_-_justify-content); flex: 1 1 0%; min-height: calc(var(--paper-item-body-two-line-min-height,72px) - 16px);"><slot name="heading"><span slot="heading">Processor usage</span></slot><div class="secondary" style="white-space: normal; display: block; padding-top: 4px; font-family: var(
          --mdc-typography-body2-font-family,
          var(--mdc-typography-font-family, Roboto, sans-serif)
        ); -webkit-font-smoothing: antialiased; font-size: var(--mdc-typography-body2-font-size, .875rem); font-weight: var(--mdc-typography-body2-font-weight,400); line-height: normal; color: var(--secondary-text-color);"><slot name="description"><div slot="description" title="" style="white-space: normal; color: var(--secondary-text-color); display: flex; justify-content: space-between;"><span class="value" style="width: 48px; padding-right: 4px; padding-inline-end: 4px; padding-inline-start: initial; flex-shrink: 0;"><div style="direction: ltr; text-align: var(--float-start);">0<span> </span>%</div></span><ha-bar class="  " style="--ha-bar-primary-color: var(
          --metric-bar-ok-color,
          var(--success-color)
        );"><svg><g><rect></rect><rect width="0%"></rect></g></svg></ha-bar></div></slot></div></div></div><div class="content" style="display: var(--settings-row-content-display,flex); justify-content: flex-end; flex: 1 1 0%; padding: 16px 0px;"><slot></slot></div></ha-settings-row></ha-metric><ha-metric><ha-settings-row style="display: flex; padding: 0px; align-content: normal; align-self: auto; align-items: center; height: 54px; width: 535px;"><div class="prefix-wrap" style="display: var(--settings-row-prefix-display); color: rgb(229, 229, 229); font-family: Roboto, sans-serif; font-size: 14px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: 0.4375px; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; background-color: rgb(12, 12, 12); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><slot name="prefix"></slot><div class="body" two-line="" style="padding-top: 8px; padding-bottom: 8px; padding-left: 0px; padding-inline: 0px 16px; overflow: hidden; display: var(--layout-vertical_-_display); flex-direction: var(--layout-vertical_-_flex-direction); justify-content: var(--layout-center-justified_-_justify-content); flex: 1 1 0%; min-height: calc(var(--paper-item-body-two-line-min-height,72px) - 16px);"><slot name="heading"><span slot="heading">Memory usage</span></slot><div class="secondary" style="white-space: normal; display: block; padding-top: 4px; font-family: var(
          --mdc-typography-body2-font-family,
[... cut ...] lass="value" style="width: 48px; padding-right: 4px; padding-inline-end: 4px; padding-inline-start: initial; flex-shrink: 0;"><div style="direction: ltr; text-align: var(--float-start);">4.1<span> </span>%</div></span><ha-bar class="  " style="--ha-bar-primary-color: var(
          --metric-bar-ok-color,
          var(--success-color)
        );"><svg><g><rect></rect><rect width="4.1%"></rect></g></svg></ha-bar></div></slot></div></div></div><br class="Apple-interchange-newline">
</body>
</html>

agners, 2024-05-08:
There is an updated kernel with the latest release, but it is the same kernel as Raspberry Pi OS is using. But quite possible that in your case this indeed caues issues :cry: 

Do yo use Hubs or similar inbetween? Did you try different ports?

You can you try boot the previous HAOS installation with:
```
ha os boot-slot other
```

revenpl, 2024-05-09:
I had the same problem on Raspberry Pi 3. Disabling Zigbee Home Automation integration and going back to previous version using command mentioned above fixed the issue. Now after reenabling ZHA everything seems to be working fine. I'm skipping 12.3 for now.

BeardElk, 2024-05-09:
I was able to go back to 12.2 and everything is working fine now, so I as well am skipping 12.3 for now. 
I usually have a short extension cable between the pi and sky connect stick, but i tried removing it and tried all the ports without any success.

gnmancini, 2024-05-09:
I had the same problem with Rasberry Pi3. To solve the issue I did a full backup and reinstallation going back to 12.2.
@agners handy command, I will use next time.

Keribi, 2024-05-09:
Zigbee broken immediately after updating to 12.3.  Pi 4, Sonoff Dongle, ZHA devices erratic and very unstable.

BeardElk, 2024-05-09:
I don't know why, but a full system reboot breaks it for me every time now, even on 12.2. 
Only way I can get it to work again is using a backup from yesterday morning (I do daily backups to my nas). 
So its working now, but I don't get one fault or anything in the logs when its not working, just says "zha failed to set up", annoying af.

joaohess, 2024-05-11:
Same problem here, but with Z2M. Managed to fix it by booting with the coordinator (here, a Sonoff Dongle-E EFR32MG21) unplugged and plugging it in after boot. 

A reboot with the dongle plugged in breaks the OS.

Keribi, 2024-05-11:
Zigbee useless after 12.3 update.  Downgraded to 12.2 and issue still exists..  Devices intermittently responsive, brightness levels are wrong etc...  this needs remedied soon! Raspi 4, sonoff dongle, ZHA

gnmancini, 2024-05-11:
@Keribi Have you tried to reboot HA with OS 12.2 without the ZHA dongle and plug it in after the reboot is completed.

BeardElk, 2024-05-11:
> @Keribi Have you tried to reboot HA with OS 12.2 without the ZHA dongle and plug it in after the reboot is completed?

I tried it and it makes no difference for me. 
Mine randomly stopped working the night to today, so I had to restore from the last working backup which was may 8.

edwin-completesolutions, 2024-05-11:
Same problem here with ConbeeII Zigbee dongle. Remove the usb zigbee dongle and it boots.
See https://www.home-assistant.io/common-tasks/os/ to switch back to OS 12.2

Keribi, 2024-05-13:
System seems stable on 12.2. My system was working reliably until the 12.3 update which immediately broke Zigbee.  Whether it is a ZHA integration issue or a USB stack issue, the update caused this problem.

walterbressers, 2024-05-13:
My Proxmox HAOS 12.2 ZHA with Conbee II setup was broken after the 12.3 upgrade, but going back to 12.2 (with the command from above) didn't fix the issue. On a separate VM on my Synology ZHA with Conbee II is working. I don't get it. I tried al sorts of setups, even new HA VM's in Proxmox but can't get to work. Strange thing is that Deconz within HAOS 12.3 does work but not the ZHA integration.

tlokaj, 2024-05-13:
I have the same issue on RPI4, I've tried everything mentioned above but nothing helped. ZHA integration cannot start.
[image]

BeardElk, 2024-05-13:
> I have the same issue on RPI4, I've tried everything mentioned above but nothing helped. ZHA integration cannot start. ![Screenshot_20240513-210238](https://private-user-images.githubusercontent.com/7229483/330175108-50a64d02-281a-48a5-bb94-3687e2c62636.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU2MjczODAsIm5iZiI6MTcxNTYyNzA4MCwicGF0aCI6Ii83MjI5NDgzLzMzMDE3NTEwOC01MGE2NGQwMi0yODFhLTQ4YTUtYmI5NC0zNjg3ZTJjNjI2MzYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDUxMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA1MTNUMTkwNDQwWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NDQ4MDBhZDY0ZDJlMjc0NjI4YzZiZDIxNGJjNWNlNWQ3Y2E1MjEyNTcwM2E4ZWY0Mzk0Zjg2YmFhNzZjYmQyZSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.pGs38suxdU87RjxQSqKDmt0v1KAu5D9masoLc-MXxgE)

I ssh into my RPI5 and used `ha os upgrade --version 12.2` to downgrade home assistant os to 12.2, then I used a working backup from when I was on 12.2, and that got it working for me! 
I've since upgraded the core to 2024.5.3 without issue.

tlokaj, 2024-05-14:
> I ssh into my RPI5 and used `ha os upgrade --version 12.2` to downgrade home assistant os to 12.2, then I used a working backup from when I was on 12.2, and that got it working for me! I've since [...]

That one helped me, thank you!!!

BeardElk, 2024-05-17:
So I've gotten everything to work with 12.3 now, for some reason was the Silicon Labs Multiprotocol Add-on not turned on for me, even tho I had set it to autostart on boot and with watchdog, so as soon as I turned that add-on on, everything just worked. 
Got no idea why autostart was off, but yeah, turning Silicon Labs Multiprotocol on fixed it for me. 
Now running Home Assistant OS 12.3 with core-2024.5.3 without issues. 
I've had it running for a couple of days just to make sure it was fixed.

tlokaj, 2024-05-17:
> That one helped me, thank you!!!

Just until next restart... now it is broken again :(

BeardElk, 2024-05-17:
> Just until next restart... now it is broken again :(

Have you checked that Silicon Labs Multiprotocol Add-on is turned on?

tlokaj, 2024-05-17:
> Have you checked that Silicon Labs Multiprotocol Add-on is turned on?

I don't have that addon installed. I've never had.

Keribi, 2024-05-18:
> Zigbee useless after 12.3 update. Downgraded to 12.2 and issue still exists.. Devices intermittently responsive, brightness levels are wrong etc... this needs remedied soon! Raspi 4, sonoff dongle, ZHA

System is stable on 2024.05.4 and 12.2.  Zigbee working fine for now.

sairon, 2024-05-20:
Anyone here with the issue, can you check if adding `dtoverlay=dwc2` to the end of `config.txt` in the boot partition helps to resolve the issues? Since the other RPi issue is apparently caused by the default RPi-specific `dwc_otg` driver, I wonder if using the upstream driver helps here too.

NodeJSmith, 2024-05-20:
I had a similar issue but with Z2M instead of ZHA. Happy to open a separate issue if that'd be better, but wanted to add that here in case its helpful. Rolling back to 12.2 with `ha os boot-slot other` and a few reboots got it working again.

capmlbpkel, 2024-05-21:
> Anyone here with the issue, can you check if adding `dtoverlay=dwc2` to the end of `config.txt` in the boot partition helps to resolve the issues?

How to can I do that? I read somewhere that the boot partition is usually /dev/vda1, but I cannot mount it:

```
/ # whoami
root
/ # mount /dev/vda1 /mnt/
mount: permission denied (are you root?)
```

capmlbpkel, 2024-05-22:
Is there anything we can do to support debugging / fixing?

sairon, 2024-05-23:
@walterbressers That's totally wrong - this thread is about Raspberry Pi issues, you're running in a VM and the change you did won't (and can't) work. AFAIK you're the only one reporting such issues on x86/VM, please create another report with full logs and issue description.

@capmlbpkel Unfortunately we still don't know if the problem is related to the other one. If you want to edit the file on the device, you need to use a root shell - the easiest way is to plug in a display and keyboard to your RPi and type `login` in the HA console. You can also edit the file on another PC - on Raspberry Pi 5 the boot partition should be visible if you take the SD card out and put it into a card reader (older RPi models have the partition hidden, so it may be more complicated to mount it).

NodeJSmith, 2024-05-23:
@sairon FWIW my issue with Z2M is on a proxmox vm on an Intel cpu, forgot to mention that in my previous comment. I can create a separate issue later if @walterbressers doesn't beat me to it

capmlbpkel, 2024-05-23:
I made the same mistake, see https://github.com/home-assistant/operating-system/issues/3390 for the separate issue.

jbouwh, 2024-06-13:
> Anyone here with the issue, can you check if adding `dtoverlay=dwc2` to the end of `config.txt` in the boot partition helps to resolve the issues? Since the other RPi issue is apparently caused by the default RPi-specific `dwc_otg` driver, I wonder if using the upstream driver helps here too.

I'm running zha with a sky connect on is 12.3 on a rpi5, an had some instability with zha. Not using the multi protocol firmware. So I have set the `dwc2` option now to see if that helps to stabilise.

walterbressers, 2024-06-16:
> @walterbressers That's totally wrong - this thread is about Raspberry Pi issues, you're running in a VM and the change you did won't (and can't) work. AFAIK you're the only one reporting such issues [...]

The issue that my Conbee II stick (which was hardware wise recognised) but failed to initialize/connect was solved with Core update 2024.6.3, probably something to do with ZHA dependancy zigpy?
https://www.home-assistant.io/blog/2024/06/05/release-20246/#202463---june-15

gnmancini, 2024-06-21:
> Anyone here with the issue, can you check if adding `dtoverlay=dwc2` to the end of `config.txt` in the boot partition helps to resolve the issues? Since the other RPi issue is apparently caused by the default RPi-specific `dwc_otg` driver, I wonder if using the upstream driver helps here too.

It worked. 
Starting with OS 12.1 using a raspberry Pi 3B
I tested the parameter with OS 12.4
I followed the step-by-step in https://github.com/home-assistant/operating-system/issues/3362 to edit the config.txt , but I was not able to find the config.txt.
I use my PC to edit the config.txt mounting the partition and editing the config.txt, I've added 'dtoverlay=dwc2' at the end under [all] section.
After that, I did the upgrade and everything was working fine.

Information about the kernel bug:
https://github.com/home-assistant/operating-system/discussions/3418

arnonsha, 2024-12-12:
Is there any progress with this issue ? 
Can anyone say if a newer OS version solves this ? 

Thanks