home-assistant/operating-system #3353: Error while setting up rpi_gpio platform for binary_sensor

Officer-on-duty, 2024-05-08:
### Describe the issue you are experiencing

Since 12.3
Logger: homeassistant.components.binary_sensor
Quelle: helpers/entity_platform.py:356
Integration: Binärsensor (Dokumentation, Probleme)
Error while setting up rpi_gpio platform for binary_sensor
[...]
RuntimeError: Failed to add edge detection

GPIO Switches works fine!

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

12.3

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. update the OS form 12.2 to 12.3
2. be sad
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
no
```

### Anything in the Host logs that might be useful for us?

```txt
no
```

### System information

Derzeit sind keine Reparaturen verfügbar

### Additional information

Backup without success since the OS is not restored.

sairon, 2024-05-09:
It's not clear what `rpi_gpio` integration you are using, because it's not a Core-bundled one, but I assume it's the `thecode/ha-rpi_gpio` which already has [an issue reported](https://github.com/thecode/ha-rpi_gpio/issues/255). Ultimately, it's a problem with `RPi.GPIO` which uses deprecated interface to configure the GPIOs, see [here](https://sourceforge.net/p/raspberry-gpio-python/tickets/210/).

Long story short, this needs to be fixed in third-party libraries, only possible workaround on the OS side is to revert to 12.2, either by `ha os boot-slot other` or `ha os upgrade --version 12.2`.

closed 2024-05-09 by sairon

Officer-on-duty, 2024-05-09:
> It's not clear what `rpi_gpio` integration you are using, because it's not a Core-bundled one, but I assume it's the `thecode/ha-rpi_gpio` which already has [an issue [...]

Very precise and helpful answer. Thank you!