home-assistant/operating-system #3356: Network storage connection fails after update to OS 12.x from 11.5

dbt63, 2024-05-09:
### Describe the issue you are experiencing

Network storage fails to connect after update from OS 11.5 to any OS version 12.
Latest attempt is with OS 12.3.  Have tried all previous os 12.x releases without success.
Reverting to os 11.5 solves the problem.
Running home assistant os on VM virtual box 7.0, on windows 11 pc.
Post update restart of homeassistant or host system does not solve the problem.

Only one network storage location is configured - a share on windows 11 PC for backups.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

6.6.29-haos

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Update via Settings / Updates, from 11.5 to 12.3.
Use CLI command  "ha os update --version 11.5" from command line to recover to working network storage connection.
Reinstating the network store manually does not work (via Settings / Storage / Network Storage).

### Anything in the Supervisor logs that might be useful for us?

```txt
2024-05-09 08:59:49.468 INFO (MainThread) [supervisor.mounts.manager] Reloading mount: HomeAssistantBackups
2024-05-09 08:59:49.511 ERROR (MainThread) [supervisor.mounts.mount] Reloading HomeAssistantBackups did not succeed. Check host logs for errors from mount or systemd unit mnt-data-supervisor-mounts-HomeAssistantBackups.mount for details.
```

### Anything in the Host logs that might be useful for us?

```txt
2024-05-09 07:59:49.484 homeassistant systemd[1]: Mounting Supervisor cifs mount: HomeAssistantBackups...
2024-05-09 07:59:49.488 homeassistant kernel: CIFS: Attempting to mount //192.168.0.200/HomeAssistantBackups
2024-05-09 07:59:49.506 homeassistant kernel: CIFS: VFS: parse_reparse_point: unhandled reparse tag: 0x9000601a
2024-05-09 07:59:49.506 homeassistant kernel: CIFS: VFS: cifs_read_super: get root inode failed
2024-05-09 07:59:49.507 homeassistant mount[6600]: mount error(95): Operation not supported
2024-05-09 07:59:49.507 homeassistant mount[6600]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log messages (dmesg)
2024-05-09 07:59:49.510 homeassistant systemd[1]: mnt-data-supervisor-mounts-HomeAssistantBackups.mount: Mount process exited, code=exited, status=32/n/a
2024-05-09 07:59:49.510 homeassistant systemd[1]: mnt-data-supervisor-mounts-HomeAssistantBackups.mount: Failed with result 'exit-code'.
2024-05-09 07:59:49.510 homeassistant systemd[1]: Failed to mount Supervisor cifs mount: HomeAssistantBackups.
2024-05-09 08:01:23.430 homeassistant systemd[1]: Mounting Supervisor cifs mount: HomeAssistantBackups...
2024-05-09 08:01:23.432 homeassistant kernel: CIFS: Attempting to mount //192.168.0.200/HomeAssistantBackups
2024-05-09 08:01:23.446 homeassistant kernel: CIFS: VFS: parse_reparse_point: unhandled reparse tag: 0x9000601a
2024-05-09 08:01:23.446 homeassistant kernel: CIFS: VFS: cifs_read_super: get root inode failed
2024-05-09 08:01:23.448 homeassistant mount[6769]: mount error(95): Operation not supported
2024-05-09 08:01:23.448 homeassistant mount[6769]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log messages (dmesg)
2024-05-09 08:01:23.450 homeassistant systemd[1]: mnt-data-supervisor-mounts-HomeAssistantBackups.mount: Mount process exited, code=exited, status=32/n/a
2024-05-09 08:01:23.450 homeassistant systemd[1]: mnt-data-supervisor-mounts-HomeAssistantBackups.mount: Failed with result 'exit-code'.
2024-05-09 08:01:23.452 homeassistant systemd[1]: Failed to mount Supervisor cifs mount: HomeAssistantBackups.
```

### System information

## System Information

version | core-2024.5.2
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.2
os_name | Linux
os_version | 6.6.29-haos
arch | x86_64
timezone | Europe/London
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4890
Installed Version | 1.34.0
Stage | running
Available Repositories | 1394
Downloaded Repositories | 9

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 21 November 2024 at 00:00
relayer_connected | true
relayer_region | eu-central-1
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | false
remote_server | eu-central-1-10.ui.nabu.casa
certificate_status | ready
instance_id | fb958b004b6f4a6f8bf49363313c5e75
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.3
-- | --
update_channel | stable
supervisor_version | supervisor-2024.05.1
agent_version | 1.6.0
docker_version | 25.0.5
disk_total | 62.3 GB
disk_used | 8.8 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Advanced SSH & Web Terminal (17.2.0), Samba share (12.3.1), Mosquitto broker (6.4.0), ESPHome (2024.4.0), Studio Code Server (5.15.0), AdGuard Home (5.0.7)

</details>

<details><summary>Dashboards</summary>

dashboards | 5
-- | --
resources | 4
views | 7
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 19 April 2024 at 09:39
-- | --
current_recorder_run | 9 May 2024 at 08:43
estimated_db_size | 597.20 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

sairon, 2024-05-09:
Very much likely a duplicate of #3208 - aren't you trying to mount a folder that is configured with OneDrive sync with files-on-demand feature enabled?

dbt63, 2024-05-09:
Yes, looks like I am...  I will try that an report back.
(Apologies if this is a duplication).

dbt63, 2024-05-09:
Moved to a non OneDrive share and now connects.  Thanks

closed 2024-05-09 by dbt63