home-assistant/operating-system #3360: Home assistant get stuck in a reboot loop after the upgrade to HA OS 12.3

danielsjf, 2024-05-10:
### Describe the issue you are experiencing

I've done the upgrade to HA OS 12.3. Usually I upgrade quite quickly and the rest of the system was up to date.
I didn't check it immediately since this update takes quite a while. A few hours later I noticed that HA was down. So I manually cycled the Raspberry Pi. After a few minutes, HA came online (port 8123). However, after a few moments I noticed it has been down again.
Is has been doing this now a few times. Any tips for logs I could check?

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

Latest (which should be 2024.05.something but I can't check due to HA not being online long enough)

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Upgrade to HA OS 12.3

### Anything in the Supervisor logs that might be useful for us?

```txt
I don't get here long enough
```

### Anything in the Host logs that might be useful for us?

```txt
I don't get here long enough
```

danielsjf, 2024-05-10:
Might be related to #3352. I have a Skyconnect for thread.

b10n, 2024-05-10:
Same issue on Raspberry Pi 3B+ (rpi3-64). Disconnecting my peripheral devices (including a CC2531 configured with Zigbee2MQTT), rebooting, and plugging them back in kept it from rebooting. I downgraded to 12.2 for the time being.

mavotronik, 2024-05-10:
> Same issue on Raspberry Pi 3B+ (rpi3-64). Disconnecting my peripheral devices (including a CC2531 configured with Zigbee2MQTT), rebooting, and plugging them back in kept it from rebooting. I downgraded to 12.2 for the time being.

Same on my Raspberry 3B+, I downgraded to 12.2 too

danielsjf, 2024-05-10:
After unplugging the USB devices and disabling thread and ZHA, I got it stable. However, some other integrations such as Homewizard (local wifi connection) also failed.

joaohess, 2024-05-10:
> Same issue on Raspberry Pi 3B+ (rpi3-64). Disconnecting my peripheral devices (including a CC2531 configured with Zigbee2MQTT), rebooting, and plugging them back in kept it from rebooting. I downgraded to 12.2 for the time being.

I’ve tried what you said and it worked, but after  rebooting (on purpose) it breaks again. Apparently the coordinator (in my case, EFR32MG21) has to be unplugged during boot.

ghostsam45, 2024-05-11:
Same issue 

[core-ssh ~]$ ha os info
board: rpi4-64
boot_slots:
   A:
      state: booted
      status: good
      version: "12.2"
   B:
      state: inactive
      status: good
      version: "12.3"
data_disk: CT240BX500SSD1-2050E4DAFBA9
update_available: true
version: "12.2"
version_latest: "12.3"

playaz44, 2024-05-11:
After update my HA is completely dead, disconnecting skyconnect does not help, I tried this command `ha os boot-slot other` from [this](https://github.com/home-assistant/operating-system/issues/3352#issuecomment-2101447704) but still nothing.

pavelrusiyan, 2024-05-13:
Same on my Raspberry 3B+, I downgraded to 12.1 And including a SONOFF_Zigbee_3.0_USB_Dongle_Plus ezsp configured with Zigbee2MQTT.

sairon, 2024-05-13:
Closing this in favor of #3362. Please see [this message](https://github.com/home-assistant/operating-system/issues/3362#issuecomment-2107327277) and report details about your setup there.

closed 2024-05-13 by sairon