home-assistant/operating-system #3362: Bootloop after 12.3 update (usb related)

WilcoSnel, 2024-05-11:
### Describe the issue you are experiencing

After upgrading to 12.3 my HA started bootlooping. Noticed that after I disconnected my Aeotec Z-Stick ZW090 it booted again and ran fine (without my Z wave network of course). 

### What operating system image do you use?

rpi3-64 (Raspberry Pi 3 64-bit OS)

### What version of Home Assistant Operating System is installed?

12.3

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Fresh install with Pi imager HA OS 12.1
2. Restore stuff from backups
3. Reboot, works
4. Update to 12.3, bootloop
5. Disconnect Z stick, reboot, works without zwave
[image]
[image]
[image]
[image]

...

### Anything in the Supervisor logs that might be useful for us?

```txt
See attached
```

### Anything in the Host logs that might be useful for us?

```txt
See attached
```

### System information

## System Information

version | core-2024.5.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.2
os_name | Linux
os_version | 6.6.28-haos-raspi
arch | aarch64
timezone | Europe/Amsterdam
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4942
Installed Version | 1.34.0
Stage | running
Available Repositories | 1394
Downloaded Repositories | 9

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.3
-- | --
update_channel | stable
supervisor_version | supervisor-2024.05.1
agent_version | 1.6.0
docker_version | 25.0.5
disk_total | 457.7 GB
disk_used | 3.8 GB
healthy | true
supported | true
board | rpi3-64
supervisor_api | ok
version_api | ok
installed_addons | AdGuard Home (5.0.7), Cloudflared (5.1.9), File editor (5.8.0), Samba NAS (12.2.0-nas2), Terminal & SSH (9.14.0), Z-Wave JS (0.5.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 2
views | 1
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | May 2, 2024 at 18:58
-- | --
current_recorder_run | May 11, 2024 at 12:22
estimated_db_size | 158.29 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

joaohess, 2024-05-11:
Same here. Probably related do #3360 and #3352.

Here, with a Raspberry Pi 3 Model B and a Zigbee Sonoff Dongle-E (EFR32MG21)

bartbrew, 2024-05-12:
Same issue with Raspberry Pi 3, 32-bit, with Zooz ZST10 usb dongle.
Downgrading back to 12.2 solved the problem.

sairon, 2024-05-13:
I tried reproducing the issue here but I might be still missing something that's triggering the issue, so I'd like to gather following information about your setups:

- Exact RPi model (for 3B please distinguish between 3B and 3B+).
- Boot medium HAOS is running from (SD card / SD card + USB data disk / USB drive as system disk).
- List of USB peripherals connected.
- Is the issue reproducible on a minimal fresh install without restoring the backup?
- Does the problem go away for all affected people here after unplugging the connected USB devices?

Also, the following logs from some of you will be very useful:

- Full `dmesg` output when the issue occurs (if it's possible to obtain it before the device reboots).
- Log from the previous boot (`ha host logs -b-1 -n1000`) when booted into working state (e.g. with USB device unplugged).

WilcoSnel, 2024-05-13:
> I tried reproducing the issue here but I might be still missing something that's triggering the issue, so I'd like to gather following information about your setups:
> 
> * Exact RPi model (for 3B please distinguish between 3B and 3B+).
> * Boot medium HAOS is running from (SD card / SD card + USB data disk / USB drive as system disk).
> * List of USB peripherals connected.
> * If you can connect your RPi to an HDMI display, do you see the same errors as in the first post in this issue?
> * Is the issue reproducible on a minimal fresh install without restoring the backup?
> * Does the problem go away for all affected people here after unplugging the connected USB devices?
> 
> Also, the following logs from some of you will be very useful:
> 
> * Full `dmesg` output when the issue occurs (if it's possible to obtain it before the device reboots).
> * Log from the previous boot (`ha host logs -b-1 -n1000`) when booted into working state (e.g. with USB device unplugged).

For me it's as follows:
- model: Pi 3B+
- Boot medium: USB SSD as system disk
- USB devices connected: USB SSD (system) , Raid box connected via USB (NAS usage), Z wave stick (model reported above)
- Yes, but it is my post so :P
- It's hard to check without backup. I'm afraid if I configure the Zstick on a fresh base, it will no longer work with my backup after.
- As reported, for me the issue does go away. Not a single error left in CLI after disconnecting the Zstick.

As far as pattern goes. It all seems to be USB protocol controllers that are having issues. For me it's a Zwave usb controller,

WilcoSnel, 2024-05-13:
Oops, closed accidentally

WilcoSnel, 2024-05-13:
Closed accidentally

joaohess, 2024-05-13:
> I tried reproducing the issue here but I might be still missing something that's triggering the issue, so I'd like to gather following information about your setups:
>
> * Exact RPi model (for 3B [...]

Here:
- Raspberry Pi 3 B
- Booting from SD Card
- Only one USB peripheral: Sonoff Zigbee Dongle-E (EFR32MG21) on a USB extension
- The problem does go away when disconnecting the dongle.
- I haven't tried a fresh install. I am back to 12.2 via `ha os boot-slot other`
- I will check the logs when I'm back home

mvdnes, 2024-05-13:
The following also crashes the system:
* Boot into system without Z-wave stick
* Attach stick
* (nothing happens yet)
* Start Zwave.js
* System crashes

playaz44, 2024-05-13:
> * Exact RPi model (for 3B please distinguish between 3B and 3B+).

i7-7700T HA OS in KVM on proxmox

> * Boot medium HAOS is running from (SD card / SD card + USB data disk / USB drive as system disk).

SSD

> * List of USB peripherals connected.

Only SkyConnect

> * If you can connect your RPi to an HDMI display, do you see the same errors as in the first post in this issue?

No

> * Is the issue reproducible on a minimal fresh install without restoring the backup?

Yes, I installed fresh HAOS on another VM and set up thread, matter, zha, bridge router, silicon labs multiprotocol, then when I tried to configure a new radio for zha, HAOS died

> * Does the problem go away for all affected people here after unplugging the connected USB devices?

No, if HAOS died after trying to configure radio then it can no longer be started

sairon, 2024-05-14:
@playaz44 This issue tracks booting problems in Raspberry Pi platform only. VM installations are running a different kernel version, please create a separate issue for that. Also, please check the host console when the issue occurs, something should be (hopefully) printed there.

sairon, 2024-05-14:
I have found https://github.com/raspberrypi/linux/issues/6100 that reported highly similar issue with 6.6 kernel a while ago, claiming 6.6.29 (which isn't in the stable branch yet) fixed the issue, so checking if it's the case for us might be worth a try as well.

Here's a custom dev build with the latest RPi Linux from the master branch: https://github.com/home-assistant/operating-system/actions/runs/9077372681

If there's anyone who could give it a try (i.e. install it on an SD card and follow the same steps that lead to the boot loop with fresh 12.3), it would be much appreciated :pray: Please note you need to be logged in on Github in order to download the artifacts, and that all the images are wrapped in an extra .zip archive, so you must first unpack that before you follow the standard flashing procedure.

mvdnes, 2024-05-14:
> I have found [raspberrypi/linux#6100](https://github.com/raspberrypi/linux/issues/6100) that reported highly similar issue with 6.6 kernel a while ago, claiming 6.6.29 (which isn't in the stable branch yet) fixed the issue, so checking if it's the case for us might be worth a try as well.
> 
> Here's a custom dev build with the latest RPi Linux from the master branch: ~https://github.com/home-assistant/operating-system/actions/runs/9077372681~ (EDIT: the build was missing 64bit variants, here's a correct one: https://github.com/home-assistant/operating-system/actions/runs/9083532010)
> 
> If there's anyone who could give it a try (i.e. install it on an SD card and follow the same steps that lead to the boot loop with fresh 12.3), it would be much appreciated 🙏 Please note you need to be logged in on Github in order to download the artifacts, and that all the images are wrapped in an extra .zip archive, so you must first unpack that before you follow the standard flashing procedure.

The 32-bit image resulted in the same crash unfortunately.

mvdnes, 2024-05-14:
> I have found [raspberrypi/linux#6100](https://github.com/raspberrypi/linux/issues/6100) that reported highly similar issue with 6.6 kernel a while ago, claiming 6.6.29 (which isn't in the stable [...]

I tried the rpi3-64 image, and got the same result.

<details>
<summary>dmesg output</summary>

```
# dmesg -w
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 6.6.30-haos-raspi (builder@79b23bbdaf26) (aarch64-buildroot-linux-gnu-gcc.br_real (Buildroot -g0a200689) 12.3.0, GNU ld (GNU Binutils) 2.40) #1 SMP PREEMPT Tue May 14 19:31:13 UTC 2024
[    0.000000] KASLR enabled
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000037400000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x0000000037400000..0x000000003b3fffff (65536 KiB) map reusable linux,cma
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] On node 0, zone DMA: 19456 pages in unavailable ranges
[    0.000000] percpu: Embedded 29 pages/cpu s80552 r8192 d30040 u118784
[    0.000000] pcpu-alloc: s80552 r8192 d30040 u118784 alloc=29*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: ARM erratum 843419
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: zram.enabled=1 zram.num_devices=3 rootwait systemd.machine_id= cgroup_enable=memory fsck.repair=yes systemd.condition-first-boot=true coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=1824 bcm2708_fb.fbheight=984 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=tty1 usb-storage.quirks=174c:55aa:u,2109:0715:u,152d:0578:u,152d:0579:u,152d:1561:u,174c:0829:u,14b0:0206:u,174c:225c:u,7825:a2a4:u,152d:0562:u,125f:a88a:u root=PARTUUID=8d3d53e3-6d49-4c38-8349-aff6859e82fd rootfstype=squashfs ro rauc.slot=A
[    0.000000] cgroup: Enabling memory control group subsystem
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 238896
[    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
[    0.000000] Memory: 859040K/970752K available (13504K kernel code, 2116K rwdata, 4196K rodata, 4608K init, 1031K bss, 46176K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 42802 entries in 168 pages
[    0.000000] ftrace: allocated 168 pages with 3 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Rude variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 100 jiffies.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: bcm2836_arm_irqchip_handle_irq
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000001] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000392] Console: colour dummy device 80x25
[    0.000409] printk: console [tty1] enabled
[    0.001408] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=19200)
[    0.001455] pid_max: default: 32768 minimum: 301
[    0.001573] LSM: initializing lsm=capability,apparmor,integrity
[    0.001761] AppArmor: AppArmor initialized
[    0.002003] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.002061] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.004943] RCU Tasks: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1.
[    0.005130] RCU Tasks Rude: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1.
[    0.005316] RCU Tasks Trace: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1.
[... 1212 lines cut ...]

[  849.095866] WARN::dwc_otg_hcd_urb_dequeue:638: Timed out waiting for FSM NP transfer to complete on 0

[  849.096306] WARN::dwc_otg_hcd_urb_dequeue:638: Timed out waiting for FSM NP transfer to complete on 2

[... cut ...]
[  849.099058] usb 1-1.5: USB disconnect, device number 4
[  849.287079] usb 1-1.5: new full-speed USB device number 6 using dwc_otg
[  849.378436] usb 1-1.5: New USB device found, idVendor=0658, idProduct=0200, bcdDevice= 0.00
[  849.378488] usb 1-1.5: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[  849.385270] cdc_acm 1-1.5:1.0: ttyACM0: USB ACM device
[  850.978049] Bluetooth: hci0: command 0x200c tx timeout
[  850.978049] Bluetooth: hci0: Opcode 0x200c failed: -110
```
</details>

sairon, 2024-05-15:
@mvdnes Thank you very much. Can you try adding `dwc_otg.fiq_enable=1 dwc_otg.fiq_fsm_enable=0` anywhere to `cmdline.txt` (it just must still be a single line)? Eventually also change the first one to `dwc_otg.fiq_enable=0`.

mvdnes, 2024-05-15:
> @mvdnes Thank you very much. Can you try adding `dwc_otg.fiq_enable=1 dwc_otg.fiq_fsm_enable=0` anywhere to `cmdline.txt` (it just must still be a single line)? Eventually also change the first one to `dwc_otg.fiq_enable=0`.

Tested with the image from above, by only changing the cmdline parameters. Unfortunately, it still crashes for both of the options.

sairon, 2024-05-15:
@mvdnes Thank you, I finally managed to reproduce it reliably here too - starting Z-Wave add-on with Z-Wave.me UZB stick seems to crash it every time. It is also reproducible on RPi OS with the `zwave-js-ui` Docker image, so that confirms it is not a HAOS-specific issue. The only thing that doesn't fit is that in my case disabling FIQ (i.e. setting `dwc_otg.fiq_enable=0 dwc_otg.fiq_fsm_enable=0`) makes it stable.

I will report this issue to the RPi kernel repo, but I want to make sure we're not chasing ghosts and the cause is the same. Can you retry (hopefully for the last time) and make sure the parameters are properly set? In `dmesg | grep FIQ` you should see:
```
[    2.771829] dwc_otg: FIQ disabled
[    2.771847] dwc_otg: FIQ split-transaction FSM disabled
```
Just disabling the FSM is not sufficient.

mvdnes, 2024-05-16:
I have added the `dwc_otg.fiq_enable=0 dwc_otg.fiq_fsm_enable=0` to a fresh install of kernel 6.6.30 above, and got a crash right after the setup of Zwave.js. I did capture the dmesg this time, and indeed the FIQ lines are present:

<details>
<summary>dmesg log of dwc_otg.fiq...=0</summary>

```
# dmesg -w
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 6.6.30-haos-raspi (builder@79b23bbdaf26) (aarch64-buildroot-linux-gnu-gcc.br_real (Buildroot -g0a200689) 12.3.0, GNU ld (GNU Binutils) 2.40) #1 SMP PREEMPT Tue May 14 19:31:13 UTC 2024
[    0.000000] KASLR enabled
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000037400000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x0000000037400000..0x000000003b3fffff (65536 KiB) map reusable linux,cma
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] On node 0, zone DMA: 19456 pages in unavailable ranges
[    0.000000] percpu: Embedded 29 pages/cpu s80552 r8192 d30040 u118784
[    0.000000] pcpu-alloc: s80552 r8192 d30040 u118784 alloc=29*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: ARM erratum 843419
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: zram.enabled=1 zram.num_devices=3 rootwait systemd.machine_id= cgroup_enable=memory fsck.repair=yes systemd.condition-first-boot=true coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=1824 bcm2708_fb.fbheight=984 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=tty1 usb-storage.quirks=174c:55aa:u,2109:0715:u,152d:0578:u,152d:0579:u,152d:1561:u,174c:0829:u,14b0:0206:u,174c:225c:u,7825:a2a4:u,152d:0562:u,125f:a88a:u dwc_otg.fiq_enable=0 dwc_otg.fiq_fsm_enable=0 root=PARTUUID=8d3d53e3-6d49-4c38-8349-aff6859e82fd rootfstype=squashfs ro rauc.slot=A
[    0.000000] cgroup: Enabling memory control group subsystem
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 238896
[    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
[    0.000000] Memory: 859040K/970752K available (13504K kernel code, 2116K rwdata, 4196K rodata, 4608K init, 1031K bss, 46176K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 42802 entries in 168 pages
[    0.000000] ftrace: allocated 168 pages with 3 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Rude variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 100 jiffies.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: bcm2836_arm_irqchip_handle_irq
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000001] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000408] Console: colour dummy device 80x25
[    0.000427] printk: console [tty1] enabled
[    0.001449] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=19200)
[    0.001496] pid_max: default: 32768 minimum: 301
[    0.001611] LSM: initializing lsm=capability,apparmor,integrity
[    0.001786] AppArmor: AppArmor initialized
[    0.002011] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.002065] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.004911] RCU Tasks: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1.
[    0.005096] RCU Tasks Rude: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1.
[    0.005283] RCU Tasks Trace: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1.
[    0.005645] rcu: Hierarchical SRCU implementation.
[    0.005672] rcu: 	Max phase no-delay instances is 400.
[... 1120 lines cut ...]
[   97.588731] audit: type=1334 audit(1715844293.085:199): prog-id=53 op=LOAD
[... cut ...]
[  127.593083] audit: type=1334 audit(1715844323.089:203): prog-id=49 op=UNLOAD
[  128.636121] audit: type=1334 audit(1715844324.132:204): prog-id=54 op=UNLOAD
[  128.636163] audit: type=1334 audit(1715844324.132:205): prog-id=53 op=UNLOAD
[  128.636179] audit: type=1334 audit(1715844324.132:206): prog-id=52 op=UNLOAD
[  133.453582] Bluetooth: hci0: Frame reassembly failed (-84)
[  133.453804] Bluetooth: hci0: Frame reassembly failed (-84)
[  136.775371] audit: type=1334 audit(1715844332.271:207): prog-id=55 op=LOAD
```
</details>

unhandyman, 2024-05-23:
I can reproduce a system freeze incl. reboot as mentioned in https://github.com/raspberrypi/linux/issues/6172. This is a fresh install.

Inserting a ConBee II into a USB slot, autodiscovery works. The system freezes right when I try to configure the ZHA integration via GUI and can't be reached via network. Comes back up after. Just did it.

<details><summary>logs before configuring ZHA</summary>
<details><summary>Core</summary>

```
2024-05-23 20:26:54.050 WARNING (Recorder) [homeassistant.components.recorder.util] The system could not validate that the sqlite3 database at //config/home-assistant_v2.db was shutdown cleanly
2024-05-23 20:26:54.172 WARNING (Recorder) [homeassistant.components.recorder.util] Ended unfinished session (id=10 from 2024-05-23 16:30:42.208624)
2024-05-23 20:27:13.287 WARNING (MainThread) [homeassistant.components.mqtt] The 'schema' option is deprecated, please remove it from your configuration
2024-05-23 20:44:01.797 ERROR (MainThread) [homeassistant.components.analytics] Error sending analytics to https://analytics-api.home-assistant.io/v1: ClientConnectorError(ConnectionKey(host='analytics-api.home-assistant.io', port=443, is_ssl=True, ssl=True, proxy=None, proxy_auth=None, proxy_headers_hash=4649981112449888728), ConnectionRefusedError(111, "Connect call failed ('::', 443, 0, 0)"))
```
</details>
<details><summary>Host</summary>

```
2024-05-23 18:26:23.134 homeassistant kernel: audit: type=1300 audit(1716488783.132:204): arch=c00000b7 syscall=211 success=yes exit=508 a0=3 a1=7fcae0cb78 a2=0 a3=1 items=0 ppid=521 pid=2149 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="iptables" exe="/usr/sbin/xtables-nft-multi" subj=unconfined key=(null)
2024-05-23 18:26:23.134 homeassistant kernel: audit: type=1327 audit(1716488783.132:204): proctitle=2F7573722F7362696E2F69707461626C6573002D2D77616974002D74006E6174002D4100444F434B4552002D7000746370002D6400302F30002D2D64706F72740038383834002D6A00444E4154002D2D746F2D64657374696E6174696F6E003137322E33302E33332E303A383838340000002D690068617373696F
2024-05-23 18:26:23.165 homeassistant kernel: audit: type=1325 audit(1716488783.163:205): table=filter:103 family=2 entries=1 op=nft_register_rule pid=2153 subj=unconfined comm="iptables"
2024-05-23 18:26:23.165 homeassistant kernel: audit: type=1300 audit(1716488783.163:205): arch=c00000b7 syscall=211 success=yes exit=648 a0=3 a1=7fd11cb2e8 a2=0 a3=1 items=0 ppid=521 pid=2153 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="iptables" exe="/usr/sbin/xtables-nft-multi" subj=unconfined key=(null)
2024-05-23 18:26:23.166 homeassistant kernel: audit: type=1327 audit(1716488783.163:205): proctitle=2F7573722F7362696E2F69707461626C6573002D2D77616974002D740066696C746572002D4100444F434B45520000002D690068617373696F002D6F0068617373696F002D7000746370002D64003137322E33302E33332E30002D2D64706F72740038383834002D6A00414343455054
2024-05-23 18:26:23.184 homeassistant kernel: audit: type=1325 audit(1716488783.183:206): table=nat:104 family=2 entries=1 op=nft_register_rule pid=2156 subj=unconfined comm="iptables"
2024-05-23 18:26:23.524 homeassistant systemd[1]: Started libcontainer container 8e8b797e54f7737dddb144f7bc78512bf908849900da4ed85ddb87308225f1d6.
2024-05-23 18:26:23.843 homeassistant kernel: eth0: renamed from veth0468713
2024-05-23 18:26:23.855 homeassistant kernel: hassio: port 6(vethf372872) entered blocking state
2024-05-23 18:26:23.855 homeassistant kernel: hassio: port 6(vethf372872) entered forwarding state
2024-05-23 18:26:23.860 homeassistant NetworkManager[419]: <info>  [1716488783.8597] device (vethf372872): carrier: link connected
2024-05-23 18:26:24.056 homeassistant systemd[1]: NetworkManager-dispatcher.service: Deactivated successfully.
2024-05-23 18:26:29.453 homeassistant systemd[1]: var-lib-docker-overlay2-170913d230c1755130466c5f9fc9a71733f5908bc705a04958bc89a3122feac2\x2dinit-merged.mount: Deactivated successfully.
2024-05-23 18:26:29.455 homeassistant systemd[1]: mnt-data-docker-overlay2-170913d230c1755130466c5f9fc9a71733f5908bc705a04958bc89a3122feac2\x2dinit-merged.mount: Deactivated successfully.
2024-05-23 18:26:29.791 homeassistant systemd[1]: Started libcontainer container 1de8e155cff7810a7b3926ef3ec776c47024b0386444c97be08faf3862e110f5.
2024-05-23 18:26:29.869 homeassistant kernel: kauditd_printk_skb: 69 callbacks suppressed
2024-05-23 18:26:29.869 homeassistant kernel: audit: type=1334 audit(1716488789.868:230): prog-id=48 op=LOAD
2024-05-23 18:26:29.872 homeassistant kernel: audit: type=1334 audit(1716488789.871:231): prog-id=49 op=LOAD
2024-05-23 18:26:29.872 homeassistant kernel: audit: type=1300 audit(1716488789.871:231): arch=c00000b7 syscall=280 success=yes exit=16 a0=5 a1=4000195840 a2=78 a3=0 items=0 ppid=2396 pid=2406 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="runc" exe="/usr/bin/runc" subj=unconfined key=(null)
2024-05-23 18:26:29.873 homeassistant kernel: audit: type=1327 audit(1716488789.871:231): proctitle=72756E63002D2D726F6F74002F7661722F72756E2F646F636B65722F72756E74696D652D72756E632F6D6F6279002D2D6C6F67002F72756E2F636F6E7461696E6572642F696F2E636F6E7461696E6572642E72756E74696D652E76322E7461736B2F6D6F62792F31646538653135356366663738313061376233393236656633
2024-05-23 18:26:29.873 homeassistant kernel: audit: type=1334 audit(1716488789.871:232): prog-id=50 op=LOAD
[... cut ...]
[... 214 lines cut ...]
version_api | ok
installed_addons | Mosquitto broker (6.4.0), Advanced SSH & Web Terminal (17.3.0), File editor (5.8.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 0
views | 0
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 22. Mai 2024 um 18:27
-- | --
current_recorder_run | 23. Mai 2024 um 20:26
estimated_db_size | 25.31 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

mfarich, 2024-05-23:
> I tried reproducing the issue here but I might be still missing something that's triggering the issue, so I'd like to gather following information about your setups:
>
> * Exact RPi model (for 3B [...]

rpi 3 v1.2
sc card for boot
in addition to zooz zwave stick also have nortec zigbee/zwave combo stick (only using zigbee from it).
boots when unplugging the zooz zwave stick on 12.3.

[image]

full system info is on the linked issue I created.

That-one-Joe, 2024-05-23:
Similar issue (imo same, just wrongly described as bootloop) here with Raspberry Pi 3B+ and Conbee II USB Zigbee dongle.

Was working up until 12.2. with Zigbee integration.

Updating to 12.3 causes:

- System booting normally and being pingable
- Stops responding (ping, webgui) as soon as Zigbee Integration starts up
- Responds again after unplugging Conbee II Stick

Downgrading to 12.2 fixes it.

sairon, 2024-05-24:
@That-one-Joe Interesting, that is slightly different from what I see with the Z-Wave stick - there the system cannot recover when I just remove the USB stick (unless watchdog kicks in and reboots the device - are you sure it's not the case? You should be able to figure it out from the system uptime). Also I wonder if this issue is also resolved by adding `dtoverlay=dtc2` to config.txt as suggested above - if not, then we're likely facing multiple issues.

WilcoSnel, 2024-05-24:
-

WilcoSnel, 2024-05-24:
I can't comment without closing the issue for some reason. So re-opened again.

jocamane, 2024-06-18:
> @mvdnes Thank you, I finally managed to reproduce it reliably here too - starting Z-Wave add-on with Z-Wave.me UZB stick seems to crash it every time. I am not able to reproduce it with any of the Zigbee sticks I have here, even if I set up a simple testing network. It is also reproducible on RPi OS with the `zwave-js-ui` Docker image, so that confirms it is not a HAOS-specific issue. The only thing that doesn't fit is that in my case disabling FIQ (i.e. setting `dwc_otg.fiq_enable=0 dwc_otg.fiq_fsm_enable=0`) makes it stable.
> 
> I will report this issue to the RPi kernel repo, but I want to make sure we're not chasing ghosts and the cause is the same. Can you retry (hopefully for the last time) and make sure the parameters are properly set? In `dmesg | grep FIQ` you should see:
> 
> ```
> [    2.771829] dwc_otg: FIQ disabled
> [    2.771847] dwc_otg: FIQ split-transaction FSM disabled
> ```
> 
> Just disabling the FSM is not sufficient. Alternatively, adding `dtoverlay=dwc2` to the end of `config.txt`, thus using a different driver for USB, also fixes it for me.

Hi,
and how do I put this line ( `dtoverlay=dwc2`) in config.txt? Where is this file and how do I access it?
Thanks

MattWestb, 2024-06-18:
Shut down the PI from HA and then done unplug the power cable to it.
Take out the SD-Card and edit the file on the boot partition and save it and unmount the card.
Put back the card and power cable and wait for HA to start

WilcoSnel, 2024-06-19:
Updated all the way to:
Core
2024.6.3
Supervisor
2024.06.0
Operating System
12.4
Frontend
20240610.1
Z-Wave JS
0.6.0

Then added dtoverlay=dwc2 to config.txt

Can confirm this works with my Zwave stick. I did have to start Zwave JS, enable watchdog and start on boot. 2 full reboots of the Pi 3b+ after that, but then started working properly. Is stable for about 12 hours now.

Cenedd, 2024-06-19:
Not sure if this is a related issue or flaky hardware.  I'm running HA-OS on a Pi5 with an SSD on an NVME Base and a PoE hat.  I've got a USB2 extension running to a cheap USB2 3-port hub and that has dongles for Bluetooth, Zigbee and RFXCom.  I've been having trouble not with boot loops but with all the devices on that USB connection going offline and showing initialisation failed, retrying.  Sometimes an unplug and re-plug fixes it but mostly it requires a reboot.  It could just be that the cheap USB2 hub was too cheap but I'm trying the dtoverlay option (I SSH'd to 22222 and then did "vi /mnt/boot/config.txt") to see whether stability improves.  It previously ran for hours or a few days before failure so I'm going to be waiting for something to NOT happen but I'll come back and update either way.

StaffordBully, 2024-06-19:
> Shut down the PI from HA and then done unplug the power cable to it.
> Take out the SD-Card and edit the file on the boot partition and save it and unmount the card.
> Put back the card and power cable and wait for HA to start

I have the usb drive connected to my pc, but i don't have the config.txt file.

So what can I do?

Cenedd, 2024-06-19:
> I have the usb drive connected to my pc, but i don't have the config.txt file.
>
> So what can I do?

If you're definitely using HomeAssistant OS on a Raspberry Pi booting from a micro SD card there should be a small FAT partition that is recognised by Windows.  You will probably need a USB reader for MicroSD cards to slot the card into. 
On the SD card you should find a text file config.txt.  That is the file you need to edit.
Windows' Notepad has a tendancy to do odd things to linux files so you may be safer to use [Notepad++](https://notepad-plus-plus.org)

StaffordBully, 2024-06-19:
> If you're definitely using HomeAssistant OS on a Raspberry Pi booting from a micro SD card there should be a small FAT partition that is recognised by Windows. You will probably need a USB reader [...]

i found that partition, but no config.txt
These files i have
[image]

Cenedd, 2024-06-19:
It's possible that your Pi isn't booting off that card and could instead be booting from something like an M.2 SSD or a USB memory stick.  You might want to run that z.exe through [virustotal](https://www.virustotal.com) as it could be malware.
You might want to move this to a new post on the [community forum](https://community.home-assistant.io) rather than us cluttering up a github issue.  Mention me as @Cenedd and it should notify me of the post.

dogtreatfairy, 2024-06-20:
A quick how-to so you don't have to remove the SD Card to edit config.txt. 

1. SSH In, or open a Terminal window through Home Assistant. You may need to use the Advanced SSH & Terminal addon, but I don't really know because I haven't tried with the other one. 

2. Run the following commands:
`docker exec -it $(docker ps -f name=homeassistant -q) bash`
`cd /mnt`
`mkdir boot`
`fdisk -l`

3. Take the "Disk" at the top and add "p1" to the end of it for the next command. 
`sudo mount /dev/mmcblk0p1 /mnt/boot`
`cd /mnt/boot`
`sudo vi config.txt`

4. Press the `i` key and find a blank line somewhere in the file before [all]. You can literally put this anywhere. 

5. Add the following line:  `dtoverlay=dwc2`

6. Hit `ESC`

7. Type `:wq`
8. Close the shell, and you're done. Now you can update Home Assistant and it should start up using the dwc2 driver.

Caligo82, 2024-06-20:
I need to foolproofly fix this without physical access to my pi4. Attaching to the container and creating new boot directories doesn't really sound foolproof to me. Is it realistic a further update with HAOS could resolve this?

Cenedd, 2024-06-20:
> Not sure if this is a related issue or flaky hardware. I'm running HA-OS on a Pi5 with an SSD on an NVME Base and a PoE hat. I've got a USB2 extension running to a cheap USB2 3-port hub and that has [...]

Just an FYI, my USB fell apart again after adding this option and required a reboot.  Still not sure quite why but it doesn't look like it's related to this issue at least.  I'll have to start digging through logs and open a different thread.  If anything does suggest it's related, I'll come back :D   Cheers all.
Screenshot of one device just for interest - but it's all the USB devices that go.

[image]

WilcoSnel, 2024-06-20:
> > Not sure if this is a related issue or flaky hardware. I'm running HA-OS on a Pi5 with an SSD on an NVME Base and a PoE hat. I've got a USB2 extension running to a cheap USB2 3-port hub and that has dongles for Bluetooth, Zigbee and RFXCom. I've been having trouble not with boot loops but with all the devices on that USB connection going offline and showing initialisation failed, retrying. Sometimes an unplug and re-plug fixes it but mostly it requires a reboot. It could just be that the cheap USB2 hub was too cheap but I'm trying the dtoverlay option (I SSH'd to 22222 and then did "vi /mnt/boot/config.txt") to see whether stability improves. It previously ran for hours or a few days before failure so I'm going to be waiting for something to NOT happen but I'll come back and update either way.
> 
> Just an FYI, my USB fell apart again after adding this option and required a reboot. Still not sure quite why but it doesn't look like it's related to this issue at least. I'll have to start digging through logs and open a different thread. If anything does suggest it's related, I'll come back :D Cheers all. Screenshot of one device just for interest - but it's all the USB devices that go.
> 
> ![Screenshot_20240620_071006_Home Assistant~2](https://private-user-images.githubusercontent.com/117521230/341431734-d032d25c-c1c9-42cd-9878-282c25b47701.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTg4OTMxMDUsIm5iZiI6MTcxODg5MjgwNSwicGF0aCI6Ii8xMTc1MjEyMzAvMzQxNDMxNzM0LWQwMzJkMjVjLWMxYzktNDJjZC05ODc4LTI4MmMyNWI0NzcwMS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjIwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYyMFQxNDEzMjVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00MmUzZmNlZjMxYWQ1YTEzOTFkYTg1M2U5NjZkZDVjNjY4ODgwM2M3MDQ4YjAwMzQxYzIzOWI0YWZlYjZiNmY3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.nGch8Yq6S6ssUJfJm2wGTJbZReFH1pVddQKago_IzOA)

It's my understanding this is a pi3(b)(+) specific kernel issue. So yes, your issue is different.

sairon, 2024-06-20:
@Cenedd Exactly, WilcoSnel is right, your issue seems to be a different one. I also think that some people took these steps unnecessarily, so the release notes have been adjusted to clarify what's the issue and how to work it around. In your case it's not clear what's the root cause, so complete Host logs would be helpful in this case - please open a different issue for that.

sairon, 2024-06-20:
@Caligo82 If you're already running HAOS 12.3 or 12.4 and your Pi **doesn't lock up and/or reboot** with the USB devices connected, you don't need to do anything. And if you're seeing any stability issues, same as above applies to you.

Caligo82, 2024-06-20:
> @Caligo82 If you're already running HAOS 12.3 or 12.4 and your Pi **doesn't lock up and/or reboot** with the USB devices connected, you don't need to do anything. And if you're seeing any stability issues, same as above applies to you.

Yes I'm currently running HAOS 12.3 on a Raspi4 with a Conbee II, and no Lockup yet. I thought this was introduced with 12.4. Earliest I will get physical access to this PI is in two weeks. So I'm hesitating to take the plunge here....

Leasider, 2024-06-21:
Same for me- HA crashed after that recent update. HA running on Pi4 with Sonoff Zibee 3.0 Hub

Leasider, 2024-06-21:
Just tried again with the SonOff Zigbee hub unplugged- HA boots happily but how can I get the Zibee stuff working again??

gnmancini, 2024-06-21:
> A quick how-to so you don't have to remove the SD Card to edit config.txt.
> 
> 1. SSH In, or open a Terminal window through Home Assistant. You may need to use the Advanced SSH & Terminal addon, but I don't really know because I haven't tried with the other one.
> 2. Run the following commands:
>    `docker exec -it $(docker ps -f name=homeassistant -q) bash`
>    `cd /mnt`
>    `mkdir boot`
>    `fdisk -l`
> 3. Take the "Disk" at the top and add "p1" to the end of it for the next command. My disk was "/dev/mmcblk0" so in my mount command it became /dev/mmcblk0p1 to mount the first partition, which is the boot partition.
>    `sudo mount /dev/mmcblk0p1 /mnt/boot`
>    `cd /mnt/boot`
>    `sudo vi config.txt`
> 4. Press the `i` key and find a blank line somewhere in the file before [all]. You can literally put this anywhere.
> 5. Add the following line:  `dtoverlay=dwc2`
> 6. Hit `ESC`
> 7. Type `:wq`
> 8. Close the shell, and you're done. Now you can update Home Assistant and it should start up using the dwc2 driver.

This worked for me, but, instead to use ssh I've mounted the partition in my linux desktop. 
Thanks for the info.

continuousmining, 2024-06-23:
Im not sure this is related, I have HAOS on virtualbox and with zwave stick my VM freezes often and without its fine, on 12.4

TheZackCodec, 2024-06-26:
> This worked for me, but, instead to use ssh I've mounted the partition in my linux desktop. Thanks for the info.

Also mounted via Linux and the fix worked!

Pico1965, 2024-06-29:
> 2\. docker exec -it $(docker ps -f name=homeassistant -q) bash

[image]

What's wrong?
RPi3

Bumblebeee19, 2024-07-05:
You just made my day! I've been trying to figure out why my system was crashing for a week.

Before updates, the system would crash approximately every 2 to 3 days, then become unavailable for 24 to 48 hours.

After the last updates, the system was completely stuck with the following configuration:

- RPI3
- Core - 2024.7.0
- Supervisor - 2024.06.2
- Operating System - 12.4
- Zigbee key - ConBee II

By unplugging the Zigbee key, I was able to access the system and disable the watchdog and autostart of Z2M, which allowed me to have access to HA. But of course no more automation including Z2M.

Thanks thanks thanks !

@Pico1965, I had the same problem here because HA is not installed via Docker, so the command is not available. I was able to access the config.txt file by plugging the SSD (for my part) into another PC then mounting the partition via Gparted (under Linux) because the partition was not mounted automatically. The [tip](https://github.com/home-assistant/operating-system/issues/3362#issuecomment-2178387864) that helped me earlier in the discussion was to find the "small FAT partition".

I hope that from now on the system will be more stable and that I will no longer have crashes after a few days!

kdickerson, 2024-07-06:
The `dtoverlay=dwc2` workaround appears to be working for me as well (RPi 3, ConBee II).

irobot73, 2024-07-06:
Seems to be working here as well (RPi 3b):

1. Updated the CONFIG.TXT, via my Linux desktop, to add the `dtoverlay=dwc2` tweak
2. Unplugged the Zooz USB Z-Wave stick
3. Started-up & updated the unit to latest/greatest on everything
4. Shut down & re-inserted USB Z-Wave
5. Started-up again, but had to add the PORT in the Z-Wave JS configuration screen
6. Rebooted & let it sit to settle out the network
7. Profit?

bcutter, 2024-07-18:
Completely irrational. Once again I could punsh me in my own face breaking the "never change a running system" rule. Everything was running perfectly for more than 2 months untouched.

Now I first updated HA OS 12.2 to 12.4 and checked twice by booting over once again after the update making sure I did not run into this issue. Everything smooth so far.

After maybe 1 to 1,5 hours I updated HA Core to a newer version. After some more time the system became very slow, unresponsive, integrations' entities rendered unavailable etc. - while the load at the same time stays rather calm (mostly load-1 < 2).

**To make this clear: HA OS boots just fine. It's just that HA Core runs like sh*** now close to unusable. Is that what you mean with "system lockups" mentioned in the [12.4 release notes](https://github.com/home-assistant/operating-system/releases/tag/12.4)?**

I'll now try to see if the `dtoverlay=dwc2` tweak makes a difference here.

> - Boot medium HAOS is running from (SD card / SD card + USB data disk / USB drive as system disk).
> - List of USB peripherals connected.
> - If you can connect your RPi to an HDMI display, do you see [...]

- Raspberry Pi 4 (64-bit)
- HAOS from SD card
- Conbee II, Rademacher duoFERN
- No screen access
- Haven't been able to do fresh installation, hell no hopefully I won't need to walk that path

**Based on my described symptoms, do those match this issue?**

G-R-O, 2024-07-19:
My system HA Yellow started crashing with latest OS without any log entries. Just on startup one message that DB wasn't shutdown properly. In the end I updated an old HACS integration with the new internal solution (Husqvarna) end since then haven't seen any crashes anymore.

pboschi, 2024-07-19:
I had the same issue on rpi3b+; but I found a workround: disable Z2M loading at startup; instead load ZHA, wait for initialization, then disable ZHA and load Z2M. This was very annoying, so I upgraded to an Odroid M1S.

bcutter, 2024-07-21:
After I sorted out the initial strange behaviour after the HA OS 12.4 update (which was a coincidence and as stated in https://github.com/home-assistant/operating-system/issues/3362#issuecomment-2237475101 very likely not related to this issue here at all) today after 3 days of uptime HA OS unexpectedly got restarted.

I tried to get the logs using `ha host logs -b-1 -n1000` but the very last lines are not helpful at all, are they?

<details><summary>HA OS 1.24 host logs from before reboot (only last 66 lines)</summary>

```
2024-07-21 14:25:12.679 Hostname systemd[1]: run-docker-runtime\x2drunc-moby-b52245d68d2779e957774ba06a838efd3f258fe4a203ac37dbdec185c32b9d4c-runc.tdOTP7.mount: Deactivated successfully.
2024-07-21 14:25:16.255 Hostname systemd[1]: run-docker-runtime\x2drunc-moby-efe7b48d9f0d7e699227ee88e8321daaa0009fed8b78815a7dcc18d42b2d9261-runc.ShLBga.mount: Deactivated successfully.
2024-07-21 14:25:46.800 Hostname systemd[1]: run-docker-runtime\x2drunc-moby-efe7b48d9f0d7e699227ee88e8321daaa0009fed8b78815a7dcc18d42b2d9261-runc.tL5OOD.mount: Deactivated successfully.
2024-07-21 14:26:22.415 Hostname systemd[1]: run-docker-runtime\x2drunc-moby-efe7b48d9f0d7e699227ee88e8321daaa0009fed8b78815a7dcc18d42b2d9261-runc.OEVfqY.mount: Deactivated successfully.
2024-07-21 14:27:40.311 Hostname systemd[1]: run-docker-runtime\x2drunc-moby-b52245d68d2779e957774ba06a838efd3f258fe4a203ac37dbdec185c32b9d4c-runc.TgVwkA.mount: Deactivated successfully.
2024-07-21 14:34:17.189 Hostname systemd[1]: run-docker-runtime\x2drunc-moby-b52245d68d2779e957774ba06a838efd3f258fe4a203ac37dbdec185c32b9d4c-runc.yFjkyv.mount: Deactivated successfully.
2024-07-21 14:36:12.927 Hostname systemd[1]: run-docker-runtime\x2drunc-moby-efe7b48d9f0d7e699227ee88e8321daaa0009fed8b78815a7dcc18d42b2d9261-runc.qLKyBW.mount: Deactivated successfully.
2024-07-21 14:37:23.471 Hostname systemd[1]: run-docker-runtime\x2drunc-moby-6d29e84717bc0cca6e544f9b25341bec14164c568063c952939c4664280cf349-runc.CBMRRD.mount: Deactivated successfully.
2024-07-21 14:37:56.044 Hostname systemd[1]: run-docker-runtime\x2drunc-moby-b52245d68d2779e957774ba06a838efd3f258fe4a203ac37dbdec185c32b9d4c-runc.d2DR39.mount: Deactivated successfully.
2024-07-21 14:38:31.649 Hostname systemd[1]: run-docker-runtime\x2drunc-moby-b52245d68d2779e957774ba06a838efd3f258fe4a203ac37dbdec185c32b9d4c-runc.zDcjHc.mount: Deactivated successfully.
2024-07-21 14:42:09.366 Hostname systemd[1]: run-docker-runtime\x2drunc-moby-efe7b48d9f0d7e699227ee88e8321daaa0009fed8b78815a7dcc18d42b2d9261-runc.5TNmVq.mount: Deactivated successfully.
2024-07-21 14:42:14.477 Hostname systemd[1]: run-docker-runtime\x2drunc-moby-efe7b48d9f0d7e699227ee88e8321daaa0009fed8b78815a7dcc18d42b2d9261-runc.S6DwdL.mount: Deactivated successfully.
2024-07-21 14:43:11.575 Hostname systemd[1]: run-docker-runtime\x2drunc-moby-b52245d68d2779e957774ba06a838efd3f258fe4a203ac37dbdec185c32b9d4c-runc.NY8vU0.mount: Deactivated successfully.
2024-07-21 14:43:46.104 Hostname systemd[1]: run-docker-runtime\x2drunc-moby-078736a8158d379b4ef46ff61088d97ed1ada8f2b7d94f153bca42b6e8e4a99f-runc.rgPD0N.mount: Deactivated successfully.
2024-07-21 14:44:17.766 Hostname systemd[1]: run-docker-runtime\x2drunc-moby-b52245d68d2779e957774ba06a838efd3f258fe4a203ac37dbdec185c32b9d4c-runc.vxlyav.mount: Deactivated successfully.
2024-07-21 14:48:00.743 Hostname systemd[1]: run-docker-runtime\x2drunc-moby-efe7b48d9f0d7e699227ee88e8321daaa0009fed8b78815a7dcc18d42b2d9261-runc.921NM3.mount: Deactivated successfully.
2024-07-21 14:48:28.667 Hostname systemd[1]: run-docker-runtime\x2drunc-moby-8adb1974aaf3501bd9f0015112d6c0e20e87ad098a638fe97742631700d3fe71-runc.kSKoy5.mount: Deactivated successfully.
2024-07-21 14:50:29.233 Hostname systemd[1]: run-docker-runtime\x2drunc-moby-b52245d68d2779e957774ba06a838efd3f258fe4a203ac37dbdec185c32b9d4c-runc.oA74TP.mount: Deactivated successfully.
2024-07-21 14:53:47.623 Hostname systemd[1]: run-docker-runtime\x2drunc-moby-b52245d68d2779e957774ba06a838efd3f258fe4a203ac37dbdec185c32b9d4c-runc.Ofqc1a.mount: Deactivated successfully.
2024-07-21 14:56:10.132 Hostname systemd[1]: run-docker-runtime\x2drunc-moby-b52245d68d2779e957774ba06a838efd3f258fe4a203ac37dbdec185c32b9d4c-runc.YuDMFE.mount: Deactivated successfully.
2024-07-21 14:56:20.304 Hostname systemd[1]: run-docker-runtime\x2drunc-moby-b52245d68d2779e957774ba06a838efd3f258fe4a203ac37dbdec185c32b9d4c-runc.mhvbhO.mount: Deactivated successfully.
2024-07-21 14:57:20.560 Hostname systemd[1]: run-docker-runtime\x2drunc-moby-efe7b48d9f0d7e699227ee88e8321daaa0009fed8b78815a7dcc18d42b2d9261-runc.edZwhE.mount: Deactivated successfully.
2024-07-21 14:58:47.088 Hostname systemd[1]: run-docker-runtime\x2drunc-moby-efe7b48d9f0d7e699227ee88e8321daaa0009fed8b78815a7dcc18d42b2d9261-runc.3pQ2wm.mount: Deactivated successfully.
2024-07-21 14:59:43.860 Hostname systemd[1]: run-docker-runtime\x2drunc-moby-b52245d68d2779e957774ba06a838efd3f258fe4a203ac37dbdec185c32b9d4c-runc.MALKcC.mount: Deactivated successfully.
2024-07-21 15:01:48.711 Hostname systemd[1]: run-docker-runtime\x2drunc-moby-078736a8158d379b4ef46ff61088d97ed1ada8f2b7d94f153bca42b6e8e4a99f-runc.ipBj9j.mount: Deactivated successfully.
2024-07-21 15:03:02.286 Hostname systemd[1]: run-docker-runtime\x2drunc-moby-b52245d68d2779e957774ba06a838efd3f258fe4a203ac37dbdec185c32b9d4c-runc.MTjZyi.mount: Deactivated successfully.
[... cut ...]
The restart happened at 2024-07-21 19:56 local time, so the last log entries don't even match that timeframe. I also had a look at `ha su logs -b-1 -n1000` but even here the last logged line is from `2024-07-21 17:38:39.772` and without relevant content.

**How can I check if this new host instability is related to this issue (or not)?**
- Should I apply the config.txt workaround (there are no side effects of doing so even when not necessary, are there?)
- Should I wait and see if/when the host crashes/restarts again next time?

**Any other logs which might be helpful here?**

chokays, 2024-08-16:
Does this bug still exist in 13.0?
I skipped 12.3 because of this issue

ZenoBell, 2024-08-16:
same , i really cannot afford atm to have HA down as i have hardwired the lights to be always on and no psysical switches. I havent still bind the switches to the lights directly 
 ..so wife will go enrage mode 😅 skipping updates till this is resolved hopefully 🤞

WilcoSnel, 2024-08-17:
> Does this bug still exist in 13.0? I skipped 12.3 because of this issue

Unfortunately I cannot check as I've migrated to a Pi5 icw NVMe a few months back. Anyone else who can check.

The dtoverlay=dwc2 workaround does work though. So you can change that and then update without any hassle.

Qhilm, 2024-08-17:
> Same issue with Raspberry Pi 3, 32-bit, with Zooz ZST10 usb dongle. Downgrading back to 12.2 solved the problem.

Did not have the issue with ZST10 on Raspberry Pi 4, I did all the 12.2, 12.3 and currently on 12.4.

gnmancini, 2024-08-17:
> same , i really cannot afford atm to have HA down as i have hardwired the lights to be always on and no psysical switches. I havent still bind the switches to the lights directly ..so wife will go enrage mode 😅 skipping updates till this is resolved hopefully 🤞

I was waiting before and until I put the parameter dtoverlay=dwc2 I continued having issues. I highly recommend doing a backup, do the upgrade and see. Worst case scenario you have to restore a backup in a fresh installation.

kozlek, 2024-08-21:
I started to suffer from instability in 12.3, and I was not able to boot on 12.4 (too instable).
I switched to dwc2 to fix my 12.4 installed and everything was working well again.

Yesterday I upgraded to 13.0 with care, and 24 hours later I can say it is stable for me, still using dwc2 (RPI 3B).

Until the issue is fixed upstream, I recommend to update carefully and be prepared to enable dwc2 if you face the symptoms.

Apart from that, everything is working properly !

TheZackCodec, 2024-08-24:
> Does this bug still exist in 13.0? I skipped 12.3 because of this issue

> > 2. docker exec -it $(docker ps -f name=homeassistant -q) bash
> 
> ![immagine](https://private-user-images.githubusercontent.com/96646106/344374884-3708087c-59d6-47f0-9e4a-c8c6f8c1774a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQ0NTkxNjgsIm5iZiI6MTcyNDQ1ODg2OCwicGF0aCI6Ii85NjY0NjEwNi8zNDQzNzQ4ODQtMzcwODA4N2MtNTlkNi00N2YwLTllNGEtYzhjNmY4YzE3NzRhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODI0VDAwMjEwOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTgyYzI2OWI3NTM5YmYwNGFiMTZiMzFlYzc2ZTkwNzQ4ZWRkMDg0ZDIwNDk5N2ZhNTc3YjA5ZTEyMzA1MWU1ZjEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.7kOwD_gHYhCHiGTF0rBAQQaAd6C1vj6ubYcnlBRzHh8)
> 
> What's wrong? RPi3

Do you have protection mode enabled?

Pico1965, 2024-08-24:
The problem still persists for me, having at least 6 reboots a day.
I do not have any device connected to the RP3 USP except a HD SSD.
For completeness I changed the case of the device with an aluminum one but the situation has not changed.

> @Pico1965 Do you have protection mode enabled?

I don't know what it is!

bcutter, 2024-08-25:
I now had these **unexpected host** incidents:
- 2024-04-29 (HA OS 12.1) - sudden reboot
- 2024-07-21 (HA OS 12.4) - sudden reboot
- 2024-08-14 (HA OS 12.4) - massive system instability, need for manual host reboot
- 2024-08-25 (HA OS 12.4) - sudden reboot (while doing a HA recorder repack)

While there's been at least one host related incident before HA OS 12.4, clearly that intensified massively once upgraded from HA OS 12.2 to 12.4 on 2024-07-18.

---

So therefore, seeing some more **random reboots**, I urgently need to ask my questions from [4 weeks ago](https://github.com/home-assistant/operating-system/issues/3362#issuecomment-2241779295) again:

> **How can I check if this new host instability is related to this issue (or not)?**
>
>     * Should I apply the config.txt workaround (there are no side effects of doing so even when not necessary, [...]

---

Sidenote: `ha host logs -b-1 -n1000` always only gives `Logs only contain 1 boots#` which therefore is not helpful at all. Seems like either the previous boot logs can not be read or they are purged after host boot.

[41 more replies]