home-assistant/operating-system #3364: Cannot boot HAOS on Raspberry Pi 3B+

dangerweenie, 2024-05-12:
### Describe the issue you are experiencing

I have a Raspberry Pi 3B and a 3B+

Using both Raspberry Pi Imager and Balena Etcher:

The 3B can boot HAOS, Octoprint, Pi OS

The 3B+ can boot Pi OS, Octoprint, but will _not_ boot HAOS

Powering up the 3B+, the red light remains solid, green light flashes a few times and stays off.  No output to HDMI - simply a blank screen.

Copying over some of the firmware files (.elf, fixup, .dtb) from the Octoprint MicroSD card, I was able to get the rainbow screensaver and the green light remained blinking on and off at regular intervals, but nothing else.

I'm not a pro user but I suspect this has something to do with firmware for the 3B+

### What operating system image do you use?

rpi3-64 (Raspberry Pi 3 64-bit OS)

### What version of Home Assistant Operating System is installed?

12.1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.Flash HAOS to MicroSD card
2.Insert card into RPi 3B+ slor
3.Attach power
...

### Anything in the Supervisor logs that might be useful for us?

```txt
no logs (no boot)
```

### Anything in the Host logs that might be useful for us?

```txt
no logs (no boot)
```

### System information

Raspberry Pi 3B+, Balena Etcher, Raspberry Pi Imager

sairon, 2024-05-13:
Unlike newer models, Raspberry Pi 3B/3B+ doesn't use EEPROM for any of the boot stages, everything is on the SD card, so we can probably rule out anything "on the device" being out of date. The firmware responsible for booting (`bootcode.bin` and other files in the boot partition) is up to date in 12.1 too.

Since you're so far the only one reporting the issue and I am not able to reproduce it on my setup here, I suggest to focus on your setup. Are there any differences when testing with the 3B and 3B+ - e.g. a different power supply? Have you a different one to test with? You can also try appeding `bootdelay=1` to the `config.txt` in the boot partition, it was sometimes reported to fix spurious issues with boot for some other people.

wsrainc, 2024-05-13:
Just to add another data point, I am also a Pi3+ user that had a failed upgrade from 12.1. System was in some sort of reboot loop, with the NIC responding to ping for about 30 seconds, then not responding for another minute or so. That cycle just repeated until I restored to the previous backup from a fresh SD install. I did not put a display on it or troubleshoot further, but I wanted to at least confirm I saw a similar problem. If it recurs when I update again, I'll try and get more details. Appreciate all the awesome work the team does!

sairon, 2024-05-14:
@wsrainc What you describe is different from the OP's problem - in that case even 12.1 fails to boot. If the issue appears after upgrade _from_ 12.1, then it can be related to #3362 (especially if you have any USB peripherals connected).

dangerweenie, 2024-05-15:
> Unlike newer models, Raspberry Pi 3B/3B+ doesn't use EEPROM for any of the boot stages, everything is on the SD card, so we can probably rule out anything "on the device" being out of date. The [...]

Hi there -

Thanks for the reply!

Yup, I know that the 3B+ doens't store on EEPROM (all on SD card) - which explains why OctoPrint can boot - it seems like the FW files in the SDcard boot partition are significant;y different.   This is furthermore supported by being able to boot other Operating Systems - it appears that something in the HAOS image that I was using just isn't playing nicely with the Rpi.

I'll try imaging the SD card again later today with 12.1 and seeing what happens.  Certainly open to any advice.

sairon, 2024-05-15:
Can you try also the latest 12.3? It should be already available in rpi-imager and offered by that by default, or you can download it from [releases](https://github.com/home-assistant/operating-system/releases/tag/12.3) here. This release has updated kernel and firmware, so it can make some difference.

dangerweenie, 2024-05-15:
> Can you try also the latest 12.3? It should be already available in rpi-imager and offered by that by default, or you can download it from [...]

I'll give it a shot and follow up soon, thanks!

CreatureFear, 2024-05-18:
Hi, 
I'm a linux noob here, but I have a RPI 3B+ and yesterday tried to update my HA setup to version 12.3 and since then I'm in the dark (HA does not boot up properly anymore).

Had the following message :
`homeasssistant a start job is running for /dev/disk/by-label/hassos-overlay`
then it goes into emergency console after a few dependencies failures.

I've just tried to install 12.3 on a fresh SD card with RPI Imager and it still goes into emergency console .

Boot up sequence now fails at the line :
`A start job is running for Wait Until Kernel Time Synchronized`

[image]

At first I thought I had a failing SD card that needed replacement, but this is def an OS version issue.

rhatguy, 2024-06-08:
This is definitely a problem with HAOS 12.3.  I unplugged by Zwave stick (a Zooz ZST-10) and the OS booted.  I was then able to downgrade the OS to 12.2.  I plugged the stick back in and all is working.  Just to be sure I upgraded the OS again to 12.3 and around 30 seconds after boot it started back up with the same errors as pasted above.  I followed the same process of pulling the stick, downgrading the OS, re-inserting the stick and booting back up and everything is fine again.

Perhaps #3301 is an issue?

Baruxa, 2024-09-06:
Fine. I'll try to install the latest version and report back

kenimou, 2024-10-13:
I am also having issuer with my pi3B/B+. For me my green light is up but nothing outputs to HDMI (no signal). Have tweaked some settings to force hdmi but no luck either. Connected to ethernet but it never joined the network according to my router. I tried HAOS 13.1 and 12.4 and nether works.

PBorowy, 2024-10-15:
I'm having the same problem as @kenimou.
I was trying different MicroSD Card, different way to flash the card, and still only one tiny green LED flash and thats it. Ethernet do not work, HDMI do not show anything on the monitor.

closed 2025-01-23 by github-actions[bot]

MikeDabrowski, 2026-02-01:
I have pi3B+ and official display. The thing goes up to the point of 'waiting for HA to start' but then fails. The logs tell that container couldn't start :( I gave up