home-assistant/operating-system #3365: HassOS data resizing timeout

agners, 2024-05-13:
**Same issue:**
**Fresh install** on a Lenovo M90n-IoT with an i3, 4gb ram and a 128gb low-performance NVMe SSD.
Computer had windows before on the drive.
Booted Ubuntu Live, and used Disk to restore and image (HAOS 12.2) as per HA tutorial.
After 1m30s, timeout to resize the data partition:
`[DEPEND] Dependency failed for HassOS data resizing.`

Command line (From CTRL-ALT-F2) with blkid showed part p8 to be correctly labeled (hassos-data)
Ping to 8.8.8.8 and local network worked.

**What worked:**
Booted Ubuntu Live, used Disk to FORMAT DISK, with option "empty no partitions" and "slow write zeroes everywhere". Then used Disk to restore an image (HAOS 12.2) as per HA tutorial.

On bootup, resizing the SWAP took 30s, resizing data was so fast I did not see going trough (so, less than 2s).
...
and now it works.

I'm enclined to say it's a bug, as strictly following instructions gives a dead-end.
Maybe the data resizing needs more time on a non-zeroed disk to perform the duty?
Why put a timeout anyway, if the whole thing fails when it fails??
If zeroing the disk is critical, probably something important is incorrectly defined in the disk image? Like not zeroing a trailing sector after a partition table that can add garbage to it?

Anyway, that is my first welcome to HA, and it's already a bug hunt.
Hope this helps other users and devs! :)

_Originally posted by @LionRelaxe in https://github.com/home-assistant/operating-system/issues/2272#issuecomment-2077001018_

bryanvaz, 2024-07-20:
+1

This issue bogged me down for 4 hours because SSD was previously used as a VM storage drive. Did a `dd if=/dev/zero` then reimaged, and the new image immediately booted into the HA cli, no delay or timeout for the resizing step.

BradB111, 2024-10-10:
+1

Trying to use a Zima board with an eMMC and this is the first time I've come across anything that reflects my experience, I have not been able to install HAOS yet however I'm mid way through a zero write. 

I would like to echo the thoughts of @LionRelaxe. A timeout during a one time setup doesn't seem necessary.

LionRelaxe, 2025-01-09:
Please! Don't let this expire :)
I would fix this myself if I knew how...

BradB111, 2025-01-09:
> Please! Don't let this expire :) I would fix this myself if I knew how...

This needs to be fixed in the installer to allow a much longer or no timeout for the process it gets stuck on.

In the meantime if you 0 write the drive you are tying to install it on it should allow for quicker installation to get through the section before the timeout. Worked straight away.

ilspleno, 2025-02-10:
> This needs to be fixed in the installer to allow a much longer or no timeout for the process it gets stuck on.
>
> In the meantime if you 0 write the drive you are tying to install it on it should [...]

THANK YOU. I spent 3-4 hours banging my head against this. Zero writing the drive fixed it for me too. 

Homelabbers frequently repurpose hardware, the "installer" should handle this better.

LionRelaxe, 2025-05-12:
Bump.
Please, remove the data resizing timeout!
Keep up the good work :)

DestrinStorm, 2025-05-28:
Just to add more votes/traffic.  My first exposure to trying to setup HA today was running into this bug - it's very not clear what was going on but a zero format on the drive did eventually fix it and get me to a working boot.  This is currently going to be a potential hard stop for users setting up on 'spare' hardware who don't know to go diving into the guts to work out why, should be fixed to make for a cleaner install

BradB111, 2025-11-14:
As far as I'm aware this is still an issue that needs to be addressed