home-assistant/operating-system #3367: Error :"  brcmfmac: brcmf_set_channel: set chanspec 0xd0...fail, reason -52 "

davidclaverie, 2024-05-13:
### Describe the issue you are experiencing

i have since last days after update this error brcmfmac: brcmf_set_channel: set chanspec 0xd0...fail, reason -52

may be the same i see here
[https://github.com/raspberrypi/linux/issues/6049](https://l.facebook.com/l.php?u=https%3A%2F%2Fgithub.com%2Fraspberrypi%2Flinux%2Fissues%2F6049%3Ffbclid%3DIwZXh0bgNhZW0CMTAAAR1gpZXwuzZB6cR5ss5pMXxhSerY9UquKbi9JeeACqeJRQ0jKf6gGedX8pY_aem_AYLmQVNpZzXJX0_tU5J9ZENZiwLRp2-jHhS0bNU-FE74Bs0beqYHo-Ya4uHhcYBFhcOBl-NGzZhZkV6D_YqBuxkT&h=AT3RWI-OQCx9_lLHl2TRVN9y6wm1f0otfeZFr85fLxTVZFi8Xk8es8bwdqrxLkKedPoaS31UgIvFlm1-jhkjxTxdB-rc1UJpZQFLRbnsbQjIiR7rcFx-rY0kmwHPeByAxQ&__tn__=R]-R&c[0]=AT0P-7cyDbcm0Ajf5rktar8i0N0RxbBkcUv6KWI5p-2OXYQSGaZJY_cMHGk_RWVmefIV9wPIJ68a5EFuEBf83rHepmp6SiW3YOFn8KMGpSrkfAMEgDm8GWrCklBRURtHx_aq0MrpqeN-0ye-lv6XUMMiCuScTu6gy3I2awFiMwTrzeAzAUqCxPN0MzL7uQU7)

Thanks to help me.
im not a 'system user'.. 

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

6.6.28-haos-raspi

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. When i reboot the system
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
2024-05-13 14:49:33.836 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2MQTT (486e6e9b_zigbee2mqtt)
2024-05-13 14:49:34.100 INFO (MainThread) [supervisor.resolution.checks.base] Run check for disabled_data_disk/system
2024-05-13 14:49:34.101 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_ipv6_error/dns_server
2024-05-13 14:49:34.231 INFO (MainThread) [supervisor.resolution.checks.base] Run check for ipv4_connection_problem/system
2024-05-13 14:49:34.232 INFO (MainThread) [supervisor.resolution.checks.base] Run check for trust/supervisor
2024-05-13 14:49:34.252 INFO (MainThread) [supervisor.resolution.checks.base] Run check for free_space/system
2024-05-13 14:49:34.253 INFO (MainThread) [supervisor.resolution.checks.base] Run check for docker_config/system
2024-05-13 14:49:34.254 INFO (MainThread) [supervisor.resolution.checks.base] Run check for no_current_backup/system
2024-05-13 14:49:34.255 INFO (MainThread) [supervisor.resolution.checks.base] Run check for multiple_data_disks/system
2024-05-13 14:49:34.256 INFO (MainThread) [supervisor.resolution.checks.base] Run check for security/core
2024-05-13 14:49:34.257 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_failed/dns_server
2024-05-13 14:49:34.383 INFO (MainThread) [supervisor.resolution.check] System checks complete
2024-05-13 14:49:34.383 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state running
2024-05-13 14:49:34.857 INFO (MainThread) [supervisor.host.services] Updating service information
2024-05-13 14:49:34.872 INFO (MainThread) [supervisor.host.network] Updating local network information
2024-05-13 14:49:35.208 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
2024-05-13 14:49:35.227 INFO (MainThread) [supervisor.host.manager] Host information reload completed
2024-05-13 14:49:35.371 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
2024-05-13 14:49:35.372 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state running
2024-05-13 14:49:35.372 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
2024-05-13 14:49:37.640 WARNING (MainThread) [supervisor.addons.addon] Watchdog found addon Linky is stopped, restarting...
2024-05-13 14:49:37.653 INFO (SyncWorker_1) [supervisor.docker.manager] Cleaning addon_cf6b56a3_linky application
2024-05-13 14:49:38.362 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on ghcr.io/bokub/ha-linky-aarch64 with version 1.4.0
2024-05-13 14:49:45.585 WARNING (MainThread) [supervisor.addons.addon] Watchdog found addon Linky is stopped, restarting...
2024-05-13 14:49:45.603 INFO (SyncWorker_2) [supervisor.docker.manager] Cleaning addon_cf6b56a3_linky application
2024-05-13 14:49:46.335 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on ghcr.io/bokub/ha-linky-aarch64 with version 1.4.0
2024-05-13 14:49:53.561 WARNING (MainThread) [supervisor.addons.addon] Watchdog found addon Linky is stopped, restarting...
2024-05-13 14:49:53.579 INFO (SyncWorker_2) [supervisor.docker.manager] Cleaning addon_cf6b56a3_linky application
2024-05-13 14:49:54.338 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on ghcr.io/bokub/ha-linky-aarch64 with version 1.4.0
2024-05-13 14:49:54.798 WARNING (MainThread) [supervisor.api.ingress] No valid ingress session 2afc3da3f78f2a83e84e0e1b3af90f0b53abf5e2cd39a9ffbf8ab9818558fba4c68a8a128c4ff16f493969e16dcf5c5d02aca14ebfb719f908f3dbfd0a7dec1b
2024-05-13 14:50:01.657 WARNING (MainThread) [supervisor.addons.addon] Watchdog found addon Linky is stopped, restarting...
2024-05-13 14:50:01.675 INFO (SyncWorker_0) [supervisor.docker.manager] Cleaning addon_cf6b56a3_linky application
2024-05-13 14:50:02.411 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on ghcr.io/bokub/ha-linky-aarch64 with version 1.4.0
2024-05-13 14:50:09.906 WARNING (MainThread) [supervisor.addons.addon] Watchdog found addon Linky is stopped, restarting...
2024-05-13 14:50:09.928 INFO (SyncWorker_3) [supervisor.docker.manager] Cleaning addon_cf6b56a3_linky application
2024-05-13 14:50:10.735 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on ghcr.io/bokub/ha-linky-aarch64 with version 1.4.0
2024-05-13 14:50:15.799 WARNING (MainThread) [supervisor.api.ingress] No valid ingress session 2afc3da3f78f2a83e84e0e1b3af90f0b53abf5e2cd39a9ffbf8ab9818558fba4c68a8a128c4ff16f493969e16dcf5c5d02aca14ebfb719f908f3dbfd0a7dec1b
2024-05-13 14:50:18.021 WARNING (MainThread) [supervisor.addons.addon] Watchdog found addon Linky is stopped, restarting...
[... 237 lines cut ...]
resources | 33
views | 65
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 3 mai 2024 à 05:34
-- | --
current_recorder_run | 13 mai 2024 à 14:47
estimated_db_size | 333.26 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

### Additional information

i think its the same like i see here??
[https://github.com/raspberrypi/linux/issues/6049](https://github.com/raspberrypi/linux/issues/6049?fbclid=IwZXh0bgNhZW0CMTAAAR1gpZXwuzZB6cR5ss5pMXxhSerY9UquKbi9JeeACqeJRQ0jKf6gGedX8pY_aem_AYLmQVNpZzXJX0_tU5J9ZENZiwLRp2-jHhS0bNU-FE74Bs0beqYHo-Ya4uHhcYBFhcOBl-NGzZhZkV6D_YqBuxkT)

but im not a "system user" 

thanks to help me

spanzetta, 2024-05-22:
Exactly the same issue here...
Using RPI3A+..
I hope it will be fixed because it is causing big instability

spanzetta, 2024-06-04:
No news on this?

I still have the problem at every boot..

How to fix it??

sadranyi, 2024-06-04:
I am also experiencing the same issue after an update

dduster, 2024-06-07:
Same for me. RPi3B+ *and* RPi5.

Core 2024.6.0
Supervisor 2024.06.0
Operating System 12.3
Frontend 20240605.0

spanzetta, 2024-06-08:
I hope that developers are aware of this problem and that they are doing something to fix it in next build..

tsanidisDev, 2024-06-11:
I am experiencing the same and HA keeps crashing.

radkokeves, 2024-06-12:
receiving brcmfmac: brcmf_set_channel: set chanspec 0xd0...fail, reason -52 as well

spanzetta, 2024-06-13:
I wonder if someone is working on fixing it..
It is clear now that the issue is happening on many Raspberry users (if not all)

Mice30, 2024-06-17:
Me too.
Rpi 4 8GB
Core 2024.6.3

spanzetta, 2024-06-23:
Solved by downgrading hassio to last working version 12.2 using: ha os update --version 12.2

Now system is again stable without that error

henkkakay, 2024-06-24:
Putting this here just in case it is related: https://github.com/mattlongman/Hassio-Access-Point/issues/75 - because the problem appeared around the same time (when I upgraded to 12.3) and gone away when I downgraded to 12.2 and is also wifi related.

The stick I am using as access point should be TP-LINK Archer T3U Plus which should have a Realtek RTL8812BU inside (so not a Broadcom)

radkokeves, 2024-06-24:
@henkkakay is "TP-LINK Archer T3U Plus" working for you OOTB with 12.2 on rpi ?

henkkakay, 2024-06-25:
> @henkkakay is "TP-LINK Archer T3U Plus" working for you OOTB with 12.2 on rpi ?

yeap - its chipset was added to the standard library of drivers I think a bit more than a year ago. 
It works well acting as a wifi access point - so my Tuya stuff latches off it, and I block internet access so all my Tuya devices are local-only (localTuya)

Note (as per the other thread) - it stopped working 12.3, havent tested for 12.4. 12.2 was (and still is) good.

beewebmeister, 2024-06-26:
Same for me. 
RPi3B+
Core 2024.6.4
Supervisor 2024.06.2
Operating System 12.4
Frontend 20240610.1
Aeotec zi-stick (zigbee) and z-stick (zwave) are both connected.

If I remove the zwave USB dongle (z-stick), the error goes away.
I confirm that everything works by downgrading the OS to the last working version 12.2.
Please fix this issue.
Thanks

BENOIT92200, 2024-07-09:
Exact same problem here since a few days now.  A lot annoying cause it happends every single restart or shutdown and cause long long delays ....for simply rebooting or SHdown.
Using UBUNTU 24.04LTS on my RPI5 .  Will try to Timeshift restore from last 2 weeks to confirm it was ok then..
Hope developer are looking at this bug soon...
Brgds

Lensters, 2024-07-27:
I'm getting the same thing since enabling my wifi with different channel numbers.

SkyConnect v1.0 supporting zha
Core 2024.7.3
Supervisor 2024.06.2
Operating System 12.4
Frontend 20240710.0

frdsky, 2024-08-01:
Can not install Homeassistant on a new Rpi4, because installation halts on this error. Other operating systems install without problem. No usb sticks attached.

sairon, 2024-08-02:
I can only acknowledge the issue but it needs to be resolved in the Raspberry Pi kernel. Please refer to there for potential solutions. Once it's resolved there, we can do some steps on the HAOS side.

frdsky, 2024-08-05:
Thanks Jan! 
I looked for a solution, but I'm not able to find it. Can only wait for an update and try again after some months. It looks like Homeassistant is too complicated for me as this is only the set-up fase..

utoddl, 2024-08-05:
I found a solution here: https://archlinuxarm.org/forum/viewtopic.php?t=17010&p=72799#p72819
Specifically, create a file `/etc/modprobe.d/brcmfmac.conf` containing this line:
```txt
options brcmfmac feature_disable=0x82000
```
and reboot.

Note: this problem is not specific to home-assistant. This solution worked for me for the exact same problem on Fedora 40 Server on a RPi3b, as well as for folks running Arch. There's a long thread in Linux Kernel about this if you want to dig deeper.

frdsky, 2024-08-05:
Wow! Thank you Tod. I 'll try it first thing tomorrow.

atith, 2024-08-07:
> I found a solution here: https://archlinuxarm.org/forum/viewtopic.php?t=17010&p=72799#p72819 Specifically, create a file `/etc/modprobe.d/brcmfmac.conf` containing this line:
>
> ```
> options brcmfmac [...]

I get the exact same error. And sorry for the stupid question but how did you create the file? I'm not able to access the home assistant through http://homeassistant.local:8123

I have tried through HA commands but failed so far.  Can you help with that?

frdsky, 2024-08-08:
Thanks you Athit! I was also not able to create the file. Tried many times
and Samba did not help. Decided to give up for now.

atith, 2024-08-08:
> Thanks you Athit! I was also not able to create the file. Tried many times and Samba did not help. Decided to give up for now.
> […](#)

same for me! Thanks for trying tho.. let's hope for an OS Update which fix that..

frdsky, 2024-09-06:
Gordon Hollingworth of the Raspberry company replied: "It looks like HomeAssistant is incorrectly configured for PI 4 Wifi."
Any news, Sairon? Or will the developers keeps pointing at each other and leave the users behind?

utoddl, 2024-09-06:
> I get the exact same error. And sorry for the stupid question but how did you create the file? I'm not able to access the home assistant through http://homeassistant.local:8123
>
> I have tried through [...]

I'm not familiar at all with the home-assistant and wouldn't know what to suggest. My hardware is a Raspberry Pi 3b where I have root access on the command line. Perhaps someone else can suggest ways to accomplish the same thing on your hardware. Sorry.

jires, 2024-09-06:
> Is there any news on this? I am still keeping HAOS 12.2.. while now there is 13.1 available (before upgrading I would like to be sure that this issue is sorted out).. Can someone confirm? Thanks

I can confirm that issue exist in version 13.1 on RPi 4

sairon, 2024-09-09:
@frdsky 
>  Any news, Sairon? Or will the developers keeps pointing at each other and leave the users behind?

Please see to the bug reported in the raspberrypi/linux repository linked in the first post of this issue, there are people who report this behavior on Raspberry Pi OS as well, so it's clearly not HAOS-only issue. But it's indeed pointless to report an issue with Home Assistant installation in Raspberry Pi OS feedback tracker, so I understand it got promptly closed. The issue is still opened in the kernel repository and no real solution is known yet. Speaking for HAOS, we do not have the knowledge and capacity needed to debug the root cause of the issue, this really needs to be addressed upstream first.

spanzetta, 2024-10-18:
I can confirm that also HAOS 13.2 still has the problem..
Last one working is still 12.2..

phuxtick, 2024-11-01:
Is this ever going to be fixed?  HAOS 13.2 on RasPi 4B _still_ has this issue and I do know how to create that modprobe file...but it flat out doesn't care about it.  That solved nothing, I'm still puking out noise to the TTY unable to connect it to anything

frdsky, 2024-11-01:
I am happy you found a way to make it work for you, phuxtick. Could you explain to the rest of us how you made the modprobe.d file stick?
I still feel the issue should be mentioned in the compatibility list. Thousands of enthusiasts have been disillusioned by this bug.

phuxtick, 2024-11-01:
> I am happy you found a way to make it work for you, phuxtick. Could you explain to the rest of us how you made the modprobe.d file stick? I still feel the issue should be mentioned in the compatibility list. Thousands of enthusiasts have been disillusioned by this bug.

Sure, here's my exact steps (I plugged in a keyboard and mouse) -

- Type `login` at the prompt
- Type `echo "options brcmfmac feature_disable=0x82000" > /etc/modprobe.d/brcmfmac.conf`
- Type `reboot`

Then when it came back up, I was able to log back in and do `nmcli device wifi connect “YOUR_SSID” password “YOUR_WIFI_PASSWORD”` and everything fired right up, WiFi connected, and I was discovering all the goodies in my network I can now take control of.

n8henrie, 2024-12-04:
Most informative post I've found on this error: https://github.com/raspberrypi/linux/issues/6049#issuecomment-2485431104

commit cbffdbc3 (2024-12-05): Bump RPi WiFi firmware to latest version

* buildroot c65b0306bb...b2077df873 (1):
  > package/brcmfmac_sdio-firmware-rpi: bump version to 4c1789e

Raspberry Pi kernel 6.6 driver for BCM43455 (used in RPi 3B+/4B) calls new API
which uses the DUMP_OBSS feature for channel selection. If it's not preset, it
results in drivers reporting errors, e.g.:

brcmf_set_channel: set chanspec 0xd099 fail, reason -52

The RPi OS firmware was updated but the package we use for this firmware
contains an old version that lacks this support. Update to latest version
synced from RPi upstream to fix the issues. Th [... cut]
commit 8d61db63 (2024-12-05): Bump RPi SDIO WiFi firmware to latest version

* buildroot c65b0306bb...b2077df873 (1):
  > package/brcmfmac_sdio-firmware-rpi: bump version to 4c1789e

Raspberry Pi kernel 6.6 driver for BCM43455 (used in RPi 3B+/4B) calls new API
which uses the DUMP_OBSS feature for channel selection. If it's not preset, it
results in drivers reporting errors, e.g.:

brcmf_set_channel: set chanspec 0xd099 fail, reason -52

The RPi OS firmware was updated but the package we use for this firmware
contains an old version that lacks this support. Update to latest version
synced from RPi upstream to fix the issue [... cut]
PR #3719 (2024-12-05): Bump RPi SDIO WiFi firmware to latest version
* buildroot c65b0306bb...b2077df873 (1):
  > package/brcmfmac_sdio-firmware-rpi: bump version to 4c1789e

Raspberry Pi kernel 6.6 driver for BCM43455 (used in RPi 3B+/4B) calls new API which uses the DUMP_OBSS feature for channel selection. If it's not preset, it results in drivers reporting errors, e.g.:

brcmf_set_channel: set chanspec 0xd099 fail, reason -52

The RPi OS firmware was updated but the package we use for this firmware contains an old version that lacks this support. Update to latest version synced from RPi upstream to fix the issues [... cut]
commit ce1e261f (2024-12-05): Bump RPi SDIO WiFi firmware to latest version (#3719)

* buildroot c65b0306bb...b2077df873 (1):
  > package/brcmfmac_sdio-firmware-rpi: bump version to 4c1789e

Raspberry Pi kernel 6.6 driver for BCM43455 (used in RPi 3B+/4B) calls new API
which uses the DUMP_OBSS feature for channel selection. If it's not preset, it
results in drivers reporting errors, e.g.:

brcmf_set_channel: set chanspec 0xd099 fail, reason -52

The RPi OS firmware was updated but the package we use for this firmware
contains an old version that lacks this support. Update to latest version
synced from RPi upstream to fix t [... cut]
closed 2024-12-05 by sairon