home-assistant/operating-system #3368: HA does not reconnect to network after router reboot

andreaswemme, 2024-05-13:
### Describe the issue you are experiencing

HAOS does not reconnect to network after router reboot. DHCP used. Changed to static IP vid address reservation in the router. 

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

12.3

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Check that HA is running ok
2. Reboot router
3. HA loosing network connection
4. HA will never regain network connection
5. Forced restart needed to regain network connection
...

### Anything in the Supervisor logs that might be useful for us?

```txt
2024-05-13 21:21:01.412 WARNING (MainThread) [homeassistant.helpers.entity] Update of media_player.spotify_wemme is taking over 10 seconds
2024-05-13 21:21:01.422 WARNING (SyncWorker_13) [urllib3.connectionpool] Retrying (Retry(total=2, connect=None, read=False, redirect=None, status=3)) after connection broken by 'NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7f5ae306c4d0>: Failed to establish a new connection: [Errno -3] Try again')': /v1/me/player?additional_types=episode
2024-05-13 21:21:12.032 WARNING (SyncWorker_13) [urllib3.connectionpool] Retrying (Retry(total=1, connect=None, read=False, redirect=None, status=3)) after connection broken by 'NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7f5ae41f68a0>: Failed to establish a new connection: [Errno -3] Try again')': /v1/me/player?additional_types=episode
2024-05-13 21:21:21.411 WARNING (MainThread) [homeassistant.components.media_player] Updating spotify media_player took longer than the scheduled update interval 0:00:30
2024-05-13 21:21:23.242 WARNING (SyncWorker_13) [urllib3.connectionpool] Retrying (Retry(total=0, connect=None, read=False, redirect=None, status=3)) after connection broken by 'NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7f5ae4258d10>: Failed to establish a new connection: [Errno -3] Try again')': /v1/me/player?additional_types=episode
2024-05-13 21:21:23.665 ERROR (MainThread) [async_upnp_client.ssdp] Received error: [Errno 101] Network unreachable, transport: <_SelectorDatagramTransport fd=76 read=polling write=<idle, bufsize=0>>, socket: <asyncio.TransportSocket fd=76, family=2, type=2, proto=0, laddr=('0.0.0.0', 49858)>
2024-05-13 21:21:23.665 ERROR (MainThread) [async_upnp_client.ssdp] Received error: [Errno 99] Address not available, transport: <_SelectorDatagramTransport fd=88 read=polling write=<idle, bufsize=0>>, socket: <asyncio.TransportSocket fd=88, family=10, type=2, proto=0, laddr=('::', 50237, 0, 0)>
2024-05-13 21:21:53.666 ERROR (MainThread) [async_upnp_client.ssdp] Received error: [Errno 101] Network unreachable, transport: <_SelectorDatagramTransport fd=76 read=polling write=<idle, bufsize=0>>, socket: <asyncio.TransportSocket fd=76, family=2, type=2, proto=0, laddr=('0.0.0.0', 49858)>
2024-05-13 21:21:53.666 ERROR (MainThread) [async_upnp_client.ssdp] Received error: [Errno 99] Address not available, transport: <_SelectorDatagramTransport fd=88 read=polling write=<idle, bufsize=0>>, socket: <asyncio.TransportSocket fd=88, family=10, type=2, proto=0, laddr=('::', 50237, 0, 0)>
2024-05-13 21:22:01.064 WARNING (SyncWorker_6) [urllib3.connectionpool] Retrying (Retry(total=2, connect=None, read=False, redirect=None, status=3)) after connection broken by 'NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7f5ae2751400>: Failed to establish a new connection: [Errno -3] Try again')': /v1/me/player/devices
2024-05-13 21:22:01.413 WARNING (MainThread) [homeassistant.helpers.entity] Update of media_player.spotify_wemme is taking over 10 seconds
2024-05-13 21:22:01.422 WARNING (SyncWorker_19) [urllib3.connectionpool] Retrying (Retry(total=2, connect=None, read=False, redirect=None, status=3)) after connection broken by 'NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7f5af1ffb980>: Failed to establish a new connection: [Errno -3] Try again')': /v1/me/player?additional_types=episode
2024-05-13 21:22:11.674 WARNING (SyncWorker_6) [urllib3.connectionpool] Retrying (Retry(total=1, connect=None, read=False, redirect=None, status=3)) after connection broken by 'NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7f5ae2fedaf0>: Failed to establish a new connection: [Errno -3] Try again')': /v1/me/player/devices
2024-05-13 21:22:12.032 WARNING (SyncWorker_19) [urllib3.connectionpool] Retrying (Retry(total=1, connect=None, read=False, redirect=None, status=3)) after connection broken by 'NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7f5ae2fef680>: Failed to establish a new connection: [Errno -3] Try again')': /v1/me/player?additional_types=episode
2024-05-13 21:22:14.001 ERROR (MainThread) [homeassistant.components.trafikverket_train.coordinator] Error requesting trafikverket_train data: Cannot connect to host api.trafikinfo.trafikverket.se:443 ssl:default [None]
2024-05-13 21:22:21.412 WARNING (MainThread) [homeassistant.components.media_player] Updating spotify media_player took longer than the scheduled update interval 0:00:30
2024-05-13 21:22:22.884 WARNING (SyncWorker_6) [urllib3.connectionpool] Retrying (Retry(total=0, connect=None, read=False, redirect=None, status=3)) after connection broken by 'NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7f5ae2fed640>: Failed to establish a new connection: [Errno -3] Try again')': /v1/me/player/devices
2024-05-13 21:22:23.242 WARNING (SyncWorker_19) [urllib3.connectionpool] Retrying (Retry(total=0, connect=None, read=False, redirect=None, status=3)) after connection broken by 'NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7f5ae2fedf10>: Failed to establish a new connection: [Errno -3] Try again')': /v1/me/player?additional_types=episode
2024-05-13 21:22:23.667 ERROR (MainThread) [async_upnp_client.ssdp] Received error: [Errno 101] Network unreachable, transport: <_SelectorDatagramTransport fd=76 read=polling write=<idle, bufsize=0>>, socket: <asyncio.TransportSocket fd=76, family=2, type=2, proto=0, laddr=('0.0.0.0', 49858)>
2024-05-13 21:22:23.667 ERROR (MainThread) [async_upnp_client.ssdp] Received error: [Errno 99] Address not available, transport: <_SelectorDatagramTransport fd=88 read=polling write=<idle, bufsize=0>>, socket: <asyncio.TransportSocket fd=88, family=10, type=2, proto=0, laddr=('::', 50237, 0, 0)>
2024-05-13 21:22:32.894 ERROR (MainThread) [homeassistant.components.spotify] Error fetching wemme Devices data: 
2024-05-13 21:22:51.237 WARNING (MainThread) [async_upnp_client.profiles.profile] Failed (re-)subscribing to: uuid:068e7780-00b6-1000-9f27-1449e06fad74, reason: UpnpConnectionError("ClientConnectorError(ConnectionKey(host='192.168.0.124', port=7676, is_ssl=False, ssl=True, proxy=None, proxy_auth=None, proxy_headers_hash=437209601818717696), OSError(101, 'Network unreachable'))", None)
2024-05-13 21:22:51.238 WARNING (MainThread) [async_upnp_client.profiles.profile] Failed (re-)subscribing to: uuid:068e7782-00b6-1000-bb1d-1449e06fad74, reason: UpnpConnectionError("ClientConnectorError(ConnectionKey(host='192.168.0.124', port=7676, is_ssl=False, ssl=True, proxy=None, proxy_auth=None, proxy_headers_hash=437209601818717696), OSError(101, 'Network unreachable'))", None)
2024-05-13 21:22:51.239 WARNING (MainThread) [async_upnp_client.profiles.profile] Failed (re-)subscribing to: uuid:068e7784-00b6-1000-95fc-1449e06fad74, reason: UpnpConnectionError("ClientConnectorError(ConnectionKey(host='192.168.0.124', port=7676, is_ssl=False, ssl=True, proxy=None, proxy_auth=None, proxy_headers_hash=437209601818717696), OSError(101, 'Network unreachable'))", None)
2024-05-13 21:22:51.287 WARNING (MainThread) [homeassistant.components.sonos.speaker] No recent activity and cannot reach Vardagsrum 2, marking unavailable
2024-05-13 21:22:53.668 ERROR (MainThread) [async_upnp_client.ssdp] Received error: [Errno 101] Network unreachable, transport: <_SelectorDatagramTransport fd=76 read=polling write=<idle, bufsize=0>>, socket: <asyncio.TransportSocket fd=76, family=2, type=2, proto=0, laddr=('0.0.0.0', 49858)>
2024-05-13 21:22:53.668 ERROR (MainThread) [async_upnp_client.ssdp] Received error: [Errno 99] Address not available, transport: <_SelectorDatagramTransport fd=88 read=polling write=<idle, bufsize=0>>, socket: <asyncio.TransportSocket fd=88, family=10, type=2, prot [... cut ...] , 0)>
```

### Anything in the Host logs that might be useful for us?

```txt
Not what I  can see
[... 79 lines cut ...]
dashboards | 2
-- | --
resources | 1
views | 7
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 3 maj 2024 11:25
-- | --
current_recorder_run | 13 maj 2024 21:24
estimated_db_size | 86.35 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

sairon, 2024-05-14:
Are there any network devices between the router and HAOS host? What device is it running on exactly? Complete host logs will be useful in this case, as they should indicate if there's something wrong with NetworkManager or link detection. You can get them also from the previous boot if you perform a forced restart using `ha host logs -b-1 -n500` (eventually adjust the `-b` parameter for older boots or `-n` for lines only containing the relevant output).

andreaswemme, 2024-05-14:
Hi @sairon! Thank you for your reply. The HAOS host is a HP ProDesk 400 G6 Mini PC and it is only running HAOS on it. No virtual machine. 

Here's the complete log file since the yesterday's router reboot tests.

```
2024-05-13 19:06:48.574 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration resrobot which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-05-13 19:06:48.575 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-05-13 19:06:48.575 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration ui_lovelace_minimalist which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-05-13 19:06:48.576 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration gtfs2 which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-05-13 19:06:48.576 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration hasl3 which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-05-13 19:06:50.041 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'hacs' accesses hass.components.frontend. This is deprecated and will stop working in Home Assistant 2024.9, it should be updated to import functions used from frontend directly at custom_components/hacs/frontend.py, line 68: hass.components.frontend.async_register_built_in_panel(, please create a bug report at https://github.com/hacs/integration/issues
2024-05-13 19:06:50.110 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'hasl3' calls `async_add_job`, which is deprecated and will be removed in Home Assistant 2025.4; Please review https://developers.home-assistant.io/blog/2024/03/13/deprecate_add_run_job for replacement options at custom_components/hasl3/__init__.py, line 311: hass.async_add_job(hass.config_entries.async_forward_entry_setup(entry, "sensor")), please create a bug report at https://github.com/hasl-sensor/integration/issues
2024-05-13 19:06:50.111 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'hasl3' calls `async_add_job`, which is deprecated and will be removed in Home Assistant 2025.4; Please review https://developers.home-assistant.io/blog/2024/03/13/deprecate_add_run_job for replacement options at custom_components/hasl3/__init__.py, line 312: hass.async_add_job(hass.config_entries.async_forward_entry_setup(entry, "binary_sensor")), please create a bug report at https://github.com/hasl-sensor/integration/issues
2024-05-13 19:06:52.951 ERROR (MainThread) [homeassistant.helpers.event] Error while processing template: Template<template=({{ state_attr('sensor.rr_740030621_740000088_route_sensor_linje1_mot_trelleborgc', 'trips')[0]['legs'][0]['stops'][0]['depTime'] }}) renders=2>
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 585, in async_render
    render_result = _render_with_context(self.template, compiled, **kwargs)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 2600, in _render_with_context
    return template.render(**kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/jinja2/environment.py", line 1304, in render
    self.environment.handle_exception()
  File "/usr/local/lib/python3.12/site-packages/jinja2/environment.py", line 939, in handle_exception
    raise rewrite_traceback_stack(source=source)
  File "<template>", line 1, in top-level template code
  File "/usr/local/lib/python3.12/site-packages/jinja2/sandbox.py", line 304, in getitem
    return obj[argument]
           ~~~^^^^^^^^^^
jinja2.exceptions.UndefinedError: None has no element 0

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 713, in async_render_to_info
    render_info._result = self.async_render(
                          ^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 587, in async_render
    raise TemplateError(err) from err
homeassistant.exceptions.TemplateError: UndefinedError: None has no element 0
2024-05-13 19:06:52.958 ERROR (MainThread) [homeassistant.components.template.template_entity] TemplateError('UndefinedError: None has no element 0') while processing template 'Template<template=({{ state_attr('sensor.rr_740030621_740000088_route_sensor_linje1_mot_trelleborgc', 'trips')[0]['legs'][0]['stops'][0]['depTime'] }}) renders=4>' for attribute '_attr_native_value' in entity 'sensor.departure_time_trelleborgc'
2024-05-13 19:06:52.959 ERROR (MainThread) [homeassistant.helpers.event] Error while processing template: Template<template=({{ state_attr('sensor.rr_740030621_740037472_route_sensor_linje1_mot_albackshallen', 'trips [... cut ...]
2024-05-13 21:23:23.669 ERROR (MainThread) [async_upnp_client.ssdp] Received error: [Errno 101] Network unreachable, transport: <_SelectorDatagramTransport fd=76 read=polling write=<idle, bufsize=0>>, socket: <asyncio.TransportSocket fd=76, family=2, type=2, proto=0, laddr=('0.0.0.0', 49858)>
2024-05-13 21:23:23.669 ERROR (MainThread) [async_upnp_client.ssdp] Received error: [Errno 99] Address not available, transport: <_SelectorDatagramTransport fd=88 read=polling write=<idle, bufsize=0>>, socket: <asyncio.TransportSocket fd=88, family=10, type=2, proto=0, laddr=('::', 50237, 0, 0)>

```

sairon, 2024-05-15:
These are core logs, **host** logs will be needed in this case, please see my previous message on how to get them.

lucode, 2024-05-15:
Had the same issue after upgrading to HAOS 12.3, but it works again now, after upgrading to to HA 2024.5.3 or ev. just another reboot. HA wasn't able to connect to other networked devices.

andreaswemme, 2024-05-15:
@sairon I'm sorry for the misunderstanding of the logs. Here's the HOST logs from one occasion of rebooting the router. I have not a clue what's interesting for you guys so here's the full log. 

```
2024-05-13 19:24:17.794 homeassistant kernel: Linux version 6.6.29-haos (builder@aa45b0ed5448) (x86_64-buildroot-linux-gnu-gcc.br_real (Buildroot -g34e790c5) 12.3.0, GNU ld (GNU Binutils) 2.40) #1 SMP PREEMPT_DYNAMIC Wed May  8 10:12:38 UTC 2024
2024-05-13 19:24:17.794 homeassistant kernel: Command line: BOOT_IMAGE=(hd0,gpt2)/bzImage root=PARTUUID=8d3d53e3-6d49-4c38-8349-aff6859e82fd rootwait zram.enabled=1 zram.num_devices=3 systemd.machine_id=34b86bea3745475b9efb48466fcaf5e1 fsck.repair=yes console=tty1 rauc.slot=A
2024-05-13 19:24:17.794 homeassistant kernel: BIOS-provided physical RAM map:
2024-05-13 19:24:17.794 homeassistant kernel: BIOS-e820: [mem 0x0000000000000000-0x000000000009efff] usable
2024-05-13 19:24:17.794 homeassistant kernel: BIOS-e820: [mem 0x000000000009f000-0x00000000000fffff] reserved
2024-05-13 19:24:17.794 homeassistant kernel: BIOS-e820: [mem 0x0000000000100000-0x00000000b8e83fff] usable
2024-05-13 19:24:17.794 homeassistant kernel: BIOS-e820: [mem 0x00000000b8e84000-0x00000000b8e84fff] ACPI NVS
2024-05-13 19:24:17.794 homeassistant kernel: BIOS-e820: [mem 0x00000000b8e85000-0x00000000b8eb6fff] usable
2024-05-13 19:24:17.794 homeassistant kernel: BIOS-e820: [mem 0x00000000b8eb7000-0x00000000b8eb7fff] reserved
2024-05-13 19:24:17.794 homeassistant kernel: BIOS-e820: [mem 0x00000000b8eb8000-0x00000000ca79afff] usable
2024-05-13 19:24:17.794 homeassistant kernel: BIOS-e820: [mem 0x00000000ca79b000-0x00000000cac9afff] type 20
2024-05-13 19:24:17.794 homeassistant kernel: BIOS-e820: [mem 0x00000000cac9b000-0x00000000cbc7efff] reserved
2024-05-13 19:24:17.794 homeassistant kernel: BIOS-e820: [mem 0x00000000cbc7f000-0x00000000cbe7efff] ACPI NVS
2024-05-13 19:24:17.794 homeassistant kernel: BIOS-e820: [mem 0x00000000cbe7f000-0x00000000cbefefff] ACPI data
2024-05-13 19:24:17.794 homeassistant kernel: BIOS-e820: [mem 0x00000000cbeff000-0x00000000cbefffff] usable
2024-05-13 19:24:17.794 homeassistant kernel: BIOS-e820: [mem 0x00000000cbf00000-0x00000000d47fffff] reserved
2024-05-13 19:24:17.794 homeassistant kernel: BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
2024-05-13 19:24:17.794 homeassistant kernel: BIOS-e820: [mem 0x0000000100000000-0x00000002277fffff] usable
2024-05-13 19:24:17.794 homeassistant kernel: NX (Execute Disable) protection: active
2024-05-13 19:24:17.794 homeassistant kernel: APIC: Static calls initialized
2024-05-13 19:24:17.794 homeassistant kernel: efi: EFI v2.7 by HP
2024-05-13 19:24:17.794 homeassistant kernel: efi: ACPI=0xcbefe000 ACPI 2.0=0xcbefe014 TPMFinalLog=0xcbe0e000 SMBIOS=0xcaf66000 ESRT=0xcaf7a218 MEMATTR=0xc6d61018 
2024-05-13 19:24:17.794 homeassistant kernel: efi: Remove mem62: MMIO range=[0xff000000-0xffffffff] (16MB) from e820 map
2024-05-13 19:24:17.794 homeassistant kernel: e820: remove [mem 0xff000000-0xffffffff] reserved
2024-05-13 19:24:17.794 homeassistant kernel: SMBIOS 3.2 present.
2024-05-13 19:24:17.794 homeassistant kernel: DMI: HP HP ProDesk 400 G6 Desktop Mini PC/871A, BIOS S23 Ver. 02.18.00 12/14/2023
2024-05-13 19:24:17.794 homeassistant kernel: tsc: Detected 2300.000 MHz processor
2024-05-13 19:24:17.794 homeassistant kernel: tsc: Detected 2299.968 MHz TSC
2024-05-13 19:24:17.794 homeassistant kernel: e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
2024-05-13 19:24:17.794 homeassistant kernel: e820: remove [mem 0x000a0000-0x000fffff] usable
2024-05-13 19:24:17.794 homeassistant kernel: last_pfn = 0x227800 max_arch_pfn = 0x400000000
2024-05-13 19:24:17.794 homeassistant kernel: MTRR map: 5 entries (3 fixed + 2 variable; max 23), built from 10 variable MTRRs
2024-05-13 19:24:17.794 homeassistant kernel: x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
2024-05-13 19:24:17.794 homeassistant kernel: last_pfn = 0xcbf00 max_arch_pfn = 0x400000000
2024-05-13 19:24:17.794 homeassistant kernel: esrt: Reserving ESRT space from 0x00000000caf7a218 to 0x00000000caf7a250.
2024-05-13 19:24:17.794 homeassistant kernel: Using GB pages for direct mapping
2024-05-13 19:24:17.794 homeassistant kernel: Secure boot disabled
2024-05-13 19:24:17.794 homeassistant kernel: ACPI: Early table checksum verification disabled
2024-05-13 19:24:17.794 homeassistant kernel: ACPI: RSDP 0x00000000CBEFE014 000024 (v02 HPQOEM)
2024-05-13 19:24:17.794 homeassistant kernel: ACPI: XSDT 0x00000000CBEB0188 0000EC (v01 HPQOEM SLIC-BPC 00000000 INTL 20160422)
2024-05-13 19:24:17.794 homeassistant kernel: ACPI: FACP 0x00000000CBEF5000 000114 (v06 HPQOEM SLIC-BPC 00000000 HP   00000001)
2024-05-13 19:24:17.794 homeassistant kernel: ACPI: DSDT 0x00000000CBED2000 01F251 (v02 HPQOEM 871A     00000000 INTL 20160527)
2024-05-13 19:24:17.794 homeassistant kernel: ACPI: FACS 0x00000000CBE0A000 000040
2024-05-13 19:24:17.794 homeassistant kernel: ACPI: SSDT 0x00000000CBEFC000 00012A (v02 HP     ShmTable 00000001 INTL 20160527)
2024-05-13 19:24:17.794 homeassistant kernel: ACPI: SSDT 0x00000000CBEF9000 0020AD (v02 CpuRef CpuSsdt  00003000 INTL 20160527)
2024-05-13 19:24:17.794 homeassistant kernel: ACPI: SSDT 0x00000000CBEF8000 000DF9 (v02 CtdpB  CtdpB    00001000 INTL 20160527)
2024-05-13 19:24:17.794 homeassistant kernel: ACPI: MSDM 0x00000000CBEF7000 000055 (v03 HPQOEM SLIC-BPC 00000000 HP   00000001)
[... cut ...]
2024-05-13 19:35:04.737 ha systemd-timesyncd[616]: Network configuration changed, trying to establish connection.
2024-05-13 19:35:04.752 ha systemd-timesyncd[616]: Contacted time server 162.159.200.123:123 (time.cloudflare.com).
2024-05-13 19:35:04.752 ha systemd[1]: Started Journal Gateway Service.
2024-05-13 19:39:22.548 ha systemd[1]: Starting Cleanup of Temporary Directories...
2024-05-13 19:39:22.582 ha systemd[1]: systemd-tmpfiles-clean.service: Deactivated successfully.
2024-05-13 19:39:22.582 ha systemd[1]: Finished Cleanup of Temporary Directories.

```

Xyz00777, 2024-05-19:
same problem here.
since 5-6 days my Home Assistant OS installed on Bare Metal cant reconnect to router when the router restarts. i first thinked it would be an addon (deactivated every addon, no its not an addon). than i thought it would be my nic who has a problem, but it was to suspicious that the connection breaks every day at the same time. to mention it, the router is configured like these since around 5 monts with no change. I think it started after the core update to 2024.5.x even after a restore of 2024.4.x its still the same problem. i cant find anything suspicious in the logs. but i also dont know where i should look because on my system, it looks like the loss of network connectivity breaks the system in sutch a way that stuth like the zigbee usb is still working and logging actions from it but other thinks like keyboard intput also breaks at the same time like network.
i would like to help to find out what the problem is but have to restarting my hass system with the power button deletes every log i try to make on my own with scripts... where can i write logs who are still there after an restart?

[hass_os_log.txt](https://github.com/home-assistant/operating-system/files/15367228/hass_os_log.txt)
[hass_os_log_prefiltered.txt](https://github.com/home-assistant/operating-system/files/15367229/hass_os_log_prefiltered.txt)

Xyz00777, 2024-05-21:
I have come across various discussions and issues that seem related to the problem I’m experiencing. A similar issue appears to have been addressed in [PR #3676](https://github.com/home-assistant/supervisor/pull/3676), which fixed network interfaces being disabled at startup.

Additionally, I found related discussions here:

[Home Assistant Reconnection/Restart on WiFi Loss](https://community.home-assistant.io/t/home-assistant-reconnection-restart-on-wifi-loss/578244)
[Home Assistant Not Accessible After Network Reset](https://www.reddit.com/r/homeassistant/comments/14o8vui/home_assistant_not_accesable_after_network_reset/)
[No IP Address After a Reboot](https://community.home-assistant.io/t/no-ip-address-after-a-reboot/411923/5)

KHV8, 2024-05-25:
This might be the same error as the one I have just created, however can not see you  have a console error log?
https://github.com/home-assistant/operating-system/issues/3392

andreaswemme, 2024-05-25:
@KHV8 - Yes we experince exactly the same errors. 

Im unable to connect a screen to my HA-server at the moment but what I'm reading in your report it look exactly the same. And Yes it has been working before.

sairon, 2024-05-27:
@Xyz00777 Thanks for the logs, it's indeed a kernel regression:
```
May 18 06:02:49 homeassistant kernel: BUG: scheduling while atomic: kworker/2:0/161179/0x00000002
May 18 06:02:49 homeassistant kernel: Modules linked in: bnep bluetooth ecdh_generic ecc nft_chain_nat xfrm_user nft_compat nf_tables cfg80211 snd_hda_codec_hdmi snd_hda_codec_realtek snd_hda_codec_generic ledtrig_audio sch_fq_codel i915 snd_hda_intel snd_intel_dspcfg snd_hda_codec x86_pkg_temp_thermal snd_hwdep snd_hda_core coretemp ttm cdc_acm drm_buddy snd_pcm drm_display_helper
May 18 06:02:49 homeassistant kernel: CPU: 2 PID: 161179 Comm: kworker/2:0 Not tainted 6.6.29-haos #1
May 18 06:02:49 homeassistant kernel: Hardware name: LENOVO 10MQS7QB00/3111, BIOS M1AKT58A 06/16/2023
May 18 06:02:49 homeassistant kernel: Workqueue: events e1000_watchdog_task
May 18 06:02:49 homeassistant kernel: Call Trace:
May 18 06:02:49 homeassistant kernel:  <TASK>
May 18 06:02:49 homeassistant kernel:  dump_stack_lvl+0x36/0x50
May 18 06:02:49 homeassistant kernel:  __schedule_bug+0x4d/0x60
May 18 06:02:49 homeassistant kernel:  __schedule+0x797/0x860
May 18 06:02:49 homeassistant kernel:  ? hrtimer_start_range_ns+0x256/0x350
May 18 06:02:49 homeassistant kernel:  schedule+0x56/0xa0
May 18 06:02:49 homeassistant kernel:  schedule_hrtimeout_range_clock+0xb9/0x130
May 18 06:02:49 homeassistant kernel:  ? __pfx_hrtimer_wakeup+0x10/0x10
May 18 06:02:49 homeassistant kernel:  usleep_range_state+0x5f/0x90
May 18 06:02:49 homeassistant kernel:  e1000e_read_phy_reg_mdic+0x7e/0x170
May 18 06:02:49 homeassistant kernel:  e1000e_update_stats+0x50e/0x730
May 18 06:02:49 homeassistant kernel:  e1000_watchdog_task+0xdc/0xa70
May 18 06:02:49 homeassistant kernel:  process_one_work+0x131/0x2f0
May 18 06:02:49 homeassistant kernel:  worker_thread+0x2f0/0x410
May 18 06:02:49 homeassistant kernel:  ? __pfx_worker_thread+0x10/0x10
May 18 06:02:49 homeassistant kernel:  kthread+0xe0/0x110
May 18 06:02:49 homeassistant kernel:  ? __pfx_kthread+0x10/0x10
May 18 06:02:49 homeassistant kernel:  ret_from_fork+0x2c/0x50
May 18 06:02:49 homeassistant kernel:  ? __pfx_kthread+0x10/0x10
May 18 06:02:49 homeassistant kernel:  ret_from_fork_asm+0x1b/0x30
May 18 06:02:49 homeassistant kernel:  </TASK>
```

Seems to be addressed in 6.6.31 kernel by this revert: https://github.com/gregkh/linux/commit/f8a139656c95db893a543159873c57a470d7376d

This means the issue should be addressed in the next OS release. The only solution for now is reverting to 12.2 or switching to the dev channel.

Xyz00777, 2024-06-01:
scheint zu funktionieren ^^
wenn es jemand nach machen könnte, auf den dev channel kann man wechseln mittels `ha supervisor options --channel dev` und halt zum zurück wechseln mit `ha supervisor options --channel stable`
wann kommt das update dann offiziel vermutlich?

Xyz00777, 2024-06-07:
ups sorry looks like i was not wake up enough and answered in german sorry :D

Seems to be working ^^

If someone wants to try it, you can switch to the dev channel using `ha supervisor options --channel dev` and switch back using `ha supervisor options --channel stable`.

When will the update be officially released, presumably?

sairon, 2024-07-09:
Should be fixed in HAOS 12.4.

closed 2024-07-09 by sairon