home-assistant/operating-system #3369: Bluetooth is not working anymore since upgrade to 12.3 from 12.2

briceruzand, 2024-05-14:
### Describe the issue you are experiencing

Since upgrade to 12.3 from 12.2 bluetooth is not working anymore

### What operating system image do you use?

yellow (Home Assistant Yellow)

### What version of Home Assistant Operating System is installed?

12.3

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Upgrade to 12.3 from 12.2

### Anything in the Supervisor logs that might be useful for us?

```txt
2024-05-12 20:25:28.620 WARNING (MainThread) [bluetooth_auto_recovery.recover] Could not determine the power state of the Bluetooth adapter hci0 [00:00:00:00:00:00] due to timeout after 5 seconds
2024-05-12 20:25:49.523 WARNING (MainThread) [bluetooth_auto_recovery.recover] Could not cycle the Bluetooth adapter hci0 [00:00:00:00:00:00]: [Errno 110] Operation timed out
2024-05-12 20:25:49.525 WARNING (MainThread) [bluetooth_auto_recovery.recover] Bluetooth management socket connection lost: [Errno 22] Invalid argument
2024-05-12 20:25:54.525 WARNING (MainThread) [bluetooth_auto_recovery.recover] Could not reset the power state of the Bluetooth adapter hci0 [00:00:00:00:00:00] due to timeout after 5 seconds
2024-05-12 20:25:54.529 WARNING (MainThread) [bluetooth_auto_recovery.recover] Closing Bluetooth adapter hci0 [00:00:00:00:00:00] failed: [Errno 9] Bad file descriptor
```
```

### Anything in the Host logs that might be useful for us?

```txt
2024-05-14 06:39:08.501 homeassistant kernel: Bluetooth: hci0: command 0xfc18 tx timeout
2024-05-14 06:39:08.501 homeassistant kernel: Bluetooth: hci0: BCM: failed to write update baudrate (-110)
2024-05-14 06:39:08.510 homeassistant kernel: Bluetooth: hci0: Failed to set baudrate
2024-05-14 06:39:10.645 homeassistant kernel: Bluetooth: hci0: command 0x1001 tx timeout
2024-05-14 06:39:10.645 homeassistant kernel: Bluetooth: hci0: BCM: Reading local version info failed (-110)
```
```

### Additional information

https://github.com/home-assistant/core/issues/117225
https://github.com/home-assistant/operating-system/issues/3359

sairon, 2024-05-14:
Seems to be similar case as #3207 - adapter ended up in some weird state and can not be reinitialized. Can you try disconnecting your Yellow from the power to perform a "cold" boot and check it then?

briceruzand, 2024-05-14:
Thx @sairon after a "cold" reboot it works well.

closed 2024-05-14 by briceruzand