home-assistant/operating-system #3371: Cannot boot HAOS 12.3 on proxmox kvm

playaz44, 2024-05-14:
### Describe the issue you are experiencing

I copied it from another issue:

> * Exact RPi model (for 3B please distinguish between 3B and 3B+).

i7-7700T HA OS in KVM on proxmox

> * Boot medium HAOS is running from (SD card / SD card + USB data disk / USB drive as system disk).

SSD

> * List of USB peripherals connected.

Only SkyConnect

> * If you can connect your RPi to an HDMI display, do you see the same errors as in the first post in this issue?

No

> * Is the issue reproducible on a minimal fresh install without restoring the backup?

Yes, I installed fresh HAOS on another VM and set up thread, matter, zha, bridge router, silicon labs multiprotocol, then when I tried to configure a new radio for zha, HAOS died

> * Does the problem go away for all affected people here after unplugging the connected USB devices?

No, if HAOS died after trying to configure radio then it can no longer be started

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

12.3

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Update from 12.2 to 12.3
2. Dead 

### Anything in the Supervisor logs that might be useful for us?

```txt
,
```

### Anything in the Host logs that might be useful for us?

```txt
.
```

### System information

i7-7700T HAOS in KVM on proxmox

sairon, 2024-05-14:
It's booting fine, but the Core container does not start. Can you check `ha core logs` in the VM console?

playaz44, 2024-05-14:
> It's booting fine, but the Core container does not start. Can you check `ha core logs` in the VM console?

[image]

I cannot access SSH and send full logs because I forgot that there is no SSH server turned on

sairon, 2024-05-16:
The cause for HA not starting is this error:
```
2024-05-14 20:39:07.797 ERROR (MainThread) [root] Uncaught exception
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/usr/src/homeassistant/homeassistant/__main__.py", line 220, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/src/homeassistant/homeassistant/__main__.py", line 208, in main
    exit_code = runner.run(runtime_conf)
                ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/runner.py", line 190, in run
    return loop.run_until_complete(setup_and_run_hass(runtime_config))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 685, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/runner.py", line 156, in setup_and_run_hass
    hass = await bootstrap.async_setup_hass(runtime_config)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/bootstrap.py", line 280, in async_setup_hass
    await hass.async_add_executor_job(conf_util.process_ha_config_upgrade, hass)
  File "/usr/local/lib/python3.12/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/config.py", line 579, in process_ha_config_upgrade
    if version_obj < AwesomeVersion("0.50"):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/awesomeversion/awesomeversion.py", line 172, in __lt__
    raise AwesomeVersionCompareException(
awesomeversion.exceptions.AwesomeVersionCompareException: Can't compare <unknown sensor:> and <SimpleVer 0.50>
```

Please, report it in the Core repo, as you'll get more targeted help there.

closed 2024-05-16 by github-actions[bot]