home-assistant/operating-system #3373: Update to OS 12.3 causing problems to VLC addon / VLC-Telnet integration

LeoCal, 2024-05-15:
### Describe the issue you are experiencing

After updating to OS 12.3, I started facing issue with VLC addon / VLC Telnet integration.
More specifically, I get an intermittent and chopping sound, which has never occurred to me in several years using VLC.
It seems multiple users are facing a similar issue, as reported to VLC addon as well: https://github.com/home-assistant/addons/issues/3594 

Here's the details of my HA setup:
Core: 2024.5.3
Supervisor: 2024.05.1
Operating System: 12.3
Frontend: 20240501.1

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

Home Assistant OS 12.3

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Upgrade to OS 12.3
2. Try to reproduce a sound through the VLC addon with VLC Telnet integration
3. Notice how sound is chopping and intermittent

### Anything in the Supervisor logs that might be useful for us?

```txt
You can refer to the issue reported to the VLC addon for more logs from me and other users:
https://github.com/home-assistant/addons/issues/3594
```

### Anything in the Host logs that might be useful for us?

```txt
You can refer to the issue reported to the VLC addon for more logs from me and other users:
https://github.com/home-assistant/addons/issues/3594
```

sakirsek, 2024-06-07:
I have this problem too. Still no solution?

LeoCal, 2024-06-07:
AFAIK, unfortunately no solution yet :-(

GunterAv, 2024-06-19:
Hello,

the same with HAOS: 12.4

I found an alternative solution (without VLC)
[https://smarterkram.de/6217/](https://github.com/home-assistant/addons/issues/url) (In German)
Also works with short mp3 files.

Best regards
Gunter

LeoCal, 2024-06-19:
Yes, I also have an MPD based addon that works fine, but ideally this needs to be fixed in VLC addon as that's part of the Home Assistant "official add-ons".

GunterAv, 2024-06-19:
Hello,

I'm right there with you.
But I need the sound output via the RasPi output and didn't want to wait any longer.

Greetings

Grafton1046, 2024-06-27:
Throwing my hat into the ring, it's not just raspberry Pi's.
After the update 12.4 I'm not able to get any sound out of my setup, running baremetal with x86-64

FadilAbdulla, 2024-07-29:
I also have the issue since I updated to 2024.5

rodolfoleo, 2024-08-16:
I had the same problem on 12.3 so I downgraded to 12.2 and the audio was fixed. 

Anybody knows if OS 13 still presents this issue?

GunterAv, 2024-08-16:
in 13.0 the proplem was NOT fixed.

basvdijk, 2024-08-16:
I experience the same on my RPI4 and noticed a very high cpu usage when playing audio or TTS. This might cause the stuttering sound.

bitbytie, 2024-08-18:
I'm also stuck on version 12.2 until this issue is resolved.  
Is there anything I can do to help diagnose the root cause?

sairon, 2024-08-19:
I had an opportunity to look at this issue a bit more in depth, and I'm afraid I don't have good news. The problem indeed happens only for VLC, which, when ran with debug logging, shows it has issues with synchronization and logs `playback way too early ... playing silence` or `playback way too late ... playing silence` couple of times. Playing audio using other tools (like `ffplay`) is fine. I checked and tried adjusting buffer values but it has no effect.

I have confirmed that Raspberry Pi OS has exact the same issue **if PulseAudio and VLC** is used. The problem only happens on latest (for RPi) 6.6 kernel, 6.1 is fine - and 12.2 to 12.3 was right where we updated to Raspberry Pi's current kernel.

Unfortunately, I don't have capacity to pursue this issue, these are the key information that could be used to report the issue upstream in [raspberrypi/linux](https://github.com/raspberrypi/linux/) repo - install PulseAudio (it's not used as the default sound server though) on the latest RPi OS and try playing audio files with VLC. If the cause is found and resolved there, we can apply the fix to HAOS as well.

ClubspeedChris, 2024-09-15:
Running the latest versions and all of my door/alarm sounds now just make weird .2 sec chirps. Really annoying. Is anyone looking at this, anything we can do to help?

NoahJamesTech, 2024-10-04:
+1, this issue is driving me crazy. Is there anything we can do to help? And/or resolve the issue temporarily on our own hardware?

mib1185, 2024-10-04:
As mentioned in https://github.com/home-assistant/operating-system/issues/3373#issuecomment-2297181620, reporting this issue to the [raspberrypi/linux](https://github.com/raspberrypi/linux/) repository by affected users would help to get it fixed upstream, thx :+1:

robertodomotico, 2024-10-08:
I have the same problem with the output on VLC only when I use the piper voice or with some mp3 audio.
The strange thing is that if I use the google voice on the VLC output it works fine.
I don't know if this can give you any ideas but I hope it gets solved

LeoCal, 2024-10-08:
In my case, instead, this is happening also using google TTS on the VLC output, on a Raspberry PI 4.

basvdijk, 2025-01-07:
Anyone tested if the latest HAOS fixed this issue?

NoahJamesTech, 2025-01-07:
I tested with the last major version (I think 14.0) and it still wasn't working. Reverted back to 12.2 where it was working.

basvdijk, 2025-01-07:
@NoahJamesTech Thanks! Looks like we are stuck with 12 or need to find another solution. I have the feeling this is not going to be fixed with future HA updates.

HiTechTLS, 2025-03-10:
Confirming this is still broken on 2025.3.1 & 14.2. Choppy, missing, and broken audio with any TTS used

mib1185, 2025-03-15:
with HAOS 15 we're changing to kernel 6.12 for all systems, except the PI, since 6.12 is not final available (_but [it is near to be](https://forums.raspberrypi.com/viewtopic.php?t=379745)_).
I've done some tests with an rpi4 ...
I'm able to reproduce the issue with the current latest kernel `6.6.74+rpt-rpi-v8` by playing an web audio stream via pullseaudio with vlc and ensure some high load with `dd if=/dev/urandom bs=10M count=1000 of=/dev/null`
Then I've updated to the current test kernel `6.12.18-v8+` and tried to reproduce the issue again, but this time the audio was playing fine all the time, even when under high load 👍 

TL;DR: with upcoming kernel [6.12](https://forums.raspberrypi.com/viewtopic.php?t=379745) the issue should be gone

basvdijk, 2025-06-14:
Thee is still no confirmation that it worked in the latest HA version. Curious to see if anyone can confirm it is fixed.

mib1185, 2025-06-14:
the kernel will be updated to 6.12 also for the RasPI in HA-OS 16 (_#4036_). so with HA-OS 16 the issue will be fixed

sairon, 2025-06-17:
OS 16.0.rc1 is out which contains the latest kernel, can you folks check how it behaves for you? I can hear couple of dropouts when the VLC stream starts but it's fine then, and load is not affecting it.

tvenkataraman, 2025-06-25:
I have updated the OS 16.0.rc1 and Core 2025.6.3 on my RasPI 4. But I am still facing the issue that the chopping sound. 
Earlier I have used 32bit OS for many years on RasPI 3 and I never had such issue.

So, I kindly request you to take a look and fix the issue please.
`Installation method
Home Assistant OS
Core
2025.6.3
Supervisor
2025.06.2
Operating System
16.0.rc1
Frontend
20250531.4`

bitbytie, 2025-07-13:
I've performed a clean installation of Home Assistant OS 16.0 on a Raspberry Pi 4. After that, I installed and enabled both the VLC add-on and the vlc-telnet device. I then uploaded a sample MP3 file (from https://pixabay.com/sound-effects/a320-tritone-chime-104562/).

Unfortunately, playback of this mp3 still suffers from dropouts, as experienced previously.

bitbytie, 2025-07-13:
Speaker is a JBL Flip 2, connected directly to the audio jack on the Pi 4.

basvdijk, 2026-02-07:
Today I had to update since my doorbell was not supported anymore in the old HA builds.

I am now running:

```
Core 2026.2.1
Supervisor 2026.01.1
Operating System 17.0
Frontend 20260128.6
```

VLC-TELNET on the builtin audio jack of my RPI4 was stuttering as expected.

I had an [USB Audioquest DragonFly Black audio device](https://www.audioquest.com/products/dragonfly-black) laying around and tested.

The hardest part for me was to find if the RPI supported the DragonFly and where to set it as audio device.

Settings -> App -> VLC -> Configuration

[image]

This worked like a charm. No more stuttering! 

So chances are that a cheap USB audio from AliExpress works as well. E.g. https://s.click.aliexpress.com/e/_c4Myxxvz

closed 2026-09-03 by github-actions[bot]