home-assistant/operating-system #3374: No sound on HAOS 12.3

macinalycan, 2024-05-15:
### Describe the issue you are experiencing

After updates to:
Core 2024.5.2
Supervisor 2024.05.1
Operating System 12.3
Frontend 20240501.1

there is no sound wiith VLC-TELNET.

After backup to Core 2024.5.1 the problem was still there.
This seems that the problem must come from the operating system 12.3.
So I tried to boot from the previous version (operating system 12.2).

From CLI: ha os boot-slot other

Now everything worked as it should again.

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

12.3

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.Update HAOS 12.2 to 12.3
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
no, because downgrade to 12.2
```

### Anything in the Host logs that might be useful for us?

```txt
no, because downgrade to 12.2
```

### System information

## System Information

version | core-2024.5.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.2
os_name | Linux
os_version | 6.1.73-haos-raspi
arch | aarch64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.2
-- | --
update_channel | stable
supervisor_version | supervisor-2024.05.1
agent_version | 1.6.0
docker_version | 25.0.5
disk_total | 57.8 GB
disk_used | 13.7 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | deCONZ (6.23.0), File editor (5.8.0), AppDaemon (0.16.5), Advanced SSH & Web Terminal (17.3.0), Whisper (2.0.0), VLC (0.3.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 6
-- | --
resources | 3
views | 12
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 11. Mai 2024 um 09:56
-- | --
current_recorder_run | 13. Mai 2024 um 18:30
estimated_db_size | 68.75 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

macinalycan, 2024-05-15:
Update to OS 12.3 causing problems to VLC addon / VLC-Telnet integration #3373

sairon, 2024-05-16:
Let's track it in #3373, as it's essentially the same issue.

closed 2024-05-16 by sairon