home-assistant/operating-system #3375: generic-x86-64 HAOS 12.3 Grub2 Freezing on all Slots since Update from 12.2. (Revert GRUB2 patch causing boot failure on some old x86_64 BIOSes (#3324))

MarvinMynx, 2024-05-15:
### Describe the issue you are experiencing

I'm not able to boot HomeAssistant OS After performing exactly this steps. I'm stuck with Grub2 freezing exactly like after applying the Update via HomeAssistant GUI.
https://www.home-assistant.io/installation/generic-x86-64#to-install-haos-via-ubuntu-from-a-usb-flash-drive

I’ve managed to successfully instal HAOS 12.2 and restore from backup, completed updates, and rebooted all systems, with the exception of HAOS 12.3. Everything appears to be functioning as normal again.

Regarding hardware, I'm utilizing this thin client: [Fujitsu ESPRIMO Q920](https://sp.ts.fujitsu.com/dmsp/Publications/public/ds-ESPRIMO-Q920.pdf). If there are any further details needed, please let me know. Thanks!

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

12.2 (Updating to 12.3 will cause boot failure)

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Install generic-x86-64 HAOS 12.2 with Prefered Method 1 Installation Methode (Ubuntu GUI)
2. Upgrade HAOS to 12.3
3. Reboot
...

### Anything in the Supervisor logs that might be useful for us?

```txt
[32m2024-05-15 15:43:50.571 INFO (MainThread) [supervisor.api.middleware.security] /core/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-05-15 15:43:50.575 INFO (MainThread) [supervisor.api.middleware.security] /supervisor/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-05-15 15:43:50.579 INFO (MainThread) [supervisor.api.middleware.security] /mounts access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-05-15 15:43:50.582 INFO (MainThread) [supervisor.api.middleware.security] /addons access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-05-15 15:44:30.698 INFO (MainThread) [supervisor.api.middleware.security] /supervisor/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-05-15 15:44:30.702 INFO (MainThread) [supervisor.api.middleware.security] /backups access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-05-15 15:44:30.719 INFO (MainThread) [supervisor.api.middleware.security] /core/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-05-15 15:44:30.722 INFO (MainThread) [supervisor.api.middleware.security] /supervisor/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-05-15 15:44:30.726 INFO (MainThread) [supervisor.api.middleware.security] /mounts access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-05-15 15:44:30.729 INFO (MainThread) [supervisor.api.middleware.security] /addons access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-05-15 15:45:50.957 INFO (MainThread) [supervisor.api.middleware.security] /supervisor/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-05-15 15:45:50.961 INFO (MainThread) [supervisor.api.middleware.security] /backups access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-05-15 15:45:51.613 INFO (MainThread) [supervisor.api.middleware.security] /core/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-05-15 15:45:51.616 INFO (MainThread) [supervisor.api.middleware.security] /supervisor/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-05-15 15:45:51.620 INFO (MainThread) [supervisor.api.middleware.security] /mounts access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-05-15 15:45:51.623 INFO (MainThread) [supervisor.api.middleware.security] /addons access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-05-15 15:48:32.060 INFO (MainThread) [supervisor.api.middleware.security] /supervisor/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-05-15 15:48:32.065 INFO (MainThread) [supervisor.api.middleware.security] /backups access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-05-15 15:48:32.082 INFO (MainThread) [supervisor.api.middleware.security] /core/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-05-15 15:48:32.085 INFO (MainThread) [supervisor.api.middleware.security] /supervisor/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-05-15 15:48:32.089 INFO (MainThread) [supervisor.api.middleware.security] /mounts access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-05-15 15:48:32.092 INFO (MainThread) [supervisor.api.middleware.security] /addons access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-05-15 15:53:52.461 INFO (MainThread) [supervisor.api.middleware.security] /supervisor/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-05-15 15:53:52.466 INFO (MainThread) [supervisor.api.middleware.security] /backups access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-05-15 15:53:52.482 INFO (MainThread) [supervisor.api.middleware.security] /core/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-05-15 15:53:52.485 INFO (MainThread) [supervisor.api.middleware.security] /supervisor/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-05-15 15:53:52.489 INFO (MainThread) [supervisor.api.middleware.security] /mounts access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-05-15 15:53:52.492 INFO (MainThread) [supervisor.api.middleware.security] /addons access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-05-15 16:03:10.677 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token[0m
[32m2024-05-15 16:04:32.725 INFO (MainThread) [supervisor.api.middleware.security] /supervisor/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-05-15 16:04:32.729 INFO (MainThread) [supervisor.api.middleware.security] /backups access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-05-15 16:04:32.746 INFO (MainThread) [supervisor.api.middleware.security] /core/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-05-15 16:04:32.749 INFO (MainThread) [supervisor.api.middleware.security] /supervisor/info access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-05-15 16:04:32.753 INFO (MainThread) [supervisor.api.middleware.security] /mounts access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-05-15 16:04:32.756 INFO (MainThread) [supervisor.api.middleware.security] /addons access from cebe7a76_hassio_google_drive_backup[0m
[32m2024-05-15 16:13:40.628 INFO (MainThread) [supervisor.hardware.monitor] Detecting remove hardware /dev/input/event11 - /dev/input/by-id/usb-Logitech_USB_Laser_Mouse-event-mouse[0m
[32m2024-05-15 16:13:40.633 INFO (MainThread) [supervisor.hardware.monitor] Detecting remove hardware /dev/hidraw2 - None[0m
[32m2024-05-15 16:13:40.633 INFO (MainThread) [supervisor.hardware.monitor] Detecting remove hardware /dev/bus/usb/001/007 - None[0m
[32m2024-05-15 16:13:47.242 INFO (MainThread) [supervisor.hardware.monitor] Detecting remove hardware /dev/input/event8 - /dev/input/by-id/usb-Dell_Dell_Multimedia_Pro_Keyboard-event-kbd[0m
[... 200 lines cut ...]
Home Assistant Supervisor
Host Operating System	Home Assistant OS 12.2
Update Channel	stable
Supervisor Version	supervisor-2024.05.1
Agent Version	1.6.0
Docker Version	25.0.5
Disk Total	234.0 GB
Disk Used	14.0 GB
Healthy	true
Supported	true
Board	generic-x86-64
Supervisor API	ok
Version API	ok
Installed Add-ons	Advanced SSH & Web Terminal (17.3.0), Studio Code Server (5.15.0), Home Assistant Google Drive Backup (0.112.1), Mosquitto broker (6.4.0), Zigbee2MQTT (1.37.1-1), Grafana (10.0.0), InfluxDB (5.0.0), Spotify Connect (0.13.0), ESPHome (2024.5.0), Local VLC (18)
Dashboards
Dashboards	3
Resources	0
Views	2
Mode	storage
Recorder
Oldest Run Start Time	May 1, 2024 at 19:27
Current Run Start Time	May 15, 2024 at 15:33
Estimated Database Size (MiB)	330.27 MiB
Database Engine	sqlite
Database Version	3.44.2

coc, 2024-05-15:
When I install 12.3 on a Thin Client Lenovo ThinkCentre M93 p and restore a full backup, File Editor and Advanced HSS & Terminal cannot be started. Also, „os Info“ show Status=bad on both partitions.
When I update to 12.2 on ha console, restore the backup, and update to 12.3 from UI, all works perfect.
I also had the problem with grub updates before …

sairon, 2024-05-15:
@MarvinMynx Thanks for the detailed report, there's already recent issue reporting this problem on Esprimo Q920, so let's stick to that one: #3348

@coc That sounds like something completely different (and so far not reported). With the GRUB incompatibility, the OS fails to load completely. Please open another issue and don't forget to attach host and Supervisor logs - that should reveal why the add-ons fail to start.

closed 2024-05-15 by sairon

MarvinMynx, 2024-06-19:
With the current POS Version i have the Same Error Again...