home-assistant/operating-system #3387: Ethernet adapter Motorcomm YT6801 Gigabit Ethernet Adapter support in HAOS

BandBxx, 2024-05-21:
### Describe the issue you are experiencing

HAOS does not recognize the Motorcomm YT6801 Gigabit Ethernet Adapter upon boot. Drivers are not integrated into the kernel.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

12.3

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.  Install HaOS on x86 platform mini-PC  Soyo M2 
2.  I get an error when loading and HAOS goes into protected mode at the CLI startup stage

### Anything in the Supervisor logs that might be useful for us?

```txt
No logs, because I didn't even start because of missing Ethernet
```

### Anything in the Host logs that might be useful for us?

```txt
Quite a popular Chinese budget miniPC with a new Ethernet adapter.
I found drivers for Windows, they work great in Win. Please add driver support to the HAOS kernel.
There are versions with different Ethernet adapters; previously I came across a Realtec-adapter.
Its my first encountered Motorcomm YT6801 Gigabit Ethernet Adapter

https://www.drvhub.net/devices/network/motorcomm/yt6801-gigabit-ethernet-adapter
```

### Additional information

Soyo M2 miniPC on Ali:   
https://aliexpress.ru/item/1005006416944353.html

BandBxx, 2024-05-21:
I found drivers for linux here:
https://www.touslesdrivers.com/index.php?v_page=23&v_code=77568&v_langue=en

sairon, 2024-05-22:
I am sorry, but I'll disappoint you here. With some very rare exceptions, HAOS only uses mainline Linux drivers. And even in those exceptions, the drivers have proper open source license, so there is no legal compliance issue when including it in our project. Which is not the case here, as indicated in the source files:
```
Copyright (c) 2021 Motorcomm, Inc.
Motorcomm Confidential and Proprietary.

This is Motorcomm NIC driver relevant files. Please don't copy, modify,
distribute without commercial permission.
```

dante1613, 2024-05-22:
I have [MLLSE M2](https://aliexpress.com/item/1005006510890816.html?) with the same controller, I managed to launch HAOS only through Proxmox. My [guide for Proxmox](https://github.com/dante1613/Motorcomm-YT6801/blob/e45f2fca4d8bac6445d3ed98b2973b7c1e42eb35/Proxmox%20-%20instruction.md)

BandBxx, 2024-05-22:
> I am sorry, but I'll disappoint you here. With some very rare exceptions, HAOS only uses mainline Linux drivers. And even in those exceptions, the drivers have proper open source license, so there [...]

Sad to hear this :( I didn’t realize that drivers respected copyrights. Device manufacturers should be interested in making sure their devices are used everywhere. Ok, for now I’ll use a USB ethernet adapter.

BandBxx, 2024-05-22:
> I have [MLLSE M2](https://aliexpress.com/item/1005006510890816.html?) with the same controller, I managed to launch HAOS only through Proxmox. My [guide for Proxmox](https://github.com/dante1613/Motorcomm-YT6801/blob/e45f2fca4d8bac6445d3ed98b2973b7c1e42eb35/Proxmox%20-%20instruction.md) install driver

Thank you, good instructions, but I would not like to use a Virtual Machine.  Native HA OS onl

closed 2024-05-23 by sairon

danielponte, 2024-07-09:
I'm in this very same boat. Darn it. Is there a way to use WiFi then?

buckwem, 2024-11-08:
This Motorcomm YT6801 Gigabit Ethernet Adapter is also used in the GEEKOM Mini Air12 Lite Mini PC and therefore only wireless works. Without mainline support for the driver there is no gaurantee of support going forward (even if I could find a driver), I am returning the NUC. 

I am noting it here for anyone who performs a Google search.

danielponte, 2024-11-08:
A lot of mini PCs are using this nic, they should consider adding the
driver to HAOS

Daniel Ponte

buckwem, 2024-11-08:
@danielponte if you are still looking - see my comment here[ ](https://community.home-assistant.io/t/problems-with-lan-adapter-geekom-mini-pc-air12-lite/792089/2)

tobzsc, 2024-12-05:
Seems like there was a commit to the mainline kernel a few days ago: https://lore.kernel.org/netdev/9506d808-85b9-4f37-baee-76f3dee56182@lunn.ch/T/

danielponte, 2024-12-05:
Amazing news! As soon as haos updates its kernel I guess well have native
support to this nic, right?

Daniel Ponte