home-assistant/operating-system #3390: Home Assistant OS 12.3 breaks ZHA on VM

capmlbpkel, 2024-05-23:
### Describe the issue you are experiencing

After upgrade to 12.3, SkyConnect/ZHA is no longer working.
Reverting the upgrade and restoring pre-12.3 backup does not solve the problem.

Seems to be very similar to https://github.com/home-assistant/operating-system/issues/3352. Maybe related to https://github.com/home-assistant/core/issues/117720.

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

12.3

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Upgrade from 12.2 to 12.3

### Anything in the Supervisor logs that might be useful for us?

```txt
no
```

### Anything in the Host logs that might be useful for us?

```txt
no
```

### System information

Core version: 2024.4.4
Supervisor version: 2024.05.1

### Additional information

I cannot find it anymore, but I remember some entries in the Core log similar to

> [homeassistant.components.zha] Failed to set up ZHA

NodeJSmith, 2024-05-23:
I'm having an issue on 12.3 in a Proxmox vm with Z2M. I'll add logs later, will have to spin up a separate vm to let break so I can get them.

sairon, 2024-05-23:
Sorry, but we can't move forward without any logs. Even if you think there's nothing useful, there could be something. Could you please reproduce the issue again and get the Host logs, and possibly also Core logs that would show why exactly the ZHA setup failed? You can use `ha host boot-slot other` to quickly swap back and forth between OS versions.

capmlbpkel, 2024-05-27:
I just learned how to enable debug logging and was able to examine the logs in more detail. As it turns out, the SkyConnect usb device was not found on the HA host at all.

Deleting and re-adding the USB host device via VM management solved the issue.

closed 2024-05-27 by capmlbpkel

walterbressers, 2024-06-16:
The issue that my Conbee II stick (which was hardware wise recognised within HA) was not working on Proxmox VM, it failed to initialise/connect, was solved with Core update 2024.6.3, probably something to do with ZHA dependancy zigpy?
https://www.home-assistant.io/blog/2024/06/05/release-20246/#202463---june-15