home-assistant/operating-system #3393: Installation process failure using Raspberry PI imager

phiGitHub, 2024-05-26:
### Describe the issue you are experiencing

Installation process failure using Raspberry PI imager.
File error ?
[image]
[image]

### What operating system image do you use?

rpi5-64  (Raspberry Pi 5 64-bit OS)

### What version of Home Assistant Operating System is installed?

None - Setup file download error

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

using Raspberry PI imager as per "Install Home Assistant Operating System" tutorial

### Anything in the Supervisor logs that might be useful for us?

```txt
no logs, see screen captures here above
```

### Anything in the Host logs that might be useful for us?

```txt
no logs, see screen captures here above
```

sairon, 2024-05-27:
This is not an HAOS issue, and some other people ran into it too, see here: https://forums.raspberrypi.com/viewtopic.php?t=367859

It may be fixed in macOS 14.5 as reported in a different issue but likely with the same root cause: https://github.com/Homebrew/homebrew-core/issues/158759#issuecomment-2123135458

Other way around would be to use a different tool or system for flashing the image. Nothing we can about it, sorry.

closed 2024-05-27 by sairon

phiGitHub, 2024-05-27:
fixed with Github direct download
thanks for swift reply & best regards