home-assistant/operating-system #3395: HP T620 doesn't boot on HA OS 12.3

TheFes, 2024-05-30:
### Describe the issue you are experiencing

System doesn't boot, screen only shows this.
[image]

After a reboot of the system, it boots in 12.2

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

6.6.25-haos

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Update from 12.2 to 12.3

### Anything in the Supervisor logs that might be useful for us?

```txt
Can't reach logs, system doesn't boot
```

### Anything in the Host logs that might be useful for us?

```txt
Can't reach logs, system doesn't boot
```

### System information

```
## System Information

version | core-2024.6.0b1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.2
os_name | Linux
os_version | 6.6.25-haos
arch | x86_64
timezone | Europe/Amsterdam
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4989
Installed Version | 1.34.0
Stage | running
Available Repositories | 1401
Downloaded Repositories | 34
HACS Data | ok

</details>

<details><summary>AccuWeather</summary>

error | failed to load: unknown
-- | --

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | April 24, 2025 at 02:00
relayer_connected | true
relayer_region | eu-central-1
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
remote_server | eu-central-1-13.ui.nabu.casa
certificate_status | ready
instance_id | c98f9013fe06431b9ceb8293608b11bf
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.2
-- | --
update_channel | beta
supervisor_version | supervisor-2024.05.2
agent_version | 1.6.0
docker_version | 25.0.5
disk_total | 116.7 GB
disk_used | 17.5 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Samba share (12.3.1), Studio Code Server (5.15.0), Terminal & SSH (9.14.0), ESPHome (2024.5.4), Home Assistant Google Drive Backup (0.112.1), AdGuard Home (5.1.0), Apache2 (2.8.2), DbStats (0.5.6), ESPHome (beta) (2024.5.4), File editor (5.8.0), MQTT Explorer (browser-1.0.3), MariaDB (2.7.1), Mosquitto broker (6.4.1), Nginx Proxy Manager (1.0.1), RPC Shutdown (2.5), SQLite Web (4.1.2), VLC (0.3.0), Zigbee2MQTT (1.37.1-1), ha-sip (3.7), openWakeWord (1.10.0), phpMyAdmin (0.9.1), porcupine1 (1.2.0), ZigStar TI CC2652P/P7 FW Flasher (0.4.0), Music Assistant (2.0.4)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 20
views | 28
mode | yaml

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | May 27, 2024 at 08:11
-- | --
current_recorder_run | May 30, 2024 at 11:37
estimated_db_size | 1048.83 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>
```

sairon, 2024-06-12:
There [was an issue](https://github.com/home-assistant/operating-system/issues/3305#issuecomment-2067699632) with T620 after the GRUB update to 2.12 but that could be resolved using a BIOS update. Can you check what BIOS version are you on?

In 12.3 we reverted a GRUB commit that  triggered boot failues on Fujitsu Esprimo Q920 (#3348) so I wonder if it could be related somehow. However, your case seems to be different, if I understand it correctly, you updated to 12.3 (which means the boot partition now contains the latest GRUB with the revert patch) and it only boots the old 12.2 entry?

TheFes, 2024-07-01:
My issue seems similar as #3425 but I already had it on 12.3

```
[core-ssh ~]$ ha os info
board: generic-x86-64
boot: A
boot_slots:
  A:
    state: booted
    status: good
    version: null
  B:
    state: inactive
    status: bad
    version: "12.4"
data_disk: PSSBN128GA87BC0-B59507980083401558877
update_available: true
version: "12.4"
version_latest: "12.4"
```

TheFes, 2024-07-01:
I must admit I didn't try myself, but the person who sold me the T620 has more of them, and tried updating the BIOS. For him it still didn't work.

driversti, 2024-07-11:
Another T620, the same issue. There are two slots (A/B), and I need to swap between them to boot HAOS. It is so annoying! I cannot update the OS while I'm not at home; otherwise, I won't be able to monitor my house.

JanHDijkstra, 2024-07-16:
Same issue with a T620. 
Need reboot 4 times to get it up and running again into not updated version.
Cannot update anymore.

TheFes, 2024-08-07:
The PR to revert the previous GRUB commit (#3470) in 13.0rc1 seem to have resolved this issue. 
This issue was not linked, but I will close it now :)

Thanks!

closed 2024-08-07 by TheFes

sairon, 2024-08-07:
@TheFes That's good news, thank you for the information! I tried to get in touch with you in a Discord DM to test it before it made its way to the release but I don't need to bother you with that anymore :)

TheFes, 2024-08-07:
Oh, must have missed that (or ignored it because I thought it was a community member looking for private support). Sorry about that :)

sjanssen15, 2024-08-17:
Just tested a T610 (older model) which had the same issue (GRUB issue after update). Reinstall with HAOS 13.0 fixed the issue on this model