home-assistant/operating-system #3396: problem with network manager "connection.autoconnect-ports: unknown property"

Stan-Gobien, 2024-05-31:
### Describe the issue you are experiencing

I'm trying to change the route-metric on the 2 NIC's I have in HomeAssistant because it's using the wrong one as default outgoing route.

Removing the DNS server or the gateway doesn't work in the UI, you can't save without them.
So i'm trying the change route-metric using the nmcli.

However nmcli is broken.
`nmcli connection modify "Supervisor enp2s4" ipv4.route-metric 103
Warning: nmcli (1.46.0) and NetworkManager (1.44.2) versions don't match. Restarting NetworkManager is advised.
Error: Failed to modify connection 'Supervisor enp2s4': connection.autoconnect-ports: unknown property
`

For now I have used ha network info to set the gateway address to it's own IP address, that seems to have helped setting the other interface default with a lower/preferred metric. 

using nmcli should work and it's broken it seems.

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

12.3

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Edit any setting using nmcli
For instance `nmcli connection modify "Supervisor enp2s4" ipv4.route-metric 103`

> Warning: nmcli (1.46.0) and NetworkManager (1.44.2) versions don't match. Restarting NetworkManager is advised.
Error: Failed to modify connection 'Supervisor enp2s4': connection.autoconnect-ports: unknown property

### Anything in the Supervisor logs that might be useful for us?

```txt
Nothing
```

### Anything in the Host logs that might be useful for us?

```txt
Nothing
```

bangcrash, 2024-06-02:
I have come across the same problem while trying to add a VLAN connection to my HA server.
The command that I used is:
`nmcli con add type vlan con-name eno1@vlan30 dev eno1 id 30 ip4 10.xx.xx.20/24 ipv4.dns 10.xx.xx.1 gw4 10.xx.xx.1`

The error given was:
`Warning: nmcli (1.46.0) and NetworkManager (1.44.2) versions don't match. Restarting NetworkManager is advised.
Error: Failed to add 'eno1@vlan30' connection: connection.autoconnect-ports: unknown property`

I'm running HASSOS 12.3 on an Intel NUC and HA 2024.5.5 with Supervisor 2024.05.1

closed 2024-06-12 by sairon

manuxio, 2024-06-18:
Please allow to add different routes.
As far as I can tell, this is not possible via "ha network" ... thank you!

victorclaessen, 2024-11-18:
I know using `nmcli` is no longer supported or advised. I just wanted to leave this here for any unfortunate soul that ended up in the same predicament as me.

A few years ago I used the `nmcli` command to add a persistent route to my HA installation. Today, I wanted to remove this persistent route and found out I couldn't because PR#761 removed networkmanager-cli. So I was stuck.

To fix it, I did this in the Advanced SSH & Web Terminal:
```
cd /tmp
wget http://dl-cdn.alpinelinux.org/alpine/v3.19/community/x86_64/libnm-1.44.4-r0.apk
wget http://dl-cdn.alpinelinux.org/alpine/v3.19/community/x86_64/networkmanager-1.44.4-r0.apk
wget http://dl-cdn.alpinelinux.org/alpine/v3.19/community/x86_64/networkmanager-cli-1.44.4-r0.apk
wget http://dl-cdn.alpinelinux.org/alpine/v3.19/community/x86_64/networkmanager-common-1.44.4-r0.apk
apk add networkmanager-1.44.4-r0.apk networkmanager-cli-1.44.4-r0.apk networkmanager-common-1.44.4-r0.apk libnm-1.44.4-r0.apk
nmcli con modify "HassOS default" -ipv4.routes "172.22.100.0/24 192.168.29.9"
```
I still got a warning about mismatched versions 1.44.4 vs 1.44.2, but now at least there was no unknown property stopping the command.

Hope that helps someone.