home-assistant/operating-system #3425: HP T620 not booting 12.4

brugk000, 2024-06-18:
### Describe the issue you are experiencing

After updating 12.3 to 12.4 the machine does not boot.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

12.3

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

## System Information

version | core-2024.6.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.2
os_name | Linux
os_version | 6.6.29-haos
arch | x86_64
timezone | Europe/Amsterdam
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4895
Installed Version | 1.34.0
Stage | running
Available Repositories | 1456
Downloaded Repositories | 18

</details>

<details><summary>AccuWeather</summary>

can_reach_server | ok
-- | --
remaining_requests | 39

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 24 augustus 2024 om 02:00
relayer_connected | true
relayer_region | eu-central-1
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
remote_server | eu-central-1-8.ui.nabu.casa
certificate_status | ready
instance_id | 84f883c958844a7eab4d8537d621d2e1
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.3
-- | --
update_channel | stable
supervisor_version | supervisor-2024.06.0
agent_version | 1.6.0
docker_version | 25.0.5
disk_total | 228.5 GB
disk_used | 28.3 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Home Assistant Google Drive Backup (0.112.1), File editor (5.8.0), Samba share (12.3.1), Terminal & SSH (9.14.0), Mosquitto broker (6.4.1), Glances (0.21.1), Zigbee2MQTT (1.38.0-1), Matter Server (6.1.2), Studio Code Server (5.15.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 11
views | 13
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 4 mei 2024 om 18:17
-- | --
current_recorder_run | 18 juni 2024 om 19:59
estimated_db_size | 14696.54 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

<details><summary>Solcast PV Forecast</summary>

can_reach_server | ok
-- | --
used_requests | 44
rooftop_site_count | 2

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

### Steps to reproduce the issue

1. Updated to 12.4
2. System restarted
3. Hard reset the machine
4.  Booted 12.3
...

### Anything in the Supervisor logs that might be useful for us?

```txt
none
```

### Anything in the Host logs that might be useful for us?

```txt
none
```

### Additional information

[image]

brugk000, 2024-06-18:
Additional: the instance HAOS as vm on Unraid works as expected

spitsy1, 2024-06-18:
Having the same problem here. Can someone tell me the steps to follow-up to fix this? I'm quite noob at this.

Thanks in advance.

markusg1234, 2024-06-19:
My HA went to onboarding after upgrade from 12.3 to 12.4, generic-x86-64.

spitsy1, 2024-06-19:
Does anyone know where i can get help to get my HA working again? I cannot reach it anymore, and really don't know where to begin. I've searched over the internet everywhere. Please help me, otherwise i can just throw the T620 in the garbage.

sairon, 2024-06-19:
Normally I would assume it's the same GRUB issue that is affecting Fujitsu Q920 - in #3395 I had the suspicion that T620 might be affected as well, but now that you say you were running 12.3 without issues, it really doesn't make sense - there were no GRUB changes between these two releases :thinking: Can you try attaching a display and checking where the boot process hangs?

sairon, 2024-06-19:
@spitsy1 If it fails while loading the system, it should boot to the previous slot after 3 failed boots, so rebooting it couple of times might get you back on track. But just as above, best would be attaching a display and checking what is wrong.

brugk000, 2024-06-19:
Will do that. Thanks

spitsy1, 2024-06-19:
> @spitsy1 If it fails while loading the system, it should boot to the previous slot after 3 failed boots, so rebooting it couple of times might get you back on track. But just as above, best would be attaching a display and checking what is wrong.

I did reboot it 4 times and now its back online. I think i just skip the update to 12.4 for now. 

Thanks a lot.

brugk000, 2024-06-19:
Solved the problem. First I updated the T620 bios to the latest version. The I booted and forced the boot via slot A. Now it works with 12.4 os

VenominousX, 2024-06-21:
Hi,
same problem here. In my case HA does not get a proper network configuration: 

`2024-06-21 04:11:46.096 ERROR (MainThread) [async_upnp_client.ssdp] Received error: [Errno 101] Network unreachable, transport: <_SelectorDatagramTransport fd=32 read=polling write=<idle, bufsize=0>>, socket: <asyncio.TransportSocket fd=32, family=2, type=2, proto=0, laddr=('0.0.0.0', 49442)>`

`2024-06-21 04:11:46.097 ERROR (MainThread) [async_upnp_client.ssdp] Received error: [Errno 99] Address not available, transport: <_SelectorDatagramTransport fd=57 read=polling write=<idle, bufsize=0>>, socket: <asyncio.TransportSocket fd=57, family=10, type=2, proto=0, laddr=('::', 55125, 0, 0)>`

Then some integrations fail to start and blocking the boot process.

My System:
Ryzen 8845HS, DDR5 RAM, PCi 4.0 SSD, 2.5Gb Ethernet

dsoveen, 2024-06-25:
> Solved the problem. First I updated the T620 bios to the latest version. The I booted and forced the boot via slot A. Now it works with 12.4 os.
> 
> New problem: I lost all my units on the sensors. Reverted to 12.3 

How did you upgrade bios? I own a t620 as well and have invested serious (google) time and effort trying to accomplish this. Didn't work out for me. Any help (outside this channel) would be appreciated!

mmirannalti, 2024-06-28:
I had the same issue after updating to 12.4. Bare metal on a nuc 7th gen.

Switched back to 12.3 boot slot and restored backup.

brugk000, 2024-06-30:
> How did you upgrade bios? I own a t620 as well and have invested serious (google) time and effort trying to accomplish this. Didn't work out for me. Any help (outside this channel) would be [...]

Take a look at this: https://support.hp.com/nl-nl/drivers/hp-t620-flexible-thin-client/5404706
Download sp115305.exe and generate a usb recovery disk. Boot the T620 with the disk inserterd

TheFes, 2024-07-01:
For me it already didn't work on 12.3, also see https://github.com/home-assistant/operating-system/issues/3395

sairon, 2024-08-07:
Per https://github.com/home-assistant/operating-system/issues/3425#issuecomment-2178517152 this original author's issue was solved. Changes in 13.0.rc1 should address other known issues with HP T620.

closed 2024-08-07 by sairon