home-assistant/operating-system #3426: USB not working on yellow after update to HAOS 12.4

bkbartk, 2024-06-18:
### Describe the issue you are experiencing

I just updated my HA yellow to HAOS 12.4 and neither of my 2 usb devices are working
those are
1. gsm module
2. sonoff zigbee stick.
I forgot to greb any logs, and already reverted to 12.3 where all works well.

I still want to mention because I would like to update in the future.

### What operating system image do you use?

yellow (Home Assistant Yellow)

### What version of Home Assistant Operating System is installed?

12.4

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

ha yellow with nvme:
-Sonoff Zigbee 3.0 USB Dongle Plus - Sonoff Zigbee 3.0 USB Dongle Plus
- SIM800C USB GSM GPRS Wireless Module Quad-Band

### Steps to reproduce the issue

update your yellow to HAOS 12.4

### Anything in the Supervisor logs that might be useful for us?

```txt
-
```

### Anything in the Host logs that might be useful for us?

```txt
-
```

### System information

## System Information

version | core-2024.6.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.2
os_name | Linux
os_version | 6.6.28-haos-raspi
arch | aarch64
timezone | Europe/Amsterdam
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.34.0
Stage | running
Available Repositories | 1394
Downloaded Repositories | 15
HACS Data | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.3
-- | --
update_channel | stable
supervisor_version | supervisor-2024.06.0
agent_version | 1.6.0
docker_version | 25.0.5
disk_total | 916.9 GB
disk_used | 299.2 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | yellow
supervisor_api | ok
version_api | ok
installed_addons | Duck DNS (1.17.0), ESPHome (2024.5.5), Home Assistant Google Drive Backup (0.112.1), Mosquitto broker (6.4.1), SQLite Web (4.1.2), Studio Code Server (5.15.0), Terminal & SSH (9.14.0), Vaultwarden (Bitwarden) (0.21.2), chrony (3.0.1), Advanced SSH & Web Terminal (18.0.0), Samba share (12.3.1), Whisper (2.1.0), Piper (1.5.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 8
-- | --
resources | 6
views | 16
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | June 10, 2024 at 17:54
-- | --
current_recorder_run | June 18, 2024 at 22:00
estimated_db_size | 1607.32 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

sairon, 2024-06-19:
This looks like #3347 which was already reported for 12.3. Can you check 12.4 again and if the devices don't appear, try rebooting it (or even cold booting) again?

bkbartk, 2024-06-19:
Hello,
thanks for the replay, there are 2 reasons why I believed this was a different issues.
3347 also mentions "Also preventing boot" I do not experience that,
and this issue was about 12.3

the strange thing now is,
I just updated again, and now both USB devices are working without an additional restart. So if USB sometimes doesn't work, the version argument for this being a different issue is invalid.

sairon, 2024-06-19:
It really doesn't prevent boot - at least it did not in my case where I was able to reproduce it - however, some things may fail to initialize and the system may be unreachable. The issue is intermittent, in my case it was rather rare but the original reporter said it happens more often to him. Unfortunately we still don't know how to reproduce it reliably, so it's hard to debug it.

bkbartk, 2024-06-19:
I that case I would assume it's the same as for
https://github.com/home-assistant/operating-system/issues/3347
I will close this one as a duplicate,
if I have any more info I will add it to the other one.

closed 2024-06-19 by bkbartk