home-assistant/operating-system #3427: 12.3 to 12.4 not working.

showupch, 2024-06-18:
### Describe the issue you are experiencing

Try to install the 12.4 . All looks normal. After rebooting the system show gain Home assistant core update available.

### What operating system image do you use?

rpi5-64  (Raspberry Pi 5 64-bit OS)

### What version of Home Assistant Operating System is installed?

Home Assistant OS 12.3

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Pi5 8GB. Nvme. 

### Steps to reproduce the issue

1. click install
2.system reboot
3.system show home assistant update available ( installed version 12.3 )
...

### Anything in the Supervisor logs that might be useful for us?

```txt
No errors
```

### Anything in the Host logs that might be useful for us?

```txt
No Errors
```

### System information

## System Information

version | core-2024.6.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.2
os_name | Linux
os_version | 6.6.28-haos-raspi
arch | aarch64
timezone | Europe/Zurich
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4646
Installed Version | 1.34.0
Stage | running
Available Repositories | 1391
Downloaded Repositories | 10

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | May 2, 2025 at 02:00
relayer_connected | true
relayer_region | eu-central-1
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | true
remote_server | eu-central-1-18.ui.nabu.casa
certificate_status | ready

can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.3
-- | --
update_channel | stable
supervisor_version | supervisor-2024.06.0
agent_version | 1.6.0
docker_version | 25.0.5
disk_total | 916.2 GB
disk_used | 11.1 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | rpi5-64
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.8.0), Terminal & SSH (9.14.0), eWeLink Smart Home (1.4.3), Mosquitto broker (6.4.1), SQLite Web (4.1.2), InfluxDB (5.0.0), Grafana (10.0.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 6
-- | --
resources | 5
views | 12
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | June 15, 2024 at 05:14
-- | --
current_recorder_run | June 18, 2024 at 22:18
estimated_db_size | 847.83 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

sairon, 2024-06-19:
Can you check if `ha host logs -n1000 -b-1` (or `-b-2`, `-b-3`, etc.) show any logs from the previous failed boots that would indicate the reason why it failed? Alternatively, can you connect a display to the Pi and check where it fails during booting?

showupch, 2024-06-19:
> Can you check if `ha host logs -n1000 -b-1` (or `-b-2`, `-b-3`, etc.) show any logs from the previous failed boots that would indicate the reason why it failed? Alternatively, can you connect a display to the Pi and check where it fails during booting?

Hi sairon. The are no error logs nowhere ( b1 b2 b3 ) I will try to add a monitor and I report back ASAP

showupch, 2024-06-19:
> Can you check if `ha host logs -n1000 -b-1` (or `-b-2`, `-b-3`, etc.) show any logs from the previous failed boots that would indicate the reason why it failed? Alternatively, can you connect a display to the Pi and check where it fails during booting?

I try also to update from terminal/ssh ha os update --version 12.4 

Same issue. The error i got before (  error -110 whilst initialising SD card. ) is nowhere anymore.

The system boot on 12.3 and is running normal.

showupch, 2024-06-19:
I found the issue. I just remove the SD card and all is running correct. Looks like I forget it inside when I was installing everything new on my P5.

I close the issue here. I'm really sorry for the troubles. 

KVT

closed 2024-06-19 by showupch