home-assistant/operating-system #3428: Upgrade 12.4 caused VMWare guest to hang

madbrain76, 2024-06-19:
### Describe the issue you are experiencing

I upgraded my HAOS VM to 12.4 today. It did not come back online.
When I logged in to the host, I saw that the guest VM had a black screen.
VMWare reported an error that itI was unable to power down the VM.
I then rebooted the host, and HAOS came back online.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

12.4

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

VMWare Workstation Pro 17.5.2 .

### Steps to reproduce the issue

1. launch HAOS 12.3 on the VM
2. get notification to upgrade to 12.4
3. install
4. VM hangs, presumably on reboot

### Anything in the Supervisor logs that might be useful for us?

```txt
Unfortunately, no. All the logs are from after the successful restart.
```

### Anything in the Host logs that might be useful for us?

```txt
Unfortunately, no. All the logs are from after the successful restart.
```

### System information

## System Information

version | core-2024.6.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.2
os_name | Linux
os_version | 6.6.33-haos
arch | x86_64
timezone | America/Los_Angeles
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4960
Installed Version | 1.34.0
Stage | running
Available Repositories | 1388
Downloaded Repositories | 6

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.4
-- | --
update_channel | stable
supervisor_version | supervisor-2024.06.0
agent_version | 1.6.0
docker_version | 26.1.4
disk_total | 251.3 GB
disk_used | 168.9 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | vmware
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Terminal & SSH (9.14.0), File editor (5.8.0), Samba Backup (5.2.0), Samba share (12.3.1), Z-Wave JS UI (3.8.0), ESPHome (2024.5.5), Network UPS Tools (0.13.0), MariaDB (2.7.1), Studio Code Server (5.15.0), Whisper (2.1.0), Piper (1.5.0), openWakeWord (1.10.0), Music Assistant (2.0.7)

</details>

<details><summary>Dashboards</summary>

dashboards | 7
-- | --
resources | 1
views | 5
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | June 9, 2024 at 10:52
-- | --
current_recorder_run | June 18, 2024 at 19:00
estimated_db_size | 2476.69 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

### Additional information

Seems like it might be a VMWare bug. The hypervisor got in a strange state.

sairon, 2024-06-19:
Not sure it's an OS bug, let's wait if we get more reports of this happening, otherwise there's not much to do about it.

madbrain76, 2024-06-19:
Yeah, it could very well be a VMWare bug. but I'm unable to open a case with Broadcom due to lacking a site ID, which I gather is only given to paying customers.

Gillennium-O, 2024-06-20:
I also had an issue with my VM running on VMware ESXi 6.7 Update 1. Where after the update HomeAssistant would load, then as soon as I accessed the GUI the VM would shut down. I upgraded the VM version to Version 14 and the VM is now stable.

grfernandes54, 2024-06-22:
I having the same issue and Discover that is something related tô Zigbee2Mqtt addon. If i try to start or stop Zigbee 2mqtt the VM free, so I disable the automaic startup at Zigbee2mqtt and with this action i was able tô reload and not get VM freeze

I open a issue at z2m but not sure what is causing the issue

madbrain76, 2024-06-22:
@grfernandes54 FYI, I'm not using Zigbee.

grfernandes54, 2024-06-22:
Any other USB device that you can remove to test?

madbrain76, 2024-06-22:
@grfernandes54 
I have a Zooz ZST39 Z-Wave stick, and 2 APC Back-UPS BX1500M UPSes attached to the VM.
I can remove the UPSes temporarily (I'm actually not doing anything with them except collect sensor data - had a short power outage this morning so it will be interesting to look at).
But for the Z-Wave stick, I have 43 Z-Wave devices that depend on it, and not really keen on removing it.
The problem isn't reproducible even with those devices, anyway. I don't see how removing them will impact the behavior.

closed 2024-09-27 by github-actions[bot]