home-assistant/operating-system #3429: Home Assistant Green 12.4 Upgrade - Broken UPnP/IGD and Nginx Proxy

rac146, 2024-06-19:
### Describe the issue you are experiencing

Upgraded my Home Assistant Green to 12.4 today and realized it was breaking the UPnP/IGD integration. Upon further analysis, my Nginx proxies were no longer working either. Downgrading back to 12.3 fixes both of these issues.

### What operating system image do you use?

green (Home Assistant Green)

### What version of Home Assistant Operating System is installed?

12.4

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Stock Home Assistant Green - I do have an extra USB ethernet adapter attached to talk to another network..

### Steps to reproduce the issue

1. Install 12.4 upgrade
2. Upon restart, UPnP/IGD Integration will not load correctly. Nginx proxies stop working
3. Downgrade back to 12.3
4. Upon restart with 12.3, UPnP/IGD Integration and Nginx working properly

### Anything in the Supervisor logs that might be useful for us?

```txt
N/A
```

### Anything in the Host logs that might be useful for us?

```txt
N/A
```

sairon, 2024-06-19:
In what way it "didn't load correctly" and what Nginx proxies? Can you please try reproducing it again and provide some logs?

rac146, 2024-06-19:
I have 1 UPnP/IGD device that looks at my main router. If I install 12.4 that device never initializes. If I delete the entity and try to re-add, the UPnP integration says "no devices found on this network".

Regarding Nginx - I am running Nginx Proxy Manager to run a proxy host. With 12.4, this proxy host becomes completely inaccessible via the internet and causes intermittent behavior locally.

Rolling back to 12.3 fixes both of these issues. I will upgrade again later today and post some logs!

rac146, 2024-06-19:
Update - after a second attempt at upgrading to 12.4, everything is working as expected. Will keep monitoring.

When I was on 12.4 the first time, I tried numerous restarts and power off/on resets to see if that resolved the issues, but the only action that fixed the problem was downgrading back to 12.3.

Could it have been a failed upgrade in a sense? Hard to pinpoint. Also, I'm not sure if this matters but the only extra USB device I have attached to the Green is an ethernet adapter so I can access a separate network.

rac146, 2024-06-19:
Second Update: The UPnP integration continues to work, but the Nginx proxy has stopped working.. Adding logs to my original post

rac146, 2024-06-19:
One more update: I no longer believe this is associated to the OS update to 12.4, but is manifesting itself during OS upgrades/downgrades or HASS restarts. The HASS Green is connected to 2 different networks (192.168.0.1 and 192.168.4.1) - both of these networks have access to the internet. I think i'm confusing HASS and it's directing internet traffic to one of these networks at random. When it goes through 192.168.0.1, internet works fine, but gets blocked on the 192.168.4.1 network. I'm going to close this and block internet access on my 4.1 network.

closed 2024-06-19 by rac146