home-assistant/operating-system #3432: Failed to install 12.4

stain3565, 2024-06-19:
### Describe the issue you are experiencing

current os version 12.3
Attempted to install 12.4 from updates
"Failed to call service update/install"
Have restared supervisor and restarted home assistant. No success.

Logs:
`
2024-06-19 13:10:50.436 INFO (MainThread) [supervisor.os.manager] Fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/12.4/haos_rpi4-64-12.4.raucb
2024-06-19 13:10:55.306 INFO (MainThread) [supervisor.os.manager] Completed download of OTA update file /data/tmp/hassos-12.4.raucb
2024-06-19 13:10:55.715 ERROR (MainThread) [supervisor.os.manager] Home Assistant Operating System update failed with: Installation error: Failed updating slot boot.0: failed to run slot hook: Child process exited with code 1
`

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

6.6

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

Just pi4

### Steps to reproduce the issue

1. Simply click install fir version 12.4
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
2024-06-19 13:10:50.436 INFO (MainThread) [supervisor.os.manager] Fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/12.4/haos_rpi4-64-12.4.raucb
2024-06-19 13:10:55.306 INFO (MainThread) [supervisor.os.manager] Completed download of OTA update file /data/tmp/hassos-12.4.raucb
2024-06-19 13:10:55.715 ERROR (MainThread) [supervisor.os.manager] Home Assistant Operating System update failed with: Installation error: Failed updating slot boot.0: failed to run slot hook: Child process exited with code 1
```

### Anything in the Host logs that might be useful for us?

```txt
Nothing useful
```

### System information

## System Information

version | core-2024.6.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.2
os_name | Linux
os_version | 6.6.28-haos-raspi
arch | aarch64
timezone | Europe/London
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.34.0
Stage | running
Available Repositories | 1404
Downloaded Repositories | 60

</details>

<details><summary>AccuWeather</summary>

can_reach_server | ok
-- | --
remaining_requests | 27

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 29 September 2024 at 01:00
relayer_connected | true
relayer_region | eu-central-1
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | true
remote_server | eu-central-1-9.ui.nabu.casa
certificate_status | ready
instance_id | 2334dad093c247fa8b69b2c522f4a8b0
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.3
-- | --
update_channel | stable
supervisor_version | supervisor-2024.06.0
agent_version | 1.6.0
docker_version | 25.0.5
disk_total | 109.3 GB
disk_used | 76.8 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | Terminal & SSH (9.14.0), Home Assistant Google Drive Backup (0.112.1), MariaDB (2.7.1), Z-Wave JS UI (3.8.0), Studio Code Server (5.15.0), Mosquitto broker (6.4.1), Samba share (12.3.1), Grafana (10.0.0), InfluxDB (5.0.0), AppDaemon (0.16.6), Duck DNS (1.17.0), File editor (5.8.0), SQLite Web (4.1.2), Z-Wave JS (0.6.0), Zigbee2MQTT (1.38.0-1), TileBoard (1.3.8), Ring-MQTT with Video Streaming (5.6.7), Samba NAS (12.2.0-nas2), eWeLink Smart Home (1.4.3), TasmoAdmin (0.30.3), Ring Livestream (1.35), Samba Backup (5.2.0), Node-RED (17.0.13), ESPHome (2024.5.5), Piper (1.5.0), Whisper (2.1.0), openWakeWord (1.10.0), Git pull (7.14.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 7
-- | --
resources | 32
views | 38
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 15 June 2024 at 22:34
-- | --
current_recorder_run | 19 June 2024 at 11:19
estimated_db_size | 1301.17 MiB
database_engine | mysql
database_version | 10.11.6

</details>

<details><summary>Sonoff</summary>

version | 3.7.3 (e240aaf)
-- | --
cloud_online | 1 / 1
local_online | 1 / 1

</details>

sairon, 2024-06-19:
Do you have direct access to the Pi to check output of `journalctl -u rauc.service`? Maybe something could be in `ha host logs -t rauc` as well but that won't be as detailed.

agners, 2024-09-12:
Typically this happens when there are two Home Assistant OS installation, on the SSD and on a internal SD card as well. If that is the case, probably best is to take a full backup and download, and start with a clean install on a SD card and use the SSD as [data disk](https://www.home-assistant.io/common-tasks/os/#using-external-data-disk).

bluenadas, 2024-09-12:
I've been running with the same hardware for a year an never had the issue previously? Assuming there are two OS versions (sd/ssd) is this behavior something that appeared in recent versions of OS?

d96moe, 2024-12-06:
I just faced the same issue trying to update 13.2 - > 14.0.
Running from ssd with nvme hat with no SD Card installed at all. Never happened before.

closed 2025-03-15 by github-actions[bot]