home-assistant/operating-system #3433: NFS share stopped working after upgrade.

spitfire, 2024-06-19:
### Describe the issue you are experiencing

After upgrading to 12.4 my mass storage NFS share stopped working in HA, trying to re-add it ends up with messages like: 

`Could not unmount s01truenas01_ha01_backup due to: Transaction for mnt-data-supervisor-mounts-s01truenas01_ha01_backup.mount/stop is destructive (mnt-data-supervisor-mounts-s01truenas01_ha01_backup.mount has 'start' job queued, but 'stop' is included in transaction).`

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

6.6.33-haos	Home Assistant OS 12.4

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Proxmox VE virtual machine 

### Steps to reproduce the issue

1. Use NFS as backup datastore
2. Upgrade HAOS to 12.4
3. See alert regarding storage device

### Anything in the Supervisor logs that might be useful for us?

```txt
2024-06-19 18:45:04.913 ERROR (MainThread) [supervisor.mounts.mount] Could not unmount s01truenas01_ha01_backup due to: Transaction for mnt-data-supervisor-mounts-s01truenas01_ha01_backup.mount/stop is destructive (mnt-data-supervisor-mounts-s01truenas01_ha01_backup.mount has 'start' job queued, but 'stop' is included in transaction).
2024-06-19 18:52:32.907 ERROR (MainThread) [supervisor.mounts.mount] Mounting s01truenas01_ha01_backup did not succeed. Check host logs for errors from mount or systemd unit mnt-data-supervisor-mounts-s01truenas01_ha01_backup.mount for details.
2024-06-19 18:52:32.920 ERROR (MainThread) [supervisor.mounts.mount] Could not unmount s01truenas01_ha01_backup due to: Transaction for mnt-data-supervisor-mounts-s01truenas01_ha01_backup.mount/stop is destructive (mnt-data-supervisor-mounts-s01truenas01_ha01_backup.mount has 'start' job queued, but 'stop' is included in transaction).
```

### Anything in the Host logs that might be useful for us?

```txt
2024-06-19 16:44:24.957 V01HA01 systemd[1]: Mounting Supervisor nfs mount: s01truenas01_ha01_backup...
2024-06-19 16:44:30.948 V01HA01 systemd[1]: Requested transaction contradicts existing jobs: Transaction for mnt-data-supervisor-mounts-s01truenas01_ha01_backup.mount/stop is destructive (mnt-data-supervisor-mounts-s01truenas01_ha01_backup.mount has 'start' job queued, but 'stop' is included in transaction).
2024-06-19 16:45:00.868 V01HA01 systemd[1]: Requested transaction contradicts existing jobs: Transaction for mnt-data-supervisor-mounts-s01truenas01_ha01_backup.mount/stop is destructive (mnt-data-supervisor-mounts-s01truenas01_ha01_backup.mount has 'start' job queued, but 'stop' is included in transaction).
2024-06-19 16:45:05.028 V01HA01 systemd[1]: Requested transaction contradicts existing jobs: Transaction for mnt-data-supervisor-mounts-s01truenas01_ha01_backup.mount/stop is destructive (mnt-data-supervisor-mounts-s01truenas01_ha01_backup.mount has 'start' job queued, but 'stop' is included in transaction).
2024-06-19 16:45:06.627 V01HA01 systemd[1]: Requested transaction contradicts existing jobs: Transaction for mnt-data-supervisor-mounts-s01truenas01_ha01_backup.mount/stop is destructive (mnt-data-supervisor-mounts-s01truenas01_ha01_backup.mount has 'start' job queued, but 'stop' is included in transaction).
2024-06-19 16:45:55.050 V01HA01 systemd[1]: mnt-data-supervisor-mounts-s01truenas01_ha01_backup.mount: Mounting timed out. Terminating.
2024-06-19 16:45:55.051 V01HA01 systemd[1]: mnt-data-supervisor-mounts-s01truenas01_ha01_backup.mount: Mount process exited, code=killed, status=15/TERM
2024-06-19 16:45:55.051 V01HA01 systemd[1]: mnt-data-supervisor-mounts-s01truenas01_ha01_backup.mount: Failed with result 'timeout'.
2024-06-19 16:45:55.051 V01HA01 systemd[1]: Failed to mount Supervisor nfs mount: s01truenas01_ha01_backup.
2024-06-19 16:52:02.865 V01HA01 systemd[1]: Mounting Supervisor nfs mount: s01truenas01_ha01_backup...
2024-06-19 16:52:32.920 V01HA01 systemd[1]: Requested transaction contradicts existing jobs: Transaction for mnt-data-supervisor-mounts-s01truenas01_ha01_backup.mount/stop is destructive (mnt-data-supervisor-mounts-s01truenas01_ha01_backup.mount has 'start' job queued, but 'stop' is included in transaction).
2024-06-19 16:52:54.061 V01HA01 systemd[1]: Starting Cleanup of Temporary Directories...
2024-06-19 16:52:54.079 V01HA01 systemd[1]: systemd-tmpfiles-clean.service: Deactivated successfully.
2024-06-19 16:52:54.080 V01HA01 systemd[1]: Finished Cleanup of Temporary Directories.
2024-06-19 16:53:33.050 V01HA01 systemd[1]: mnt-data-supervisor-mounts-s01truenas01_ha01_backup.mount: Mounting timed out. Terminating.
2024-06-19 16:53:33.050 V01HA01 systemd[1]: mnt-data-supervisor-mounts-s01truenas01_ha01_backup.mount: Mount process exited, code=killed, status=15/TERM
2024-06-19 16:53:33.051 V01HA01 systemd[1]: mnt-data-supervisor-mounts-s01truenas01_ha01_backup.mount: Failed with result 'timeout'.
2024-06-19 16:53:33.051 V01HA01 systemd[1]: Failed to mount Supervisor nfs mount: s01truenas01_ha01_backup.
2024-06-19 16:53:33.070 V01HA01 systemd[1]: Mounting Supervisor nfs mount: s01truenas01_ha01_backup...
2024-06-19 16:54:03.094 V01HA01 systemd[1]: Requested transaction contradicts existing jobs: Transaction for mnt-data-supervisor-mounts-s01truenas01_ha01_backup.mount/stop is destructive (mnt-data-supervisor-mounts-s01truenas01_ha01_backup.mount has 'start' job queued, but 'stop' is included in transaction).
2024-06-19 16:55:03.300 V01HA01 systemd[1]: mnt-data-supervisor-mounts-s01truenas01_ha01_backup.mount: Mounting timed out. Terminating.
2024-06-19 16:55:03.302 V01HA01 systemd[1]: mnt-data-supervisor-mounts-s01truenas01_ha01_backup.mount: Mount process exited, code=killed, status=15/TERM
```

### System information

## System Information

version | core-2024.6.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.2
os_name | Linux
os_version | 6.6.33-haos
arch | x86_64
timezone | Europe/Warsaw
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.34.0
Stage | running
Available Repositories | 1470
Downloaded Repositories | 75
HACS Data | ok

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.4
-- | --
update_channel | stable
supervisor_version | supervisor-2024.06.0
agent_version | 1.6.0
docker_version | 26.1.4
disk_total | 137.1 GB
disk_used | 59.0 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | kvm
board | ova
supervisor_api | ok
version_api | ok
installed_addons | RPC Shutdown (2.5), Studio Code Server (5.15.0), Tautulli (4.0.1), Mosquitto broker (6.4.1), phpMyAdmin (0.9.1), Tor (5.0.1), Uptime Kuma (0.12.1), ESPHome (2024.5.5), Signal Messenger (0.84.0), Let's Encrypt (5.0.18), Traccar (0.25.0), MariaDB (2.7.1), Cloudflared (5.1.12), Nginx Proxy Manager (1.0.1), eufy-security-ws (1.8.0-2), Samba share (12.3.1), Advanced SSH & Web Terminal (18.0.0), AdGuard Home (5.1.1), AppDaemon (0.16.6), Grocy (0.21.0), Zigbee2MQTT (1.38.0-1), ZigStar TI CC2652P7 FW Flasher (0.2.0), ZigStar TI CC2652P/P7 FW Flasher (0.4.0), Piper (1.5.0), go2rtc (1.9.4), Zigbee2MQTT Edge (edge), Log Viewer (0.17.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 7
-- | --
resources | 29
views | 19
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 11 czerwca 2024 16:03
-- | --
current_recorder_run | 19 czerwca 2024 18:39
estimated_db_size | 2618.97 MiB
database_engine | mysql
database_version | 10.11.6

</details>

<details><summary>Solcast PV Forecast</summary>

can_reach_server | ok
-- | --
used_requests | 6
rooftop_site_count | 2

</details>

### Additional information

NFS share hosted on Truenas Scale Dragonfish-24.04.1.1, worked great until HAOS upgrade

spitfire, 2024-06-19:
I did try to reboot it again, then tried to add the mount:
Supervisor logs says
```
2024-06-19 20:50:49.012 ERROR (MainThread) [supervisor.mounts.mount] Mounting s01truenas01backup did not succeed. Check host logs for errors from mount or systemd unit mnt-data-supervisor-mounts-s01truenas01backup.mount for details.
2024-06-19 20:50:49.024 ERROR (MainThread) [supervisor.mounts.mount] Could not unmount s01truenas01backup due to: Transaction for mnt-data-supervisor-mounts-s01truenas01backup.mount/stop is destructive (mnt-data-supervisor-mounts-s01truenas01backup.mount has 'start' job queued, but 'stop' is included in transaction).
```
host log (for some reason it seems to be using UTC time, not local?) says:
```
2024-06-19 18:50:18.997 V01HA01 systemd[1]: Mounting Supervisor nfs mount: s01truenas01backup...
2024-06-19 18:50:49.024 V01HA01 systemd[1]: Requested transaction contradicts existing jobs: Transaction for mnt-data-supervisor-mounts-s01truenas01backup.mount/stop is destructive (mnt-data-supervisor-mounts-s01truenas01backup.mount has 'start' job queued, but 'stop' is included in transaction).
2024-06-19 18:51:49.050 V01HA01 systemd[1]: mnt-data-supervisor-mounts-s01truenas01backup.mount: Mounting timed out. Terminating.
2024-06-19 18:51:49.051 V01HA01 systemd[1]: mnt-data-supervisor-mounts-s01truenas01backup.mount: Mount process exited, code=killed, status=15/TERM
2024-06-19 18:51:49.051 V01HA01 systemd[1]: mnt-data-supervisor-mounts-s01truenas01backup.mount: Failed with result 'timeout'.
2024-06-19 18:51:49.051 V01HA01 systemd[1]: Failed to mount Supervisor nfs mount: s01truenas01backup.
```

closed 2024-09-25 by github-actions[bot]

rsch90, 2024-10-18:
I have the same behaviour since two weeks or so

campfred, 2024-10-30:
Same as rsch90 but on Raspberry Pi.
HA OS 13.2
HA Supervisor 2024.10.2
HA Core 2024.10.2
Running baremetal on rpi4-64

shlomiassaf, 2024-12-19:
Same here!

HA OS 14.0
HA Core 2024.12.4

On intel NUC

For me, its about mnt-data-supervisor-mounts-**frigate**

Remove frigate completely and it still error's.

Maybe some rule got stuck and did not clean when I uninstalled

rsch90, 2024-12-19:
For me it's resolved. Have done nothing to resolve it tried it with the latest versions beginning of this week and it's working again.
I am on OS 14.1, Supervisor 2014.12.0 and Core 2024.12.4
I have a second HA for my parents they never had any issues at all.

Hondahu, 2025-01-24:
Same here.

I just can’t add the NFS storage from my Synology NAS.

System > storage > add network storage > 

Give a name: SynologyNFS
Select usage as Media
Put the IP of my NAS
Select NFS
Put the volume path: /volume1/home-assistant 
—> didn’t work.. 

I also mounted the NFS into another raspberry pi running raspbian in the same network as the Homeassistant raspberry pi. It did work without any issue.

I’m also not being able to ping the NAS from the HA terminal(timeout), which is possible from any other capable device on my network.

I suspect this is a DNS problem but I added the DNS sever of my router into the network settings of HA but still no luck. I also added the nameserver of my local dns on the file /etc/resolv.conf but it just won’t work.

I restarted the system multiple times. :( 

Please help. Getting this working would be very cool. Thanks!!

My setup:

Version	core-2025.1.4
Installationstyp	Home Assistant OS
Entwicklung	false
Supervisor	true
Docker	true
Benutzer	root
Virtuelle Umgebung	false
Python-Version	3.13.1
Betriebssystemfamilie	Linux
Betriebssystem-Version	6.6.62-haos-raspi
CPU-Architektur	aarch64
Zeitzone	Europe/Berlin
Konfigurationsverzeichnis	/config

These are my logs:

### **Home Assistant Core log:**
Logger: homeassistant.components.hassio
Quelle: components/hassio/websocket_api.py:135
Integration: Home Assistant Supervisor (Dokumentation, Probleme)
Erstmals aufgetreten: 17:13:53 (1 Vorkommnisse)
Zuletzt protokolliert: 17:13:53

Failed to to call /mounts - Could not unmount SynologyNFS due to: Transaction for mnt-data-supervisor-mounts-SynologyNFS.mount/stop is destructive (mnt-data-supervisor-mounts-SynologyNFS.mount has 'start' job queued, but 'stop' is included in transaction).

### **Supervisor logs:**

2025-01-24 17:13:23.502 INFO (MainThread) [supervisor.mounts.manager] Creating or updating mount: SynologyNFS
2025-01-24 17:13:53.613 WARNING (MainThread) [supervisor.mounts.mount] Mount SynologyNFS still in state activating after waiting for 30 seconds to complete
2025-01-24 17:13:53.614 ERROR (MainThread) [supervisor.mounts.mount] Reloading SynologyNFS did not succeed. Check host logs for errors from mount or systemd unit mnt-data-supervisor-mounts-SynologyNFS.mount for details.
2025-01-24 17:13:53.653 ERROR (MainThread) [supervisor.mounts.mount] Could not unmount SynologyNFS due to: Transaction for mnt-data-supervisor-mounts-SynologyNFS.mount/stop is destructive (mnt-data-supervisor-mounts-SynologyNFS.mount has 'start' job queued, but 'stop' is included in transaction).