home-assistant/operating-system #3434: [WARN] Home Assistant CLI is not running! Jump into emergency console

Mrsash2020, 2024-06-20:
### Describe the issue you are experiencing

After update I get this message in the console. I am on VirtualBox 7 latest on Win 11 x86 Machine.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

Home Assistant OS 12.4

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

I am on VirtualBox 7 latest on Win 11 x86 Machine
Intel 13600K
Asrock Z790 Steel Legend WiFi
DDR5 32GB
SLZB-06 POE for Z2MQTT

### Steps to reproduce the issue

1.Launch VirtualBox > HASS 
2.[WARN] Home Assistant CLI is not running! Jump into emergency console
3.Link to get to home assistant still works

### Anything in the Supervisor logs that might be useful for us?

```txt
[32m2024-06-20 09:55:15.317 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-audio with version 2023.12.0[0m
[32m2024-06-20 09:55:15.346 INFO (MainThread) [supervisor.plugins.audio] Starting Audio plugin[0m
[32m2024-06-20 09:55:17.920 INFO (MainThread) [supervisor.docker.audio] Starting Audio ghcr.io/home-assistant/amd64-hassio-audio with version 2023.12.0 - 172.30.32.4[0m
[32m2024-06-20 09:55:17.923 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-observer with version 2023.06.0[0m
[32m2024-06-20 09:55:18.203 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-multicast with version 2024.03.0[0m
[32m2024-06-20 09:55:18.277 INFO (MainThread) [supervisor.plugins.multicast] Starting Multicast plugin[0m
[32m2024-06-20 09:55:19.329 INFO (MainThread) [supervisor.docker.multicast] Starting Multicast ghcr.io/home-assistant/amd64-hassio-multicast with version 2024.03.0 - Host[0m
[32m2024-06-20 09:55:19.448 INFO (MainThread) [supervisor.homeassistant.secrets] Loaded 2 Home Assistant secrets[0m
[32m2024-06-20 09:55:20.363 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/qemux86-64-homeassistant with version 2024.6.3[0m
[32m2024-06-20 09:55:20.796 INFO (MainThread) [supervisor.os.manager] Detect Home Assistant Operating System 12.4 / BootSlot A[0m
[32m2024-06-20 09:55:36.844 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/a889bffc repository[0m
[32m2024-06-20 09:55:36.844 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/core repository[0m
[32m2024-06-20 09:55:36.845 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/5c53de3b repository[0m
[32m2024-06-20 09:55:36.846 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/45df7312 repository[0m
[32m2024-06-20 09:55:36.847 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/15d21743 repository[0m
[32m2024-06-20 09:55:36.848 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/a0d7b954 repository[0m
[32m2024-06-20 09:56:34.774 INFO (MainThread) [supervisor.store] Loading add-ons from store: 85 all - 85 new - 0 remove[0m
[32m2024-06-20 09:56:34.803 INFO (MainThread) [supervisor.addons.manager] Found 11 installed add-ons[0m
[32m2024-06-20 09:56:35.020 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/amd64-addon-mosquitto with version 6.4.1[0m
[32m2024-06-20 09:56:35.093 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/ssh/amd64 with version 18.0.0[0m
[32m2024-06-20 09:56:35.203 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/vscode/amd64 with version 5.15.0[0m
[32m2024-06-20 09:56:35.232 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/amd64-addon-duckdns with version 1.17.0[0m
[32m2024-06-20 09:56:35.286 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/amd64-addon-configurator with version 5.8.0[0m
[32m2024-06-20 09:56:35.339 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/amd64-addon-samba with version 12.3.1[0m
[32m2024-06-20 09:56:35.375 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/amd64-addon-nginx_proxy with version 3.9.0[0m
[32m2024-06-20 09:56:35.480 INFO (MainThread) [supervisor.docker.interface] Attaching to alexxit/go2rtc with version 1.9.4[0m
[32m2024-06-20 09:56:35.629 INFO (MainThread) [supervisor.docker.interface] Attaching to zigbee2mqtt/zigbee2mqtt-amd64 with version 1.38.0-1[0m
[32m2024-06-20 09:56:35.760 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/esphome/esphome-hassio with version 2024.6.0[0m
[32m2024-06-20 09:56:35.775 INFO (MainThread) [supervisor.docker.interface] Attaching to thomasmauerer/samba-backup-amd64 with version 5.2.0[0m
[32m2024-06-20 09:56:35.951 INFO (MainThread) [supervisor.backups.manager] Found 7 backup files[0m
[32m2024-06-20 09:56:36.375 INFO (MainThread) [supervisor.discovery] Loaded 2 messages[0m
[32m2024-06-20 09:56:36.375 INFO (MainThread) [supervisor.ingress] Loaded 6 ingress sessions[0m
[32m2024-06-20 09:56:36.375 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state setup[0m
[32m2024-06-20 09:56:36.375 INFO (MainThread) [supervisor.resolution.checks.base] Run check for disabled_data_disk/system[0m
[32m2024-06-20 09:56:36.375 INFO (MainThread) [supervisor.resolution.checks.base] Run check for detached_addon_removed/addon[0m
[32m2024-06-20 09:56:36.375 INFO (MainThread) [supervisor.resolution.checks.base] Run check for multiple_data_disks/system[0m
[32m2024-06-20 09:56:36.375 INFO (MainThread) [supervisor.resolution.checks.base] Run check for detached_addon_missing/addon[0m
[... 378 lines cut ...]

<details><summary>Dashboards</summary>

dashboards | 6
-- | --
resources | 34
views | 39
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | June 19, 2024 at 11:49 PM
-- | --
current_recorder_run | June 20, 2024 at 9:57 AM
estimated_db_size | 5.45 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

### Additional information

I started having this issue after I updated yesterday. I have tried typing login and other suggestions including going to a previous backup being restored but this message on the console stays put. thanks

Mrsash2020, 2024-06-22:
Todays update fixed the issue

closed 2024-06-22 by Mrsash2020

puffo1969, 2024-07-14:
I have the same issue with raspberry pi 

I used 
[image]

Thank you in advance