home-assistant/operating-system #3437: Home Assistant OS 12.4 does not boot with an SD card from a Rasp Pi 5

BatuhanBasaran, 2024-06-20:
### Describe the issue you are experiencing

The SD card was flashed with the RaspberryPi Imager 1.8.5 and for test purposes with BalenaEtcher.
I used the following image with BalenaEtcher:
https://github.com/home-assistant/operating-system/releases/download/12.4/haos_rpi5-64-12.4.img.xz

Unfortunately the Pi5 is stuck in the boot menu. No matter if with the RaspberryPi Imager or with BalenaEtcher.

[image]

Finally I downloaded haos_rpi5-64-11.5.img and flashed it. This works perfectly on the Pi5.

### What operating system image do you use?

rpi5-64  (Raspberry Pi 5 64-bit OS)

### What version of Home Assistant Operating System is installed?

12.4

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Raspberry Pi 5 with 8GB Ram

### Steps to reproduce the issue

1.Flash HAOS 12.4 on SD-Card
2.Boot the Pi5 with the SD-Card
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
-
```

### Anything in the Host logs that might be useful for us?

```txt
-
```

Baxxy13, 2024-06-20:
Can confirm that booting OS 12.4 isn't possible with SD-Card or USB-SSD. 
(NVME not tested)

Kitekater, 2024-06-20:
me too.
Same picture

adam-jones-net, 2024-06-20:
Not sure if this is the same issue but I ran the latest update when prompted to install on my mobile app. I swear it was 2024.6.4 even though today checking on github the latest version is listed as 2024.6.3.

gokartn, 2024-06-20:
I had the same issue.  
Thought my Raspberry Pi 5 was broken, the only thing keeping me going was the boot screen Switching from "Detected" to "not detected" when pluggin in and taking out the SD Card. As well as Raspberry Pi Os Lite working. 

Used the Imager to get the latet 12.4 and it gave the exact same image as OP. 
Downloaded 12.3 from [Here](https://github.com/home-assistant/operating-system/releases/download/12.3/haos_rpi5-64-12.3.img.xz) and flashed it using the imager, and it worked on first boot.

DonnieNarco, 2024-06-21:
Yes, exact same issue with a brand new Pi5. I thought I was going crazy because no one seemed to have same issue, until I found this. So, I will use 12.3 for now.......Does anyone know if I should then avoid any and all updates until an official statement comes out that says it's fixed??

tobtobi, 2024-06-21:
Exactly the same issue here also on a Pi5 4GB
BalenaEtecher and Pi Imager both same problem
Only Solution was running the older 12.3 Version

UF-Elektron, 2024-06-21:
Same issue! Brand new Raspberry Pi 5, Home Assistant OS 12.4 does not boot.
SD Card: SanDisk Extreme PRO 32GB
SD Card is detected, but boot process keeps looping.

Tried if my raspi works by inserting SD Card with Raspberry Pi OS and it booted without any issues.

Baxxy13, 2024-06-21:
The problem only occurs with fresh flashed `haos_rpi5-64-12.4.img.xz`.
If you start with `haos_rpi5-64-12.3.img.xz` you can safely update to 12.4 from within HA-UI.

Just tested this and it works without problems.

DonnieNarco, 2024-06-21:
> The problem only occurs with fresh flashed `haos_rpi5-64-12.4.img.xz`. If you start with `haos_rpi5-64-12.3.img.xz` you can safely update to 12.4 from within HA-UI.
> 
> Just tested this and it works without problems.

I have read from others that upgrading to 12.4 from within HA also makes this get stuck in a boot loop.....

DonnieNarco, 2024-06-21:
> Hmm, are those readings Pi5B related?
> I flashed OS 12.3 to new SD-Card, booted and doing the onboarding.
> After that the 12.4 Update was announced in the UI and i installed it.
> Now i'm on 12.4 with my Pi5B8G.

I do recall a few people saying RPi5, but I will give that a shot once I
get this in place. I have successfully installed 12.3 now and restored from
a backup. I am at work now, but once I get home, I will put this in and see
how it goes.

FiftyBillion, 2024-06-21:
Exactly the same issue and error I got.
I just got the new Raspberry pi 5 and a whole new SD card yesterday.
I have tried both Raspberry pi imager and Balena etcher, same issue.

mwortham94, 2024-06-21:
Gosh, wish I'd thought to actually look it up vice tag-teaming with ChatGPT for like 5 hours. Same issue. Glad to know I can downgrade to 12.3. RPi5 that I bought a few days ago.

ababij, 2024-06-21:
> I'm having the same issue with a Pi 5/8GB. I attempted to run the update in HA but when it restarted it got stuck in the boot menu. I then attempted a new install using RaspberryPi Imager 1.8.5 and had the same result. Tried a couple SD cards but still the same. Downloaded a 12.3 image and everything worked again.

After reading Baxxy13's comments, I tried to run the 12.4 update again from the HA UI and it worked this time.  Maybe it was cause I had a fresh 12.3 install on the SD?  
Whatever it was, when I started with a new 12.3 install and then used the HA UI to do the update to 12.4, it worked.

GlobierH, 2024-06-22:
I have the exact same issue.
Installed 12.4 via Imager 1.8.5 and have the same boot issue
Installed 12.3 via img download then via imager and boots fine.
Havent tried the update from 12.3 yet as just want to get things restored and back running. Will monitor this issue for when the issue is fixed.

TomMarks, 2024-06-22:
Exactly same issue with Raspberry Pi 5, Pi Imager, and HA installation 12.4.
Downloading 12.3 image from here and installing with Pi Imager seems to work.

rodpayne, 2024-06-23:
I received a new Raspberry Pi today, and it had the same issue with hanging up booting the HA OS 12.4 image from SD. It turns on the red LED momentarily and then switches to solid green, with no flashing indicating activity.

It tried it with the 12.3 image, and it seemed to get past the initial boot, **but it would not connect over the ethernet port**. The amber and green LEDs on the ethernet port are on solid, with no flashing indicating network activity. (I also tried the 11.5 image, with no luck.)

Baxxy13, 2024-06-23:
The issue here is that the downloadable, or directly via Pi-Imager flashable, 12.4 image (for Pi5B) is somehow faulty.
The issue is not about 12.3 image or any other problems, ist just "Pi5B doesn't boot with freshly flashed OS 12.4".

And of course, to prevent others to get into this issue, the faulty image should be removed from downloads and Pi-Imager till a fixed image is available. 

But it's weekend and it seems nobody cares... 😉

sairon, 2024-06-24:
> And of course, to prevent others to get into this issue, the faulty image should be removed from downloads and Pi-Imager till a fixed image is available.

I quickly verified on my Pi5/8G to check if we should take this measure, and I'm not able to reproduce it here, so there are some other factors at play (very likely the EEPROM firmware). I will look into it deeper, but at this point it's clear that not everyone is affected and there will be likely also a different way to address it.

MrKlacker, 2024-06-24:
> I quickly verified on my Pi5/8G to check if we should take this measure, and I'm not able to reproduce it here, so there are some other factors at play (very likely the EEPROM firmware). I will look [...]

I can confirm I had the same. Purchased a fresh RPI5 with a fresh microSD and tried to flash 12.4. with balenaEtcher and RPI Imager. Both failed. Tried 12.3 and booted on first try.

DonnieNarco, 2024-06-24:
I'm not sure how you aren't able to reproduce it, when everyone here has
the same issue. Unless you are running some custom Pi Firmware or
something?

sairon, 2024-06-24:
@DonnieNarco That's the one million dollar question right now. Here's a screenshot of my RPi hanging in the bootloader (by adjusting the path to the kernel to prevent it from booting). You can see the version is the same as in the (so far only) image posted here:
[image]

DonnieNarco, 2024-06-24:
That is indeed interesting....I hope this gets fixed soon!

Baxxy13, 2024-06-24:
> and I'm not able to reproduce it here

Ok, and sorry for my comment about the faulty image.

It looks like the problem is with the imager programs (RPI imager / Etcher) or the underlaying os, and has nothing to do with the bootloader.
I normally use the RPI imager under Windows. 
For a test, I started Raspbian-OS Lite on the Pi5, downloaded the Pi5B HA-OS 12.4 image and flashed it to an SD card.
With this SD card, the Pi5B boots HA-OS 12.4 without any problems.

I used the following to flash the SD card:
`sudo dcfldd bs=1M if=haos_rpi5-64-12.4.img of=/dev/sdb`

I hope this helps with troubleshooting.

Translated with DeepL.com (free version)

sairon, 2024-06-24:
@Baxxy13 Thank you very much, that's really helpful, at least I know I (probably) don't need to chase ghosts in HAOS anymore.

That doesn't exactly answer the $1M question but gives a major clue. In my testing I tried using both RPi Imager and balenaEtcher (both on Linux though) and it worked for me as well, so either it affects these utilities only on Windows, or, more likely, for some reason the 12.4 image you (and the remaining people here) downloaded was corrupted somehow. For example balenaEtcher verifies the integrity of the written data, but it uses the original file on your PC as the reference, so it can't determine if the source file isn't faulty. Maybe Github had some trouble at the time, maybe some antivirus software interferes there, it's hard to tell at this point.

The simplest (but not completely reliable) method is to check the filesize of the image - it should be **exactly** 282171876 bytes in size. Uncompressed it's 2147483648 bytes, but I assume that corruption of the archive would result in file that can't be extracted.

More reliable way is to check the checksum of the files, which is following for SHA-256 algorithm:
```
d3c779c1af62300a689cb9f6db72c8bada1a4e02649c63922f0c8db9495f2a56  haos_rpi5-64-12.4.img
019a662e8898cc957cc9614f60782526de3675b6def6b148bcd7ef57184acd13  haos_rpi5-64-12.4.img.xz
```
For example [here](https://techdocs.akamai.com/download-ctr/docs/verify-checksum) is a guide how to check it on all major operating systems.

Baxxy13, 2024-06-24:
Hmm, the checksum for the downloaded `haos_rpi5-64-12.4.img.xz` is correct, so no download problems.
```
C:\WINDOWS\system32>certutil -hashfile F:\haos_rpi5-64-12.4.img.xz sha256
SHA256-Hash von F:\haos_rpi5-64-12.4.img.xz:
019a662e8898cc957cc9614f60782526de3675b6def6b148bcd7ef57184acd13
CertUtil: -hashfile-Befehl wurde erfolgreich ausgeführt.
```

If i unpack the image (using WinRar) the checksum of the `haos_rpi5-64-12.4.img` is also correct, so no interference from os or antivirus.
```
C:\WINDOWS\system32>certutil -hashfile F:\haos_rpi5-64-12.4.img sha256
SHA256-Hash von F:\haos_rpi5-64-12.4.img:
d3c779c1af62300a689cb9f6db72c8bada1a4e02649c63922f0c8db9495f2a56
CertUtil: -hashfile-Befehl wurde erfolgreich ausgeführt.
```
But the Pi5B won't boot...

sairon, 2024-06-24:
Thanks again, with the Windows version of the RPi Imager I was able to reproduce it. It shows that for some reason the partition table is altered in a way that the RPi bootloader is not able to read the boot partition. Moreover, I get exactly the same output with balenaEtcher and Rufus, so it's like if the GPT header was altered by Windows itself :thinking: 

I was testing on a laptop with Windows 11, I am curious if someone could try with an older Windows version and tell if it also generates unbootable images.

Baxxy13, 2024-06-24:
I'm using older Windows 10 Home 22H2.

sairon, 2024-06-25:
I tried using Raspberry Pi Imager on Windows 7 and it doesn't have the same flaw - the SD card is byte-to-byte identical with the source image, which makes me believe that the tools themselves aren't to blame. I'll continue looking into that but it seems Windows is doing something nasty and unexpected when it encounters the slightly changed GPT header that HAOS 12.4 has :( I have updated the issue title accordingly and pinned it for visibility.

**Currently the only workaround is either to use the older HAOS version and do an OS upgrade afterwards, or not use Windows 10+ (not sure about Windows 8) for creating the bootable device for RPi 5.**

Baxxy13, 2024-06-25:
Really strange.
I did some test's yesterday with other "windows-imaging-programs" (besides Etcher and Pi-Imager), but all fail to create a bootable 12.4 sd-card.

- Rufus 4.5 portable
- Win32DiskImager 1.0.0
- HDD Raw Copy Tool 1.20
  
So ist seems that really Windows is the issue... 😟

qemajlosmani, 2024-06-25:
I had the exact same issue. Was trying for 3 Hours to install HAOS. Tried different SD Cards, SSD Drives and then after flashing my SD Card with the version 13.3, installed and running without any problem.
Raspberry PI with 8GB

manstatt85, 2024-06-26:
Same for me - 12.4 stuck while booting, after installing 12.3 everything is fine

Patches520, 2024-06-27:
I have the same issue as everyone, but I'm stuck in a bad loop trying to download the updated supervisor and cant get install to boot. Any suggestions?

sairon, 2024-06-27:
There's not much more information to gather at this point, so I'm locking this issue to prevent spamming all the people who contributed with their findings. For general discussion please use Home Assistant Community forums or GH discussions.

PR #3497 (2024-07-30): Patch genimage to force first usable LBA @ 34 to fix RPi 5/Windows issues
Genimage sets the first usable LBA to the offset of the first partition. While it shouldn't be an issue in theory, Windows may do some nasty things with the GPT header afterwards which breaks the Raspberry Pi bootloader, manifesting as

Before purpose of this behavior is clarified in [1], add a downstream patch that sets the first usable LBA back to 34, which was the value that was used before migrating to Genimage in #3388. Since changing this value (hopefully) doesn't have any other consequences, and the images now sho [... cut]
commit d51a0d3c (2024-07-30): Patch genimage to force first usable LBA @ 34 to fix RPi 5/Windows issues (#3497)

Genimage sets the first usable LBA to the offset of the first partition. While
it shouldn't be an issue in theory, Windows may do some nasty things with the
GPT header afterwards which breaks the Raspberry Pi bootloader, manifesting as

Before purpose of this behavior is clarified in [1], add a downstream patch
that sets the first usable LBA back to 34, which was the value that was used
before migrating to Genimage in #3388. Since changing this value (hopefully)
doesn't have any other consequences, and the image [... cut]
closed 2024-07-30 by sairon