home-assistant/operating-system #3440: OS not properly reboot host

VenominousX, 2024-06-21:
### Describe the issue you are experiencing

Hello,
I restart my hosts every morning, as this provides the best reliability of HA for me. However, with OS version 12.4, it seems that the **host_reboot** command is not working properly. After executing the command, the machine does not reboot completely, which is indicated by the status LED on the Mini PC. This does not flash, but lights up constantly as in normal operation. Subsequently, HA does not receive a correct network configuration: 

`2024-06-21 04:11:46.096 ERROR (MainThread) [async_upnp_client.ssdp] Received error: [Errno 101] Network unreachable, transport: <_SelectorDatagramTransport fd=32 read=polling write=<idle, bufsize=0>>, socket: <asyncio. TransportSocket fd=32, family=2, type=2, proto=0, laddr=('0.0.0.0', 49442)>`

2024-06-21 04:11:46.097 ERROR (MainThread) [async_upnp_client.ssdp] An error was received: [Errno 99] address not available, transport: <_SelectorDatagramTransport fd=57 read=polling write=<idle, bufsize=0>>, socket: <asyncio.TransportSocket fd=57, family=10, type=2, proto=0, laddr=('::', 55125, 0, 0)>`

Then some integrations like AsusRouter and remote_homeassistant cannot be started and block the boot process. However, deactivating the integrations would not help here either, as controlling the master via tablets does not work without a network configuration on the master.

However, a manual hard reset will get the system up and running again.

Best regards

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

12.4

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

AMD Ryzen 8845HS
64GB DDR5 RAM
1TB PCi 4.0 SSD
Dual Edge TPU
2.5Gb Ethernet
Conbee2 Stick

### Steps to reproduce the issue

1. Home Assistant Command: Host reboot

### Anything in the Supervisor logs that might be useful for us?

```txt
2024-06-21 08:11:37.962 INFO (MainThread) [supervisor.dbus.systemd] Load dbus interface org.freedesktop.systemd1
2024-06-21 08:11:37.962 INFO (MainThread) [supervisor.dbus.timedate] Load dbus interface org.freedesktop.timedate1
2024-06-21 08:11:38.023 INFO (MainThread) [supervisor.host.services] Updating service information
2024-06-21 08:11:38.024 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
2024-06-21 08:11:38.026 WARNING (SyncWorker_0) [supervisor.host.sound] Can't update PulseAudio data: Failed to connect to pulseaudio server
2024-06-21 08:11:38.075 INFO (MainThread) [supervisor.host.network] Updating local network information
2024-06-21 08:11:38.091 INFO (MainThread) [supervisor.host.apparmor] Loading AppArmor Profiles: {'hassio-supervisor'}
2024-06-21 08:11:48.913 WARNING (MainThread) [supervisor.utils.whoami] Can't fetch Whoami data: Timeout
2024-06-21 08:11:48.913 WARNING (MainThread) [supervisor.core] Can't adjust Time/Date settings: Can't fetch Whoami data: Timeout
2024-06-21 08:11:48.917 INFO (MainThread) [supervisor.docker.monitor] Started docker events monitor
2024-06-21 08:11:59.913 WARNING (MainThread) [supervisor.jobs] 'Updater.fetch_data' blocked from execution, no supervisor internet connection
2024-06-21 08:11:59.922 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-cli with version 2024.05.0
2024-06-21 08:11:59.932 INFO (MainThread) [supervisor.plugins.cli] Starting CLI plugin
2024-06-21 08:12:00.129 INFO (MainThread) [supervisor.docker.cli] Starting CLI ghcr.io/home-assistant/amd64-hassio-cli with version 2024.05.0 - 172.30.32.5
2024-06-21 08:12:00.135 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-dns with version 2024.04.0
2024-06-21 08:12:00.144 INFO (MainThread) [supervisor.plugins.dns] Starting CoreDNS plugin
2024-06-21 08:12:00.322 INFO (MainThread) [supervisor.docker.dns] Starting DNS ghcr.io/home-assistant/amd64-hassio-dns with version 2024.04.0 - 172.30.32.3
2024-06-21 08:12:00.322 INFO (MainThread) [supervisor.plugins.dns] Updated /etc/resolv.conf
2024-06-21 08:12:10.943 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-audio with version 2023.12.0
2024-06-21 08:12:10.952 INFO (MainThread) [supervisor.plugins.audio] Starting Audio plugin
2024-06-21 08:12:11.155 INFO (MainThread) [supervisor.docker.audio] Starting Audio ghcr.io/home-assistant/amd64-hassio-audio with version 2023.12.0 - 172.30.32.4
2024-06-21 08:12:11.156 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-observer with version 2023.06.0
2024-06-21 08:12:11.163 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-multicast with version 2024.03.0
2024-06-21 08:12:11.167 INFO (MainThread) [supervisor.plugins.multicast] Starting Multicast plugin
2024-06-21 08:12:11.257 INFO (MainThread) [supervisor.docker.multicast] Starting Multicast ghcr.io/home-assistant/amd64-hassio-multicast with version 2024.03.0 - Host
2024-06-21 08:12:11.259 INFO (MainThread) [supervisor.homeassistant.secrets] Loaded 2 Home Assistant secrets
2024-06-21 08:12:11.263 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/generic-x86-64-homeassistant with version 2024.6.3
[... 223 lines cut ...]

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 0
views | 6
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 12. Juni 2024 um 20:38
-- | --
current_recorder_run | 21. Juni 2024 um 08:12
estimated_db_size | 275.79 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

### Additional information

As already mentioned, the master starts normally when I perform a hard reset. My master and the voice assist satellites are in a seperate subnet which blocks internet connection.

cogneato, 2024-06-25:
I'm not sure how related this is but you have multiple HA instances on the network? There appears to be a duel for the hostname

```
Hostname conflict, changing published hostname from 'homeassistant' to 'homeassistantX'. 
```
This appears several times with different attempts to assign a number.
You can rename one of your hosts to avoid this at Settings > System > Network in HA.
Are you rebooting both at the same time?

VenominousX, 2024-06-25:
Hi, that's a good thought. However, this does not explain why a hard reset works without problems despite unchanged customization. It seems to have a different cause, as a fallback to 12.3 also works again without customization.

My satellites are started 1 minute earlier than the master. Both satellites already have 12.4 installed, start at the same time and have the same integrations. The difference to the master is that the satellites are Raspberry Pi's 4 and have no problems.

But I will try that, thanks

VenominousX, 2024-06-26:
Update:
I have changed the network name of all of my satellites, installed the OS 12.4 on my master and the host does not reboot after the host_reboot command, so the error still exists.

sairon, 2024-06-27:
What does it mean "does not reboot completely"? What happens when you run the host_reboot service? Anything in Core or Host logs after that point? Does it behave correctly if you run `ha host reboot` in the terminal?

VenominousX, 2024-06-27:
Hi, after the upgrade from 12.3 to 12.4 the host starts as usual. If a reboot is initiated with 12.4, the host does not boot. It is noticeable that the status LEDs on the LAN port then remain off. The host also does not come up again via the terminal. Since a hard reset solves the problem, I suspect the reboot process under 12.4. As soon as I go back to 12.3, everything works as usual.
There are no anomalies in the home-assistant.log.1 or core log. However, since the host only boots after the hard reset, I don't know whether these statements are completely accurate.

[home-assistant_2024-06-27T11-08-32.266Z.log](https://github.com/user-attachments/files/16012515/home-assistant_2024-06-27T11-08-32.266Z.log)
[host_2024-06-27T11-14-03.724Z.log](https://github.com/user-attachments/files/16012516/host_2024-06-27T11-14-03.724Z.log)

VenominousX, 2024-06-27:
Looks like the computer stucks in grub

VenominousX, 2024-06-27:
There are 4 Options in Grub:
Slot A (OK=1 TRY=0)
Slot B (OK=1 TRY=1 - increments on boot, when not hard reseted)
Slot A (rescue shell
Slot B (rescue shell)

Slot A boots 12.3
Slot B boots 12.4

VenominousX, 2024-06-27:
I did a complete reinstallation with the 12.4 image. The problem still exists. Going back to 12.3

VenominousX, 2024-08-17:
Fixed with the OS version 13.0
Thank you very much :)

closed 2024-08-17 by VenominousX