home-assistant/operating-system #3447: Ethernet jack not working on fresh installation

OnSive, 2024-06-26:
### Describe the issue you are experiencing

I tried the raspberry installer and flashing it on my own (12.4/haos_rpi5-64-12.4.img.xz) but either way the ethernet jack does not work on my raspberry pi 5.

I tried raspberry os and the jack worked fine. Don't even know how I could access my raspberry in this state to debug further.

### What operating system image do you use?

rpi5-64  (Raspberry Pi 5 64-bit OS)

### What version of Home Assistant Operating System is installed?

12.4

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

Raspberry Pi 5 (with and without POE header)

### Steps to reproduce the issue

1. Get a Raspberry Pi 5
2. Install the latest Raspberry Home Assistant OS

### Anything in the Supervisor logs that might be useful for us?

```txt
Can't even reach the device
```

### Anything in the Host logs that might be useful for us?

```txt
Can't even reach the device
```

OnSive, 2024-06-26:
Just flashed version 12.3 and Ethernet is working (with POE header)

sairon, 2024-06-27:
If you used Windows for flashing the image, then it's almost surely the pinned issue (#3437).

OnSive, 2024-06-27:
@sairon Thanks, then it's probably this issue, thought that mine may be unrelated 👍

closed 2024-06-27 by OnSive