home-assistant/operating-system #3449: Cannot access HA after upgrade

doreiny, 2024-06-28:
### Describe the issue you are experiencing

Hi guys.
After the update from 2024.4 all went well, then after a couple of hours suddenly I cannot access HA, not from inside or outside.
If I reboot the PC off a couple of times then it works again for an half our and then again no access.
My HA runs on X86 directly.
PC confirmation:
Core I5 12 Gen.
16gb ram
512 ssd nvme.
[image]
[image]
[image]
[image]

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

2024.6

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

PC confirmation:
Core I5 12 Gen.
16gb ram
512 ssd nvme.

### Steps to reproduce the issue

Just let the system run

### Anything in the Supervisor logs that might be useful for us?

```txt
Don't have access to the system
```

### Anything in the Host logs that might be useful for us?

```txt
Don't have access
```

doreiny, 2024-06-29:
[inbound6819451688617370460.txt](https://github.com/user-attachments/files/16038616/inbound6819451688617370460.txt)

doreiny, 2024-06-29:
I managed to get more log as soon as it worked again
[inbound1926459041292440268.txt](https://github.com/user-attachments/files/16039736/inbound1926459041292440268.txt)
[inbound7615523733385297362.txt](https://github.com/user-attachments/files/16039741/inbound7615523733385297362.txt)
[inbound7433614066653038588.txt](https://github.com/user-attachments/files/16039743/inbound7433614066653038588.txt)
[inbound8454135636696586527.txt](https://github.com/user-attachments/files/16039744/inbound8454135636696586527.txt)

doreiny, 2024-06-30:
Can someone help please, I'm willing to pay to solve this issue

Jackman3005, 2024-07-03:
I believe this is the same issue I had today. It was so locked up I was not able to restart the VM within my esxi host and needed to shut down the whole machine..

closed 2024-10-09 by github-actions[bot]