home-assistant/operating-system #3450: Build does not support aarch64 native builder

claplace, 2024-06-29:
### Describe the issue you are experiencing

While working on a fix for #3247, I started an haos build on my own arm64 server. After 1+ hour of building it failed with the error:

```# Building C bootstrap tool.
cmd/dist
go tool dist: unknown architecture: aarch64
make[1]: *** [package/pkg-generic.mk:283: /build/output/build/host-go-bootstrap-stage1-1.4-bootstrap-20171003/.stamp_built] Error 1
make[1]: Leaving directory '/build/buildroot'
make: *** [Makefile:33: generic_aarch64] Error 2```

### What operating system image do you use?

generic-aarch64 (Generic UEFI capable aarch64 systems)

### What version of Home Assistant Operating System is installed?

dev

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

debian arm64 builder VM:
```
cyprien@haos-builder:~/operating-system$ neofetch
       _,met$$$$$gg.          cyprien@haos-builder
    ,g$$$$$$$$$$$$$$$P.       --------------------
  ,g$$P"     """Y$$.".        OS: Debian GNU/Linux 12 (bookworm) aarch64
 ,$$P'              `$$$.     Host: VMware20,1 1
',$$P       ,ggs.     `$$b:   Kernel: 6.1.0-21-arm64
`d$$'     ,$P"'   .    $$$    Uptime: 2 hours, 24 mins
 $$P      d$'     ,    $$P    Packages: 532 (dpkg)
 $$:      $$.   -    ,d$$'    Shell: bash 5.2.15
 $$;      Y$b._   _,d$P'      Resolution: 1024x768
 Y$$.    `.`"Y$$$$P"'         Terminal: /dev/pts/0
 `$$b      "-.__              CPU: (64)
  `Y$$                        GPU: 00:0f.0 VMware Device 0406
   `Y$$.                      Memory: 679MiB / 60158MiB
     `$$b.
       `Y$$b.
          `"Y$b._
              `"""
```

### Steps to reproduce the issue

on an arm64 system: `sudo scripts/enter.sh make generic_aarch64`

### Anything in the Supervisor logs that might be useful for us?

```txt
this is a build issue.
```

### Anything in the Host logs that might be useful for us?

```txt
this is a build issue.
```

claplace, 2024-06-29:
it looks like a buildroot issues, which has been discussed back in 2023, but has never made its way upstream.

https://patchwork.ozlabs.org/project/buildroot/patch/20230707185106.98208-1-ckhardin@gmail.com/

I've applied/updated the proposed patch and was able to pass the go build issue.

I'll close the issue because it's not a haos bug, but it's sad to not be able to build the operating system natively on arm64.

closed 2024-06-29 by claplace

claplace, 2024-07-04:
@sairon would you be interested in a patch for buildroot to use go1.19 native binaries as bootstrap, allowing native building of haos on arm64? I'm trying to make one, but if there's no interest I can drop it.

sairon, 2024-07-04:
@claplace You can try reopening the topic in upstream mailing list but it's definitely not something I want to carry in our fork only.

Monofin, 2024-09-25:
Looks like it has been [fixed in upstream](https://github.com/buildroot/buildroot/commit/7b2a164b7460c07471080fe565bf5da76b032e59), but just missed on the latest tag that was merged to the fork.
I'll test it as a patch on a local system, it may require a Dockerfile update to install local host go if building on aarch64. That might be a bit of an image size bump, but if it works, makes my life a lot easier!

sairon, 2024-09-25:
Backporting it to 2024.02.x would take a little more effort - the LTS branch contains 1.21.x and only security fixes are applied there, while the master uses 1.22.x already, so this patch would not apply cleanly. Even worse, there was even more refactoring done for the go package, so I'm not sure I want to introduce the maintenance burden for the following ~5 months before the next LTS is branched off.

But IIUC it shouldn't be needed to adjust the build environment's Dockerfile in any way (although it's not an issue if it were needed), the difference would be just the Buildroot would start using a pre-built binary of the host-go package instead of building it from sources.

Monofin, 2024-09-25:
My mistake here on the Dockerfile - I had assumed host-go required installation, when there are (as you say) the binary packages available now.

Comment updated above - I'm going to attempt to patch and build anyway, (mostly because I'm only on aarch64 boxes here... its true!)

claplace, 2024-09-25:
> Looks like it has been [fixed in upstream](https://github.com/buildroot/buildroot/commit/7b2a164b7460c07471080fe565bf5da76b032e59)

Amazing :-) thanks for sharing the info!

> I'm not sure I want to introduce the maintenance burden for the following ~5 months before the next LTS is branched off.

I completely understand and agree. I was able to patch my build to use that host go prebuilt package, so there is a solution for the – still – few of us running on aarch64 boxes.

Monofin, 2024-09-25:
I've successfully back ported (via cherry-pick) the (rather extensive) patch chain from upstream, and also tried (and updated) the patch mentioned above - both work, but both are definitely a bit kludgy.

The patch chain backport is the cleanest, and compatible with upstream, and does actually work, but is quite invasive in terms of the changes needed to the config and build process - possibly not worth it in terms of stability and testing etc, let alone a nice clean merge later on!

Happy to wait until the LTS branches :-)

rpardini, 2025-01-31:
Sorry for the necro. Trying to build hassos on arm64 host, late January'2025, and hitting the `go tool dist: unknown architecture: aarch64`. This is more generally relevant as GitHub has released arm64 GitHub Action runners recently.

sairon, 2025-01-31:
@rpardini Once Buildroot 2025.02 is released, we'll update HAOS to it, so the patches linked above will be available here soon(-ish).