home-assistant/operating-system #3453: Can't change default network adapter nor choose network adapter for WOL integration

StijnWolf, 2024-07-01:
### Describe the issue you are experiencing

I've been using home assistant for the past few years and love it. I use the Wake On Lan integration daily with the following configuration:

```
switch:
    - platform: wake_on_lan
      mac: "00:11:22:33:44:55"
      name: "Stijn PC"
      host: "192.168.x.x"
      turn_off:
        - service: shell_command.turn_off_stijn_desktop_windows
        - service: shell_command.turn_off_stijn_desktop_linux
shell_command:
  reboot_stijn_desktop_linux: "ssh -i /config/ssh_keys/id_rsa_homeassistant -o 'StrictHostKeyChecking=no' homeassistant@192.168.x.x sudo poweroff --reboot"
  reboot_stijn_desktop_windows: "ssh -i /config/ssh_keys/id_rsa_homeassistant -o 'StrictHostKeyChecking=no' rpc_user@192.168.x.x shutdown /r"
  turn_off_stijn_desktop_linux: "ssh -i /config/ssh_keys/id_rsa_homeassistant -o 'StrictHostKeyChecking=no' homeassistant@192.168.x.x sudo poweroff"
  turn_off_stijn_desktop_windows: "ssh -i /config/ssh_keys/id_rsa_homeassistant -o 'StrictHostKeyChecking=no' rpc_user@192.168.x.x shutdown /s"

```
This has been working perfectly fine, but I recently introduced an extra router to my setup, specifically for VR streaming. So the server is still connected to the other router, while the new router is connected to the old one and my gaming pc is connected to the new router; I put a new router between the gaming pc and old router.
This broke my WOL integration in HA since the gaming PC is now connected to a different subnet. As a countermeasure I decided to buy a USB Wi-Fi dongle and pass it to my HAOS VM. I'm able to connect my HA instance to the same router as my pc via the Wi-Fi dongle, but here's the kicker. Home Assistant doesn't seem to send the WOL packet to all network adapters, but only the default one. 

This means that while I've configured everything correctly, the package won't get sent to the right adapter. I searched around for changing the default network adapter using nmcli (like in this [thread](https://community.home-assistant.io/t/selector-under-settings-system-network-that-sets-default-adapter/419329)), but was then greeted with the following error:
```
Warning: nmcli (1.46.0) and NetworkManager (1.44.2) versions don't match. Restarting NetworkManager is advised.
Error: Failed to modify connection 'Supervisor enp1s0': connection.autoconnect-ports: unknown property
```
I also tried using the Wi-Fi adapter as the only network interface in HA, and low and behold, the magic packet gets sent properly! Unfortunately I need the other network adapters so this solution isn't viable for me, but simply confirmed my suspicions.

There are two possible solutions to my problem as far as I'm aware:

1. Being able to set the used network adapter in the Wake On Lan integration
2. Being able to change my default network adapter in Settings > System > Network.

The most frustrating part of this is knowing that this can work, but doesn't. Other devices (such as my smartphone) are able to send a WOL packet just fine to my gaming pc, so I know that this issue boils down to HA not sending the packet to the right destination.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

Home Assistant OS 12.4

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

HAOS vm running via QEMU/KVM with virt-manager.

Specs:

CPU: Intel I5 6600k
RAM: 16gb DDR4@2400mhz
Storage: 1TB m.2 ssd
GPU: none
PSU: 650W silverstone (don't know the model)

OS: Ubuntu Desktop 22.04

### Steps to reproduce the issue

1. Have a separate network with a device to be used with Wake On Lan
2. Add network adapter connected to said network to HA
3. Ensure the new adapter is not the default network adapter in Settings > System > Network
4. Try sending WOL packet to device 
5.  Sending WOL packet to device does not work
[... 281 lines cut ...]

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 3
views | 5
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | June 22, 2024 at 11:02 AM
-- | --
current_recorder_run | July 1, 2024 at 12:54 PM
estimated_db_size | 165.14 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

### Additional information

HA runs in a VM on a server running Ubuntu Desktop 22.04. My gaming PC and server are both hardwired to the same router. The gaming pc has a Windows (11) + Linux (Manjaro) dual boot, this doesn't seem to matter but should be mentioned nonetheless.

sairon, 2024-07-08:
Currently using a multi-homed network setup is something that's rather unsupported in HA, and enabling configurability of the default network adapter is something that would need changes in multiple layers of the stack. We might consider it in the future but it's not a viable solution for your issue.

On the other hand, the way around it could be adding option to specify interface where the magic packet should be sent from. This is something that is available in the underlying [wakeonlan](https://pypi.org/project/wakeonlan/) library used by the integration so if the integration allowed to specify it, it could be a solution to your problem.

StijnWolf, 2024-07-24:
I made an attempt to write a custom_component to add this feature to the existing wakeonlan integration. Sadly I couldn't get the interface parameter in the [wakeonlan](https://pypi.org/project/wakeonlan/)  library used by the integration to work. It'd succesfully execute the command but my target machine would still not receive a magic packet. I succesfully added a copy of the integration to my home assistant instance inside the custom_components folder, so I was able to add custom logic to the existing integration locally. Sadly the wakeonlan integration just didn't want to recognize my second network interface. I tried passing a hardcoded value for this parameter to no avail.

AFAIK this issue can't be supported by home assistant. I would've loved to make a merge request with my fix but the problem doesn't seem to be fixed by adding the interface parameter to the [wakeonlan](https://pypi.org/project/wakeonlan/) calls within the wakeonlan home assistant integration.

I confirmed my suspicions by writing a python script which uses [wakeonlan](https://pypi.org/project/wakeonlan/) on the server which runs the home assistant vm, and that worked as expected; the target machine receives the magic packet, but only if I add the interface parameter. This made me give up on the wakeonlan integration because I just couldn't figure out another probable cause than adding the interface parameter after this test.

I instead opted for a custom switch with very similar behavior using a custom binary sensor. My HA vm uses SSH to connect to the server hosting the HA vm, and then the server itself sends the magic packet to my target machine. This gives me the desired behavior for a wakeonlan switch. I'm happy I was able to build a solution myself, but am disappointed that it doesn't make use of the seemingly fitting wakeonlan integration.

I added my configuration as a file should it be of any use

[wolpc.txt](https://github.com/user-attachments/files/16361609/wolpc.txt)

I suppose this issue can be closed since I don't think this issue can be resolved within the wakeonlan integration and I've provided an alternative for my use case.

closed 2024-08-02 by sairon