home-assistant/operating-system #3455: Official fan is not working on Khadas Vim3

Nicooow, 2024-07-02:
### Describe the issue you are experiencing

The official fan does not work on Khadas VIM3. It never turns on, and no commands seem to activate it. It appears that a driver is missing in the OS.

I have tried everything I could find, but my Linux skills are too limited to make any progress.
If needed, I have two VIM3 devices available with Home Assistant for testing.

Here is the doc : https://docs.khadas.com/products/sbc/vim3/configurations/cooling-fan 

[image]
[image]

### What operating system image do you use?

khadas-vim3 (Khadas VIM3)

### What version of Home Assistant Operating System is installed?

12.4

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

Khadas vim3 with 3705 Cooling Fan (https://www.khadas.com/product-page/3705-cooling-fan)

### Steps to reproduce the issue

Plug the fan

### Anything in the Supervisor logs that might be useful for us?

```txt
no
```

### Anything in the Host logs that might be useful for us?

```txt
no
```

LaZyCraZy71, 2024-09-21:
hi there,

any update from your issue ?

closed 2024-11-11 by Nicooow

LaZyCraZy71, 2024-11-11:
nice! thanks for your great work !! 👍