home-assistant/operating-system #3459: Raspberry Pi Image is broken

Krispkiwi, 2024-07-04:
### Describe the issue you are experiencing

When loading in with a fresh install on a Rpi 4, you can get to the login/create account screen however upon pressing create account, nothing happens. Other people have reported the same issue within the discord server as well. Thread ID 1258250777470959636

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

Latest OS

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

Raspberry Pi 4

### Steps to reproduce the issue

1. Install home assistant on a raspberry pi 4
2. Get to the login screen
3. Try to create account

### Anything in the Supervisor logs that might be useful for us?

```txt
Can't get this far
```

### Anything in the Host logs that might be useful for us?

```txt
Again I don't get this far
```

Impact123, 2024-07-05:
Proper link to thread: https://discord.com/channels/330944238910963714/1258250777470959636

Krispkiwi, 2024-07-05:
> That thread id is kinda useless. I assume you meant this thread? https://discord.com/channels/330944238910963714/1258250777470959636

Yeah, It's not the end of the world for me, I have backups i can get in from but for fresh installs its problematic

sairon, 2024-07-08:
Per the linked discussion, the problem is not the OS. I've created a frontend issue for tracking the root cause: https://github.com/home-assistant/frontend/issues/21333

closed 2024-07-08 by sairon