home-assistant/operating-system #3472: Faulty USB drivers on aarch64 VM images?

pacco788, 2024-07-13:
### Describe the issue you are experiencing

Upgraded my Homeserver from Mac mini Intel to Mac mini M2. Its used for a Debian VM, surveillance station, PLEX, tvheadend server, Homebridge and for Home Assistant.

5 USB Devices for EnOcean, Z-Wave, Zigbee and two energy meters are being forwarded to the VM. Was using VirtualBox on the Intel machine, now UTM on Apple Silicon. When USB Devices are attached (no matter which and how many) to the VM the machine almost instantly crashes with this failure on UTM `"Assertion failed: (p->stream |I QTAILQ_FIRST&ep->queue) == p), function usb_packet_complete_one, file core.c, line 470."`. I also tried VMware Fusion and Parallels. The VM keeps running unter VMware and Parallels, but XHCI-Errors are show on the Home Assistant CLI screen.

`xhci_hcd 0000 error failed finding new dequeue state`
and
```
xhci_hcd 0000:13:00.0: Error Transfer event TRB DMA prt not part of current TD ep_index 2 comp_code 1
xhci_hcd 0000:13:00.0: Error Transfer event TRB DMA prt not part of current TD ep_index 2 comp_code 1
xhci_hcd 0000:13:00.0: Error Transfer event TRB DMA prt not part of current TD ep_index 2 comp_code 1

```

Everything seems to work for some hours, till the alert "Running out of memory" appears and the VM stops running.

Tried to run the OVA x64-image on UTM with x86_64 emulation. This seems to work quite well for now, no XHCI errors, no memory failures. But very slow, high CPU load and high energy consumption.

I already tested the setup some months ago with my MacBook Pro M2. This was running nicely. Tested it again with the latest aarch64-image on the same MacBook shows XHCI-errors/quits UTM.

Tried to run the VM on the mini and MacBook with an old aarch image shows no alert messages.

### What operating system image do you use?

generic-aarch64 (Generic UEFI capable aarch64 systems)

### What version of Home Assistant Operating System is installed?

12.4

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

Mac mini M2, 16 GB with forwarded USB Dongles to VM
UTM/QEMU-Backend

Sonoma 14.5

### Steps to reproduce the issue

1. Load aarch64 image to UTM
2. Start the VM
3. Attach any USB Device

### Anything in the Supervisor logs that might be useful for us?

```txt
---
```

### Anything in the Host logs that might be useful for us?

```txt
2024-07-13 22:18:00.247 homeassistant kernel: xhci_hcd 0000:12:00.0: Error: Failed finding new dequeue state
2024-07-13 22:18:00.248 homeassistant kernel: xhci_hcd 0000:12:00.0: ERROR Transfer event TRB DMA ptr not part of current TD ep_index 4 comp_code 6
2024-07-13 22:18:00.248 homeassistant kernel: xhci_hcd 0000:12:00.0: Looking for event-dma 0000000083c2c890 trb-start 0000000083c2c870 trb-end 0000000083c2c870 seg-start 0000000083c2c000 seg-end 0000000083c2cff0
```

sairon, 2024-07-23:
I don't have a Mac to test this on, but the UTM error seems to be a bug in UTM itself, see also https://github.com/utmapp/UTM/issues/5949

For the other hypervisors, can you narrow down which OS version started to have this issue? It is probably an upstream kernel regression, so knowing which version broke it could help a lot. It'd be also worth trying the [latest dev build](https://os-artifacts.home-assistant.io/index.html) which contains the latest 6.6 kernel.

pacco788, 2024-07-24:
So I tried the dev build with UTM: same error, in Parallels: 4 XHCI errors show up right after startup but CLI stays clear for a day now, memory usage is quite high compared to emulated x68 but not going up.

Will test some aarch64 images and report.

Thank you so far!

pacco788, 2024-08-22:
Didn't have the opportunity to test the older images. Sorry about that.

VM on Parallels just shows some errors right after startup, but EnOcean wall switches (via EnOcean USB Stick) didn't respond at all or were very laggy. Zwave, Zigbee and the energy-meters were reliable.

Tested the VM on the latest VirtualBox for aarch64 developer preview. No errors, no problems adding USB devices on startup, no problems with EnOcean. The machine is running for 1,5 weeks now.

Problem is solved. Will close the issue.

Seems to be an issue with UTM/Parallels.

closed 2024-08-22 by pacco788