home-assistant/operating-system #3474: ASUS NUC 14 Pro will not boot

tane64, 2024-07-14:
### Describe the issue you are experiencing

Home Assistant OS 12.4 Will not boot on a new ASUS NUC 14
Booting  ‘Slot A (OK=0 TRY+0)

System will freeze after that

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

12.4

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

Asus NUC 14 Pro Intel 125, 16 GB Memory 4 TB SSD NVME

### Steps to reproduce the issue

Fresh install

Ubuntu and Proxmox will install and boot fine. 

### Anything in the Supervisor logs that might be useful for us?

```txt
no
```

### Anything in the Host logs that might be useful for us?

```txt
Botsing ‘Slot A (OK=0 TRY+0)

System  Will freeze
```

sairon, 2024-07-23:
Can you please test if the [latest dev build](https://os-artifacts.home-assistant.io/13.0.dev20240722/haos_generic-x86-64-13.0.dev20240722.img.xz) (as of today) boots correctly? Could be fixed by 63a05a5a1ec6b8c8f49cb3e2812894087a52d01a.

tane64, 2024-07-23:
Same result. Will not boot
After Booting Slot A system will freeze

sairon, 2024-07-23:
If there nothing else shown after selecting the boot slot, GRUB is likely failing to load kernel. It can be a bug in the UEFI firmware or GRUB itself. HAOS 12.2 brought GRUB 2.12 that hasn't made it way into most of the current mainstream distros, and for example in case of Ubuntu, it's heavily patched. Can you please also check if [HAOS 12.1](https://github.com/home-assistant/operating-system/releases/tag/12.1) doesn't behave differently by any chance?

tane64, 2024-07-24:
Same result with HAOS 12.1
Also downgraded the Bios firmware to
38
39
41
42
43

All same result after after GRUB menu it will freeze

riemers, 2024-08-03:
[image]

Playing with these settings made it boot on my side. In particular the intel VT for directed I/O (VT-d) needs to be unticked. I have verified this with opmised bios settings, then disabling secure boot and unticking above named item.

sairon, 2024-08-05:
Nice find, @riemers! Although toggling VT-d shouldn't make any difference if HAOS boots or not, disabling it doesn't do any harm, because HAOS doesn't make any use of it. Disabling secure boot is mandatory though.

sairon, 2024-08-08:
Unfortunately, without the device on hand, it's shooting in the dark. It could also be the other way, because HAOS already uses GRUB 2.12 bootloader, which is not yet in most of the others (or not heavily unpatched).

The only other idea I have right now is to check if it really fails to boot or just the graphics driver doesn't correctly initialize. To test that, re-enable the BIOS option and check if HA become reachable after some time.

riemers, 2024-08-08:
Thought about that too, but never noticed an ip after longer waiting. Also the keyboard locks up too. So for example caps lock lights don’t work anymore either.

themajesticking, 2024-08-12:
> ![WhatsApp Image 2024-08-03 at 20 37 12_738e4e07](https://private-user-images.githubusercontent.com/682041/354836878-f0c092d4-8412-46c3-8a4c-1a2ba1c370fa.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjM0MjE3MjIsIm5iZiI6MTcyMzQyMTQyMiwicGF0aCI6Ii82ODIwNDEvMzU0ODM2ODc4LWYwYzA5MmQ0LTg0MTItNDZjMy04YTRjLTFhMmJhMWMzNzBmYS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODEyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgxMlQwMDEwMjJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wNWViYTUzYTJlMjZiMmE3OGM0NjdmYzcxYjJkYTMyYTExNjJlMTg0YTY4ZjBkY2ZhODg0YWIxZGJkNTM5M2RjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.3p7xlh7SaDoxmWg7WXNI1SSXSDWef0ui0EXyPIbLUgM)
> 
> Playing with these settings made it boot on my side. In particular the intel VT for directed I/O (VT-d) needs to be unticked. I have verified this with opmised bios settings, then disabling secure boot and unticking above named item.

OMGG I've been fighting with this for 4 hours on my NUC 14 Pro and this was the fix. I ticked/unticked the boxes in this photo and the "Booting ‘Slot A (OK=0 TRY+0)" flashed for a second and HA finally completed its first boot. Thanks!

xeor, 2024-09-03:
Confirming this solved the exact same issue I had on NUC 14 as well. There should be a "known fixes" section in some docs somewhere :)

Deckoz2302, 2024-11-18:
I'd argue that disabling VT-D is not a fix, as it disables the iGPU on Ultra 5 125H, making it unable to pass through the iGPU for hardware accelerated decoding of streams within Frigate Add-On. 

Sure it makes home assistant OS boot, but disabling hardware for I/O pass through in the process.

alvinchen1, 2024-11-18:
Confirming that this resolved issue getting Home Assistant installed on Bee-Link GTi14 

BIOS - Press F7
	Advanced - CPU Configuration - Intel (VMX) Virtualization Technology - from Enabled to Disabled
	Security - Secure Boot - Secure Boot - from Enabled to Disabled
	Save and Exit

waypts, 2024-11-27:
**RIEMERS POST SOLVED THIS ISSUE!!!!**  I had this exact problem trying to run Home Assistant on the **Asus NUC 14 Pro+**. I followed the Home Assistant Setup x86 Secondary Instructions. **Under the BIOS Security Tab, be sure to UNCHECK the Intel VT for Directed I/O (VT-d) option.** I spent hours trying to figure out why Home Assistant wouldn't run and would freeze. I appreciate Reimer's post. Home Assistant either needs to add this to their install instructions or troubleshooting notes or fix this issue. This was a pain! 

> ![WhatsApp Image 2024-08-03 at 20 37 [...]

closed 2025-06-09 by github-actions[bot]

JerJetta, 2025-07-08:
I want to use a new Dell Pro Micro Plus QBM1250 with an Intel Core Ultra 5 245 processor. I just can't boot HAOS. I want to use this PC to run Frigate, because this PC has a 13 TOPS NPU. But with this PC it just can't.