home-assistant/operating-system #3477: HA OS (generic x86_64) any OS beyond 12.1 will lose WiFi (Intel(R) AX210)

bearhntr, 2024-07-15:
### Describe the issue you are experiencing

I have successfully been running HA OS 12.1 on my HP T620 for quite some time.  I have upgraded to 12.2, 12.3 and now 12.4.  All of them fail to see the WiFi on the Intel(R) AX210 (WiFi 6E/BT) card (Bluetooth still works).  As soon as I drop the OS back to 12.1, the WiFi is seen again and I get a WLP1S0 adapter to connect to my home WiFi.  

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

6.6.20-haos

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

HP T620 with Intel(R) Wi-Fi 6 AX210 card installed (WiFi 6E and Bluetooth) 

### Steps to reproduce the issue

1.  Run INSTALL on the OS Upgrade
2.  Reboot
3.  WiFi adapter is missing.
4.   Revert OS to 12.1 and WiFi comes back
...

### Anything in the Supervisor logs that might be useful for us?

```txt
[32m2024-07-15 16:55:26.438 INFO (MainThread) [supervisor.addons.manager] Found 7 installed add-ons[0m
[32m2024-07-15 16:55:26.528 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/ssh/amd64 with version 18.0.0[0m
[32m2024-07-15 16:55:26.543 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/amd64-addon-configurator with version 5.8.0[0m
[32m2024-07-15 16:55:26.559 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/log-viewer/amd64 with version 0.17.0[0m
[32m2024-07-15 16:55:26.570 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/amd64-addon-mariadb with version 2.7.1[0m
[32m2024-07-15 16:55:26.582 INFO (MainThread) [supervisor.docker.interface] Attaching to homeassistant/amd64-addon-samba with version 12.3.1[0m
[32m2024-07-15 16:55:26.598 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/vscode/amd64 with version 5.15.0[0m
[32m2024-07-15 16:55:26.611 INFO (MainThread) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/zwave-js-ui/amd64 with version 3.8.5[0m
[32m2024-07-15 16:55:26.837 INFO (MainThread) [supervisor.backups.manager] Found 11 backup files[0m
[32m2024-07-15 16:55:26.896 INFO (MainThread) [supervisor.discovery] Loaded 1 messages[0m
[32m2024-07-15 16:55:26.896 INFO (MainThread) [supervisor.ingress] Loaded 1 ingress sessions[0m
[32m2024-07-15 16:55:26.896 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state setup[0m
[32m2024-07-15 16:55:26.897 INFO (MainThread) [supervisor.resolution.checks.base] Run check for disabled_data_disk/system[0m
[32m2024-07-15 16:55:26.897 INFO (MainThread) [supervisor.resolution.checks.base] Run check for detached_addon_missing/addon[0m
[32m2024-07-15 16:55:26.897 INFO (MainThread) [supervisor.resolution.checks.base] Run check for detached_addon_removed/addon[0m
[32m2024-07-15 16:55:26.898 INFO (MainThread) [supervisor.resolution.checks.base] Run check for multiple_data_disks/system[0m
[32m2024-07-15 16:55:26.898 INFO (MainThread) [supervisor.resolution.check] System checks complete[0m
[32m2024-07-15 16:55:26.898 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state setup[0m
[32m2024-07-15 16:55:26.899 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete[0m
[32m2024-07-15 16:55:26.900 INFO (MainThread) [supervisor.jobs] 'ResolutionFixup.run_autofix' blocked from execution, system is not running - setup[0m
[32m2024-07-15 16:55:26.900 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state setup[0m
[32m2024-07-15 16:55:26.901 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete[0m
[32m2024-07-15 16:55:26.902 INFO (MainThread) [__main__] Running Supervisor[0m
[32m2024-07-15 16:55:26.919 INFO (MainThread) [supervisor.os.manager] Rauc: B - marked slot kernel.1 as good[0m
[32m2024-07-15 16:55:26.946 INFO (MainThread) [supervisor.addons.manager] Phase 'initialize' starting 0 add-ons[0m
[32m2024-07-15 16:55:26.953 INFO (MainThread) [supervisor.addons.manager] Phase 'system' starting 2 add-ons[0m
[32m2024-07-15 16:55:26.983 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mariadb_pass[0m
[32m2024-07-15 16:55:26.984 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mariadb_user[0m
[32m2024-07-15 16:55:26.984 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mariadb_user[0m
[32m2024-07-15 16:55:26.993 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mariadb_pass[0m
[32m2024-07-15 16:55:26.994 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mariadb_user[0m
[32m2024-07-15 16:55:27.994 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mariadb_user[0m
[32m2024-07-15 16:55:27.625 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on homeassistant/amd64-addon-mariadb with version 2.7.1[0m
[32m2024-07-15 16:55:28.281 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on ghcr.io/hassio-addons/zwave-js-ui/amd64 with version 3.8.5[0m
[33m2024-07-15 16:55:29.252 WARNING (MainThread) [supervisor.services.modules.mysql] Can't remove not exists services[0m
[32m2024-07-15 16:55:29.671 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mariadb_pass[0m
[32m2024-07-15 16:55:29.671 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mariadb_user[0m
[32m2024-07-15 16:55:29.671 INFO (MainThread) [supervisor.homeassistant.secrets] Request secret mariadb_user[0m
[32m2024-07-15 16:55:30.411 INFO (MainThread) [supervisor.services.modules.mysql] Set core_mariadb as service provider for MySQL[0m
[... 227 lines cut ...]
version_api | ok
installed_addons | Advanced SSH & Web Terminal (18.0.0), File editor (5.8.0), Log Viewer (0.17.0), MariaDB (2.7.1), Samba share (12.3.1), Studio Code Server (5.15.0), Z-Wave JS UI (3.8.5)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 3
views | 6
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | July 5, 2024 at 17:15
-- | --
current_recorder_run | July 15, 2024 at 16:56
estimated_db_size | 273.55 MiB
database_engine | mysql
database_version | 10.11.6

</details>

bearhntr, 2024-07-15:
This is what I see after 12.4 update:

[image]

This is after reversion to 12.1

[image]

Attached the HOST logs after reversion to 12.1
[12-1_host_2024-07-15T21-09-12.592Z.log](https://github.com/user-attachments/files/16240358/12-1_host_2024-07-15T21-09-12.592Z.log)

sairon, 2024-07-23:
The logs don't contain enough information to see what went wrong, could you please send the complete logs of the current boot, i.e. output of  `ha host logs -n2000 -b0`? Ideally reboot the host first to make sure there's just the information we need and no clutter from the running system.

bearhntr, 2024-07-23:
> The logs don't contain enough information to see what went wrong, could you please send the complete logs of the current boot, i.e. output of `ha host logs -n2000 -b0`? Ideally reboot the host first to make sure there's just the information we need and no clutter from the running system.

Please elaborate-- do you want me to UPGRADE again to 12.4 or do you want logs from 12.1 (using the command you mention) - or do you want both?   I take it that I should collect these as soon after a reboot as possible?

sairon, 2024-07-23:
The logs should be from the version that doesn't work, so 12.4 in your case. And yes, run the command shortly after it boots. You don't necessarily need to _upgrade_ again, if `ha os info` shows the 12.4 version for the other boot slot that the currently booted one, you can also simply switch to it using `ha os boot-slot other` (and then do the same to revert the "upgrade").

bearhntr, 2024-07-26:
> The logs should be from the version that doesn't work, so 12.4 in your case. And yes, run the command shortly after it boots. You don't necessarily need to _upgrade_ again, if `ha os info` shows the [...]

I will pull the logs - if you can advise how to get them.   If I go into TERMINAL from the HA GUI (web page) and issue that command --- they scroll off the screen and I cannot scroll up to copy/paste them.  

I want to pull 2 sets - one as current 12.1 after a fresh reboot - then upgrade to 12.4 which appears to be the latest version out there - then pull them again.

bearhntr, 2024-07-26:
Dropped it back to 12.1 - and WiFi is back

[image]

[image]

bearhntr, 2024-08-07:
Any updates?  Do you need anything more from me?

bearhntr, 2024-08-16:
@sairon 

Looks like there is now an OS 13.0   -- does this fix the WiFi issue with the INTEL AX210?

sairon, 2024-08-19:
@bearhntr Sorry for leaving you hanging. The log files have been useful, I will prepare a fix and try to squeeze it in the 13.1 release.

bearhntr, 2024-08-19:
> @bearhntr Sorry for leaving you hanging. The log files have been useful, I will prepare a fix and try to squeeze it in the 13.1 release.

Pleased just let me know.  I am not doing any upgrades at this time until this part is fixed.  I am still running (see image) - as I am trying to build a completely new system which will use the same WiFi/BT card but under Proxmox.  I have it built...and it does the same thing when I upgrade it beyond OS 12.1.

[image]

commit d3a43a4c (2024-08-21): Fix iwlwifi driver failing to find .pnvm files for Intel 6E (AX210) (#3549)

Since updating to Buildroot 2024.02, the iwlwifi loads a different version of
firmware for Intel AX cards (reported on AX210) which also needs the pnvm file.
However, unlike firmwares, the load method is different and the driver can't
load a compressed file. Disable compression for all .pnvm files to fix this.

* buildroot baa16784d2...55be56d521 (1):
  > package/linux-firmware: exclude some files from compression

Fixes #3477
commit 1a7dc0ca (2024-08-21): Fix iwlwifi driver failing to find .pnvm files for Intel 6E (AX210) (#3549)

Since updating to Buildroot 2024.02, the iwlwifi loads a different version of
firmware for Intel AX cards (reported on AX210) which also needs the pnvm file.
However, unlike firmwares, the load method is different and the driver can't
load a compressed file. Disable compression for all .pnvm files to fix this.

* buildroot baa16784d2...55be56d521 (1):
  > package/linux-firmware: exclude some files from compression

Fixes #3477

(cherry picked from commit d3a43a4ca4fcdcb4b0ba9a6017c3340773011166)
PR #3556 (2024-08-23): Fix missing Intel AX210 firmware files, remove unused firmwares
Fix AX210 firmware files, this time for real. In #3549 we still had only ucode and pnvm files for AX211, AX210 was still missing the pnvm file, because its ucode was provided through IWLWIFI_22000 without appropriate pnvm file. Both AX210 and AX211 firmwares are now installed with IWLWIFI_6E which includes both pnvm and ucode files.

Also some firmwares which are not used by the current kernel can be removed, because the kernel always only loads the most recent one.

* buildroot 01188d9c38...a2c10a16a0 (2):
  > Revert "package/lin [... cut]
commit c6d4e278 (2024-08-23): Fix missing Intel AX210 firmware files, remove unused firmwares (#3556)

Fix AX210 firmware files, this time for real. In #3549 we still had only
ucode and pnvm files for AX211, AX210 was still missing the pnvm file,
because its ucode was provided through IWLWIFI_22000 without appropriate
pnvm file. Both AX210 and AX211 firmwares are now installed with
IWLWIFI_6E which includes both pnvm and ucode files.

Also some firmwares which are not used by the current kernel can be
removed, because the kernel always only loads the most recent one.

* buildroot 01188d9c38...a2c10a16a0 (2):
  > Revert "pa [... cut]
closed 2024-08-21 by sairon