home-assistant/operating-system #3479: Error installing Home Assistant within Synology DSM7

slszmgl, 2024-07-18:
### Describe the issue you are experiencing

Hi, I import Home Assistant OVI file into Synology DSM7 Virtual Machine Manager. When I fist startup and trying to setup, it showed "Error installing Home Assistant" on web.

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

cannot find, download on 18th, July 2024

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

DS1621+ Snology

### Steps to reproduce the issue

1. Import OVI file.
2. Fire up virtual machine.
3. connnect with https://localip:8123
4. "Error installing Home Assistant"
5. The log is below
...

### Anything in the Supervisor logs that might be useful for us?

```txt
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/udev.sh
[00:27:56] INFO: Using udev information from host
cont-init: info: /etc/cont-init.d/udev.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun supervisor (no readiness notification)
services-up: info: copying legacy longrun watchdog (no readiness notification)
[00:27:56] INFO: Starting local supervisor watchdog...
s6-rc: info: service legacy-services successfully started
2024-07-18 00:28:00.709 INFO (MainThread) [__main__] Initializing Supervisor setup
2024-07-18 00:28:00.730 INFO (MainThread) [supervisor.docker.network] Can't find Supervisor network, creating a new network
2024-07-18 00:28:01.055 INFO (MainThread) [supervisor.bootstrap] Setting up coresys for machine: qemux86-64
2024-07-18 00:28:01.061 INFO (MainThread) [supervisor.docker.supervisor] Attaching to Supervisor ghcr.io/home-assistant/amd64-hassio-supervisor with version 2024.06.0
2024-07-18 00:28:01.061 INFO (MainThread) [supervisor.docker.supervisor] Connecting Supervisor to hassio-network
2024-07-18 00:28:01.335 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state initialize
2024-07-18 00:28:01.337 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
2024-07-18 00:28:01.337 INFO (MainThread) [__main__] Setting up Supervisor
2024-07-18 00:28:01.974 INFO (MainThread) [supervisor.api] Starting API on 172.30.32.2
2024-07-18 00:28:02.052 INFO (MainThread) [supervisor.hardware.monitor] Started Supervisor hardware monitor
2024-07-18 00:28:02.054 INFO (MainThread) [supervisor.dbus.manager] Connected to system D-Bus.
2024-07-18 00:28:02.055 INFO (MainThread) [supervisor.dbus.agent] Load dbus interface io.hass.os
2024-07-18 00:28:02.055 INFO (MainThread) [supervisor.dbus.hostname] Load dbus interface org.freedesktop.hostname1
2024-07-18 00:28:02.055 INFO (MainThread) [supervisor.dbus.logind] Load dbus interface org.freedesktop.login1
2024-07-18 00:28:02.056 INFO (MainThread) [supervisor.dbus.network] Load dbus interface org.freedesktop.NetworkManager
2024-07-18 00:28:02.056 INFO (MainThread) [supervisor.dbus.rauc] Load dbus interface de.pengutronix.rauc
2024-07-18 00:28:02.056 INFO (MainThread) [supervisor.dbus.resolved] Load dbus interface org.freedesktop.resolve1
2024-07-18 00:28:02.056 INFO (MainThread) [supervisor.dbus.systemd] Load dbus interface org.freedesktop.systemd1
2024-07-18 00:28:02.056 INFO (MainThread) [supervisor.dbus.timedate] Load dbus interface org.freedesktop.timedate1
2024-07-18 00:28:02.250 INFO (MainThread) [supervisor.host.services] Updating service information
2024-07-18 00:28:02.255 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
2024-07-18 00:28:02.261 WARNING (SyncWorker_0) [supervisor.host.sound] Can't update PulseAudio data: Failed to connect to pulseaudio server
2024-07-18 00:28:02.963 INFO (MainThread) [supervisor.host.network] Updating local network information
2024-07-18 00:28:03.008 INFO (MainThread) [supervisor.host.apparmor] Loading AppArmor Profiles: {'hassio-supervisor'}
[... 193 lines cut ...]
2024-07-18 08:31:03.704 INFO (MainThread) [supervisor.resolution.module] Create new suggestion execute_remove - store / 5c53de3b
2024-07-18 08:31:03.704 INFO (MainThread) [supervisor.resolution.module] Create new issue fatal_error - store / 5c53de3b
2024-07-18 08:31:03.708 INFO (MainThread) [supervisor.store] Loading add-ons from store: 0 all - 0 new - 0 remove
2024-07-18 08:31:03.708 INFO (MainThread) [supervisor.addons.manager] Found 0 installed add-ons
2024-07-18 08:31:03.710 INFO (MainThread) [supervisor.backups.manager] Found 0 backup files
2024-07-18 08:31:03.711 INFO (MainThread) [supervisor.discovery] Loaded 0 messages
2024-07-18 08:31:03.711 INFO (MainThread) [supervisor.ingress] Loaded 0 ingress sessions
2024-07-18 08:31:03.711 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state setup
2024-07-18 08:31:03.711 INFO (MainThread) [supervisor.resolution.checks.base] Run check for multiple_data_disks/system
2024-07-18 08:31:03.711 INFO (MainThread) [supervisor.resolution.checks.base] Run check for detached_addon_removed/addon
2024-07-18 08:31:03.711 INFO (MainThread) [supervisor.resolution.checks.base] Run check for detached_addon_missing/addon
2024-07-18 08:31:03.711 INFO (MainThread) [supervisor.resolution.checks.base] Run check for disabled_data_disk/system
2024-07-18 08:31:03.711 INFO (MainThread) [supervisor.resolution.check] System checks complete
2024-07-18 08:31:03.712 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state setup
2024-07-18 08:31:03.712 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
2024-07-18 08:31:03.712 INFO (MainThread) [supervisor.jobs] 'ResolutionFixup.run_autofix' blocked from execution, system is not running - setup
2024-07-18 08:31:03.712 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state setup
2024-07-18 08:31:03.712 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
2024-07-18 08:31:03.713 INFO (MainThread) [__main__] Running Supervisor
2024-07-18 08:31:03.806 INFO (MainThread) [supervisor.os.manager] Rauc: A - marked slot kernel.0 as good
2024-07-18 08:31:03.816 INFO (MainThread) [supervisor.supervisor] Fetching AppArmor profile https://version.home-assistant.io/apparmor_stable.txt
2024-07-18 08:31:04.496 INFO (MainThread) [supervisor.host.apparmor] Adding/updating AppArmor profile: hassio-supervisor
2024-07-18 08:31:04.602 INFO (MainThread) [supervisor.supervisor] Update Supervisor to version 2024.06.2
2024-07-18 08:31:04.603 INFO (MainThread) [supervisor.docker.interface] Downloading docker image ghcr.io/home-assistant/amd64-hassio-supervisor with tag 2024.06.2.
```

t1214275321, 2024-07-18:
我也是  怎么解决？

alientim, 2024-08-02:
Check the Connection?!
`curl 28 Failed to connect to github.com port 443 after 135289 ms: Couldn't connect to server`
The Problem is here that u cant curl from Github it seems ur VM has no External Access to the Internet

sairon, 2024-08-02:
Indeed, it seems the VM is not connected to the internet. Do other VMs created the same way have internet connectivity I'm not familiar with Synology VMM but can you connect to the VM's console and run `ha host logs` and `ha network info` there?

slszmgl, 2024-08-02:
Thank you for all the comments. The Synology device is physically located in mainland China, and the VM couldn't connect to GitHub for some reason. When I tried the same installation method on another familiar device outside mainland China, the problem didn't occur.

sairon, 2024-08-05:
I see, it is true that accessing Github can be problematic from Chinese internet. This is almost definitely a networking issue the OS can't affect.

closed 2024-08-05 by sairon