home-assistant/operating-system #3489: Problem with disk space

devbysb, 2024-07-23:
### Describe the issue you are experiencing

Hi, I have a problem with disk space. 
Home assistant show 95% used with 0.3GB free (hardisk 50GB)
I have some difficulty to show the disk usage.

When I make df -h
[image]

When I make du -a /dir/ | sort -n -r | head -n 20 to show biggest file
[image]

I make my backup on samba server, I have deleted all backups to see if disk space change but nothing.

I don't know where the disk space go... I have some addons but I can't show the disk usage.

Thank your help

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

12.4

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

I use virtual box 6.1

### Steps to reproduce the issue

1. df -h to show disk usage
2. du -ah to show directory size
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
2024-07-23 16:36:23.658 INFO (MainThread) [supervisor.resolution.checks.base] Run check for security/core
2024-07-23 16:36:23.659 INFO (MainThread) [supervisor.resolution.checks.base] Run check for multiple_data_disks/system
2024-07-23 16:36:23.660 INFO (MainThread) [supervisor.resolution.check] System checks complete
2024-07-23 16:36:23.660 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state running
2024-07-23 16:36:23.926 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
2024-07-23 16:36:23.926 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state running
2024-07-23 16:36:23.926 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
2024-07-23 16:37:27.649 WARNING (MainThread) [supervisor.addons.options] Option 'anonymous' does not exist in the schema for Mosquitto broker (core_mosquitto)
2024-07-23 16:42:27.674 WARNING (MainThread) [supervisor.addons.options] Option 'anonymous' does not exist in the schema for Mosquitto broker (core_mosquitto)
2024-07-23 16:43:16.540 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
2024-07-23 16:47:27.637 WARNING (MainThread) [supervisor.addons.options] Option 'anonymous' does not exist in the schema for Mosquitto broker (core_mosquitto)
2024-07-23 16:52:27.633 WARNING (MainThread) [supervisor.addons.options] Option 'anonymous' does not exist in the schema for Mosquitto broker (core_mosquitto)
2024-07-23 16:57:27.665 WARNING (MainThread) [supervisor.addons.options] Option 'anonymous' does not exist in the schema for Mosquitto broker (core_mosquitto)
2024-07-23 17:02:27.656 WARNING (MainThread) [supervisor.addons.options] Option 'anonymous' does not exist in the schema for Mosquitto broker (core_mosquitto)
2024-07-23 17:07:27.656 WARNING (MainThread) [supervisor.addons.options] Option 'anonymous' does not exist in the schema for Mosquitto broker (core_mosquitto)
2024-07-23 17:12:27.663 WARNING (MainThread) [supervisor.addons.options] Option 'anonymous' does not exist in the schema for Mosquitto broker (core_mosquitto)
2024-07-23 17:13:17.144 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
2024-07-23 17:17:27.644 WARNING (MainThread) [supervisor.addons.options] Option 'anonymous' does not exist in the schema for Mosquitto broker (core_mosquitto)
2024-07-23 17:22:27.662 WARNING (MainThread) [supervisor.addons.options] Option 'anonymous' does not exist in the schema for Mosquitto broker (core_mosquitto)
2024-07-23 17:27:27.668 WARNING (MainThread) [supervisor.addons.options] Option 'anonymous' does not exist in the schema for Mosquitto broker (core_mosquitto)
2024-07-23 17:32:10.235 INFO (MainThread) [supervisor.host.info] Updating local host information
2024-07-23 17:32:10.879 INFO (MainThread) [supervisor.host.services] Updating service information
2024-07-23 17:32:10.890 INFO (MainThread) [supervisor.host.network] Updating local network information
2024-07-23 17:32:11.158 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
2024-07-23 17:32:11.179 INFO (MainThread) [supervisor.host.manager] Host information reload completed
2024-07-23 17:32:27.667 WARNING (MainThread) [supervisor.addons.options] Option 'anonymous' does not exist in the schema for Mosquitto broker (core_mosquitto)
[... 244 lines cut ...]
version_api | ok
installed_addons | Mosquitto broker (6.4.1), File editor (5.8.0), Terminal & SSH (9.14.0), InfluxDB (5.0.0), Grafana (10.0.0), Node-RED (18.0.4), ESPHome (2024.6.6), Whisper (2.1.2), Piper (1.5.0), Cloudflared (5.1.14)

</details>

<details><summary>Dashboards</summary>

dashboards | 4
-- | --
resources | 14
views | 16
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 15 juillet 2024 à 08:50
-- | --
current_recorder_run | 19 juillet 2024 à 11:52
estimated_db_size | 4956.61 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

sairon, 2024-07-23:
I'm a bit doubtful the OS is to blame but let's see. When using `du -h`, also use `sort -h`, otherwise you're sorting the numbers regardless of the units. Then it should be obvious what's the culprit.

Impact123, 2024-07-24:
You won't be able to see much when using an addon. You have to urn this in the OS' CLI.
I wrote down some steps how to check space usage in the HA discord: https://discord.com/channels/330944238910963714/332167321311510530/1098322242812784793
I think the discord server or forum would be a much better choice for this conversation.

devbysb, 2024-07-24:
Hi 

It's true that the ssh addon has limited rights... Which I hadn't seen. I managed to find some files that were a bit heavy.

40.0M   /var/log/journal/faa3064398ef4cde823f6de94fd5e324
483.6M  /var/log/journal/9e39450d53ae434ea2f84a09c481e419
3.2G    /var/log/journal/aa4917c818474fccb1fd12291af65040

I made snapshot and delete aa4917c818474fccb1fd12291af65040. 

I'm taking this opportunity to see if it's possible to add tools to empty the logs, or to help clean up, see the disk space used by the addons, the database, the plugins. Maybe also activate developer ssh more easily because in a virtual machine it's a bit complicated.
These are just ideas :)

closed 2024-07-29 by devbysb