home-assistant/operating-system #3490: Some USB devices not recognized

justingoeden, 2024-07-24:
### Describe the issue you are experiencing

I have 7 USB ports, 6 of which are USB-A style (2 of which are powered) and 1 is a USB-C style.
I recently lost USB recognition on two devices.  My Zigbee and ZWave dongles are found, regardless of where they are plugged in, but the Coral USB and the BT antenna just can't been found/seen by HAOS.  I've tried every plug, every combination, changed cables, etc.  It always recognizes the Zigbee and ZWave.

It used to work just fine, I had the Coral TPU running in my Frigate addon, but one day a few months ago I noted that my cameras weren't working, then after digging, found that Frigate couldn't see the Coral TPU, then figured out that my whole Intel NUC and/or HAOS can't seem to see some of the USB devices.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

12.4

### Did the problem occur after upgrading the Operating System?

Yes

### Hardware details

I've got an HP NUC with Intel i7 with HAOS installed straight on to the SSD.

### Steps to reproduce the issue

1.
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
2024-07-23 21:10:09.505 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
2024-07-23 21:11:51.589 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
2024-07-23 21:11:52.626 INFO (MainThread) [supervisor.host.info] Updating local host information
2024-07-23 21:11:53.163 INFO (MainThread) [supervisor.host.services] Updating service information
2024-07-23 21:11:53.167 INFO (MainThread) [supervisor.host.network] Updating local network information
2024-07-23 21:11:53.228 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
2024-07-23 21:11:53.232 INFO (MainThread) [supervisor.host.manager] Host information reload completed
2024-07-23 21:41:51.671 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
2024-07-23 22:10:09.506 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state running
2024-07-23 22:10:09.506 INFO (MainThread) [supervisor.resolution.checks.base] Run check for docker_config/system
2024-07-23 22:10:09.506 INFO (MainThread) [supervisor.resolution.checks.base] Run check for disabled_data_disk/system
2024-07-23 22:10:09.506 INFO (MainThread) [supervisor.resolution.checks.base] Run check for security/core
2024-07-23 22:10:09.507 INFO (MainThread) [supervisor.resolution.checks.base] Run check for multiple_data_disks/system
2024-07-23 22:10:09.507 INFO (MainThread) [supervisor.resolution.checks.base] Run check for trust/supervisor
2024-07-23 22:10:09.511 INFO (MainThread) [supervisor.resolution.checks.base] Run check for pwned/addon
2024-07-23 22:10:09.612 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_ipv6_error/dns_server
2024-07-23 22:10:09.612 INFO (MainThread) [supervisor.resolution.checks.base] Run check for ipv4_connection_problem/system
2024-07-23 22:10:09.612 INFO (MainThread) [supervisor.resolution.checks.base] Run check for free_space/system
2024-07-23 22:10:09.612 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_failed/dns_server
2024-07-23 22:10:09.612 INFO (MainThread) [supervisor.resolution.checks.base] Run check for no_current_backup/system
2024-07-23 22:10:09.613 INFO (MainThread) [supervisor.resolution.check] System checks complete
2024-07-23 22:10:09.613 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state running
2024-07-23 22:10:09.671 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
2024-07-23 22:10:09.671 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state running
2024-07-23 22:10:09.671 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
2024-07-23 22:11:48.700 INFO (MainThread) [supervisor.backups.manager] Found 125 backup files
2024-07-23 22:11:50.081 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
2024-07-23 22:11:51.759 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
2024-07-23 22:11:59.317 INFO (MainThread) [supervisor.api.middleware.security] /supervisor/info access from cebe7a76_hassio_google_drive_backup
2024-07-23 22:11:59.319 INFO (MainThread) [supervisor.api.middleware.security] /backups access from cebe7a76_hassio_google_drive_backup
2024-07-23 22:41:51.829 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
2024-07-23 23:00:24.499 INFO (MainThread) [supervisor.api.middleware.security] /supervisor/info access from cebe7a76_hassio_google_drive_backup
2024-07-23 23:00:24.501 INFO (MainThread) [supervisor.api.middleware.security] /backups access from cebe7a76_hassio_google_drive_backup
2024-07-23 23:10:09.675 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state running
2024-07-23 23:10:09.675 INFO (MainThread) [supervisor.resolution.checks.base] Run check for docker_config/system
2024-07-23 23:10:09.675 INFO (MainThread) [supervisor.resolution.checks.base] Run check for disabled_data_disk/system
2024-07-23 23:10:09.675 INFO (MainThread) [supervisor.resolution.checks.base] Run check for security/core
[... 243 lines cut ...]
version_api | ok
installed_addons | Z-Wave JS (0.6.2), Terminal & SSH (9.14.0), Google Assistant SDK (2.5.0), Home Assistant Google Drive Backup (0.112.1), ESPHome (2024.7.0), Studio Code Server (5.15.0), File editor (5.8.0), Mosquitto broker (6.4.1), Zigbee2MQTT (1.39.0-1), Frigate (Full Access) (0.13.2)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 67
views | 28
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | July 14, 2024 at 3:27 AM
-- | --
current_recorder_run | July 23, 2024 at 3:00 AM
estimated_db_size | 437.18 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

sairon, 2024-07-29:
Can you be more specific about the NUC model? The logs are missing lines from the kernel log that could shed some light on what's wrong, ideally reboot the board and run `ha host logs -n2000 -b0` - that should contain complete information about the system initialization.

> It used to work just fine

Can you tell when it stopped working? You can boot to the previous installed OS version using `ha os boot-slot other`. If it doesn't fix it, try downgrading to a specific version using e.g. `ha os update --version 12.3`.

justingoeden, 2024-07-30:
> Can you be more specific about the NUC model? The logs are missing lines from the kernel log that could shed some light on what's wrong, ideally reboot the board and run `ha host logs -n2000 -b0` - that should contain complete information about the system initialization.

Yup!  It's an HP EliteDesk 800 G2 Mini
-i7-6700T
-Motherboard: Intel Q270
-256Gb SSD
-16gb DDR4 RAM

When I ran the 'ha host logs -n2000 -b0', it gave me like 1800 lines of data.  I can copy and paste here, but I feel like that's a little much.  Thoughts? I can try and look for stuff within that, but mind you, I know nothing about kernels and all that stuff.

> Can you tell when it stopped working? You can boot to the previous installed OS version using `ha os boot-slot other`. If it doesn't fix it, try downgrading to a specific version using e.g. `ha os update --version 12.3`.

All I can say right now is roughly.  I've found that some people are having "similar" issues, but not quite the same with some of the newer updates.  But I honestly can't recall specifics at this moment.  I can try and install older versions, hopefully it will let me update back to today's should something go haywire.

justingoeden, 2024-07-30:
Thank you Jan for replying!

Preface: I'm essentially a computer novice so I'll try and answer
everything appropriately.  I did reply on the Github issue, but wasn't sure
of the best way to respond, so I'll reply to your email as well.

here's what I replied to you with:

---Can you be more specific about the NUC model? The logs are missing lines
from the kernel log that could shed some light on what's wrong, ideally
reboot the board and run ha host logs -n2000 -b0 - that should contain
complete information about the system initialization.

Yup! It's an HP EliteDesk 800 G2 Mini
-i7-6700T
-Motherboard: Intel Q270
-256Gb SSD
-16gb DDR4 RAM

When I ran the 'ha host logs -n2000 -b0', it gave me like 1800 lines of
data. I can copy and paste here, but I feel like that's a little much.
Thoughts? I can try and look for stuff within that, but mind you, I know
nothing about kernels and all that stuff.

---Can you tell when it stopped working? You can boot to the previous
installed OS version using ha os boot-slot other. If it doesn't fix it, try
downgrading to a specific version using e.g. ha os update --version 12.3.

All I can say right now is roughly. I've found that some people are having
"similar" issues, but not quite the same with some of the newer updates.
But I honestly can't recall specifics at this moment. I can try and install
older versions, hopefully it will let me update back to today's should
something go haywire.

justingoeden, 2024-07-30:
To add in more details.  I've "back-installed" older OS's incrementally to 10.5 with no luck.  

I've also plugged in both my Coral TPU on to my main Windows desktop and it comes up as "Unknown Device" on Device Manager.  My BT antenna installed itself and recognized itself as a BT LE antenna .  So both items seem to be functional.

Also, I forgot to put this in the original, and I doubt it matters, but the only other hardware change I made was upgrading from the Conbe Zigbee controller to the Sonoff 3.0 dongle, and migrated the radios.  But that shouldn't have anything to do with the USB's not being recognized, one would think.

sairon, 2024-07-31:
> To add in more details. I've "back-installed" older OS's incrementally to 10.5 with no luck.

Then I'm not sure it's really an OS regression. If you say it used to work few months ago but rolling back to 10.5 (which is almost a year old) didn't help, there might be something else in play.

The host logs would definitely help. If you're using the web terminal, you can save the logs to a text file in the config folder (i.e. where your `configuration.yaml` is) using `ha host logs -n2000 -b0 > /config/host_logs.txt` and then download it from there.

justingoeden, 2024-08-01:
> The host logs would definitely help. If you're using the web terminal, you can save the logs to a text file in the config folder (i.e. where your `configuration.yaml` is) using `ha host logs -n2000 -b0 > /config/host_logs.txt` and then download it from there.

Oh thank you so much!  Ok it should be attached!! I've since went back to the latest version, so if it shows 12.x, that's all it is.  Oh shoot, I'm just thinking out loud.  When I ran this log file, I do NOT have the antenna or TPU currently plugged in to NUC (was testing them on my regular Windows PC).  Do they need to be plugged into the NUC for you to see what you need to see?  If so, I can plug them in and re-run the log file.

THanks!!!!
[host_logs_8124.txt](https://github.com/user-attachments/files/16459861/host_logs_8124.txt)

sairon, 2024-08-02:
This log is looking fine, please send a new one with the non-working devices connected. Ideally connect them to the ports where the Zigbee and Z-Wave sticks are now connected and place those to some of other ports.

justingoeden, 2024-08-03:
Ok, I've done exactly as you said.  I removed the Zigbee and Zwave controllers and placed them in different ports. Then I placed the BLE antenna and Coral in the previous ports that had the Zigbee and Zwave.  Attached is the log.
[host_logs_8324.txt](https://github.com/user-attachments/files/16481075/host_logs_8324.txt)

sairon, 2024-08-05:
Hmm, while the Bluetooth dongle appears (and there is no indication it should **not** work), the other port (`usb 1-3` from the previous log) doesn't even puts a log message there is a device plugged in. If it were a Linux driver issue, it should at least print some log messages about device enumeration, yet here it's completely silent. This IMO points to some hardware issue :thinking:

justingoeden, 2024-08-08:
Well thank you all for the help.  Guess I'm out of luck lol.  Don't know what to do from here.  I appreciate it though!

closed 2024-11-13 by github-actions[bot]