home-assistant/operating-system #3494: Nvme driver support missing in HaOS 12.4 for HP laptop 15-dw3023nia

SelalexHA, 2024-07-28:
### Describe the issue you are experiencing

Hello! I was trying to install HaOS on an HP 15-dw3023nia using a live Linux USB drive. The image flashed successfully, but there was no boot - it got stuck after USB device enumeration.
New plan: install HaOS on an SD card and move the HaOS data disk to the NVMe drive. HaOS installed and booted fine from SdCard, but the NVMe drive is not listed in devices (OS datadisk list shows no devices/disks), and no new storage hardware from web interface as well. So i can not move haos data drive to Nvme drive.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

Home Assistant OS 12.4

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

system description - https://support.hp.com/th-en/document/c08005860 No additional diveces intalles.

### Steps to reproduce the issue

1. Install HaOS on sd card, following instructions from https://www.home-assistant.io/installation/generic-x86-64/ on HP HP Laptop 15-dw3023nia
2. type in ha console > os datadisk list - NVMe SSd is missing.

### Anything in the Supervisor logs that might be useful for us?

```txt
no errors
```

### Anything in the Host logs that might be useful for us?

```txt
no errors
```

### System information

## System Information

version | core-2024.7.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.4
os_name | Linux
os_version | 6.6.33-haos
arch | x86_64
timezone | Europe/Moscow
config_dir | /config

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.4
-- | --
update_channel | stable
supervisor_version | supervisor-2024.06.2
agent_version | 1.6.0
docker_version | 26.1.4
disk_total | 6.6 GB
disk_used | 4.9 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | 

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 0
views | 0
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | July 27, 2024 at 11:12 PM
-- | --
current_recorder_run | July 28, 2024 at 2:12 AM
estimated_db_size | 0.24 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

SelalexHA, 2024-07-28:
Some updates - i have tied 13.0 latest dev build - same problem. 
[image]
Also - photo of screen while trying to boot from HaOs image installed on Nvme. No changes after screenshot.

sairon, 2024-07-29:
Unfortunately the HW description doesn't give much details about the chipset. Could you pull please pull a full `dmesg` output from the running system (no matter if HAOS or any other Linux distribution)? Alternatively, maybe `lspci` could be sufficient.

SelalexHA, 2024-07-30:
> Unfortunately the HW description doesn't give much details about the chipset. Could you pull please pull a full `dmesg` output from the running system (no matter if HAOS or any other Linux distribution)? Alternatively, maybe `lspci` could be sufficient.

I realy appretiate your help! Here is lspci output from live usb drive and attached dmesg output:

0000:00:00.0 Host bridge: Intel Corporation Device 9a04 (rev 01)
0000:00:02.0 VGA compatible controller: Intel Corporation Tiger Lake-LP GT2 [UHD Graphics G4] (rev 01)
0000:00:04.0 Signal processing controller: Intel Corporation TigerLake-LP Dynamic Tuning Processor Participant (rev 01)
0000:00:08.0 System peripheral: Intel Corporation GNA Scoring Accelerator module (rev 01)
0000:00:0e.0 RAID bus controller: Intel Corporation Volume Management Device NVMe RAID Controller
0000:00:14.0 USB controller: Intel Corporation Tiger Lake-LP USB 3.2 Gen 2x1 xHCI Host Controller (rev 20)
0000:00:14.2 RAM memory: Intel Corporation Tiger Lake-LP Shared SRAM (rev 20)
0000:00:15.0 Serial bus controller: Intel Corporation Tiger Lake-LP Serial IO I2C Controller #0 (rev 20)
0000:00:15.1 Serial bus controller: Intel Corporation Tiger Lake-LP Serial IO I2C Controller #1 (rev 20)
0000:00:16.0 Communication controller: Intel Corporation Tiger Lake-LP Management Engine Interface (rev 20)
0000:00:17.0 System peripheral: Intel Corporation RST VMD Managed Controller
0000:00:1d.0 PCI bridge: Intel Corporation Tiger Lake-LP PCI Express Root Port #9 (rev 20)
0000:00:1d.1 PCI bridge: Intel Corporation Tiger Lake-LP PCI Express Root Port #10 (rev 20)
0000:00:1e.0 Communication controller: Intel Corporation Tiger Lake-LP Serial IO UART Controller #0 (rev 20)
0000:00:1e.3 Serial bus controller: Intel Corporation Tiger Lake-LP Serial IO SPI Controller #1 (rev 20)
0000:00:1f.0 ISA bridge: Intel Corporation Tiger Lake-LP LPC Controller (rev 20)
0000:00:1f.3 Multimedia audio controller: Intel Corporation Tiger Lake-LP Smart Sound Technology Audio Controller (rev 20)
0000:00:1f.4 SMBus: Intel Corporation Tiger Lake-LP SMBus Controller (rev 20)
0000:00:1f.5 Serial bus controller: Intel Corporation Tiger Lake-LP SPI Controller (rev 20)
0000:01:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8211/8411 PCI Express Gigabit Ethernet Controller (rev 15)
0000:02:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8821CE 802.11ac PCIe Wireless Network Adapter
10000:e0:17.0 SATA controller: Intel Corporation Tiger Lake-LP SATA Controller (rev 20)
10000:e0:1c.0 System peripheral: Intel Corporation RST VMD Managed Controller
10000:e0:1c.4 PCI bridge: Intel Corporation Tiger Lake-LP PCI Express Root Port #5 (rev 20)
10000:e1:00.0 Non-Volatile memory controller: Samsung Electronics Co Ltd NVMe SSD Controller 980 (DRAM-less)
[dmesg.txt](https://github.com/user-attachments/files/16427699/dmesg.txt)

closed 2024-11-07 by github-actions[bot]