home-assistant/operating-system #3525: Error updating Home Assistant Operating System: Can't fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/13.0/haos_rpi4-64-13.0.raucb: Cannot connect to host github.com:443 ssl:default [Connect call failed ('20.205.243.166', 443)]

liamlain, 2024-08-14:
### Describe the issue you are experiencing

Error when updating OS 

```
Error updating Home Assistant Operating System: Can't fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/13.0/haos_rpi4-64-13.0.raucb: Cannot connect to host github.com:443 ssl:default [Connect call failed ('20.205.243.166', 443)]
```

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

12.4

### Did the problem occur after upgrading the Operating System?

No

### Hardware details

pi4 4gb 

### Steps to reproduce the issue

1. when click update OS
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
[32m2024-08-14 15:57:23.030 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_ipv6_error/dns_server[0m
[32m2024-08-14 15:57:23.056 INFO (MainThread) [supervisor.resolution.checks.base] Run check for ipv4_connection_problem/system[0m
[32m2024-08-14 15:57:23.057 INFO (MainThread) [supervisor.resolution.checks.base] Run check for pwned/addon[0m
[32m2024-08-14 15:57:23.074 INFO (MainThread) [supervisor.resolution.checks.base] Run check for security/core[0m
[32m2024-08-14 15:57:23.075 INFO (MainThread) [supervisor.resolution.checks.base] Run check for multiple_data_disks/system[0m
[32m2024-08-14 15:57:23.075 INFO (MainThread) [supervisor.resolution.checks.base] Run check for free_space/system[0m
[32m2024-08-14 15:57:23.076 INFO (MainThread) [supervisor.resolution.checks.base] Run check for dns_server_failed/dns_server[0m
[32m2024-08-14 15:57:23.100 INFO (MainThread) [supervisor.resolution.checks.base] Run check for docker_config/system[0m
[32m2024-08-14 15:57:23.101 INFO (MainThread) [supervisor.resolution.check] System checks complete[0m
[32m2024-08-14 15:57:23.101 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state running[0m
[32m2024-08-14 15:57:23.322 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete[0m
[32m2024-08-14 15:57:23.322 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state running[0m
[32m2024-08-14 15:57:23.323 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete[0m
[32m2024-08-14 16:02:12.900 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/allenporter/stream-addons repository[0m
[32m2024-08-14 16:02:12.907 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/thomasmauerer/hassio-addons repository[0m
[32m2024-08-14 16:02:12.914 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/mKeRix/hassio-repo repository[0m
[32m2024-08-14 16:02:12.921 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/dianlight/hassio-addons repository[0m
[32m2024-08-14 16:02:12.928 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/adamoutler/HassOSConfigurator repository[0m
[32m2024-08-14 16:02:12.943 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/ repository[0m
[32m2024-08-14 16:02:12.956 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/music-assistant/home-assistant-addon repository[0m
[32m2024-08-14 16:02:12.971 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/sabeechen/hassio-google-drive-backup repository[0m
[32m2024-08-14 16:02:12.983 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/adamoutler/HassOSArgonOneAddon repository[0m
[32m2024-08-14 16:02:13.000 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/blakeblackshear/frigate-hass-addons repository[0m
[32m2024-08-14 16:02:13.014 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/AlexxIT/hassio-addons repository[0m
[32m2024-08-14 16:02:13.023 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/adamoutler/Addons repository[0m
[32m2024-08-14 16:02:13.039 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/mercenaruss/zigstar_addons repository[0m
[32m2024-08-14 16:02:13.050 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/brenner-tobias/ha-addons repository[0m
[32m2024-08-14 16:02:13.055 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/zigbee2mqtt/hassio-zigbee2mqtt repository[0m
[32m2024-08-14 16:02:13.069 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/CoolKit-Technologies/ha-addon repository[0m
[32m2024-08-14 16:02:13.084 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/esphome/home-assistant-addon repository[0m
[32m2024-08-14 16:02:13.087 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/home-assistant/addons repository[0m
[32m2024-08-14 16:02:13.096 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/hassio-addons/repository repository[0m
[33m2024-08-14 16:02:17.554 WARNING (SyncWorker_2) [supervisor.addons.validate] Add-on have full device access, and selective device access in the configuration. Please report this to the maintainer of Frigate (Full Access)[0m
[33m2024-08-14 16:02:17.570 WARNING (SyncWorker_2) [supervisor.addons.validate] Add-on have full device access, and selective device access in the configuration. Please report this to the maintainer of Frigate (Full Access) Beta[0m
[33m2024-08-14 16:02:17.829 WARNING (SyncWorker_0) [supervisor.addons.validate] Add-on have full device access, and selective device access in the configuration. Please report this to the maintainer of DeskPi Pro Active Cooling[0m
[33m2024-08-14 16:02:18.437 WARNING (SyncWorker_4) [supervisor.addons.validate] Add-on have full device access, and selective device access in the configuration. Please report this to the maintainer of Frigate (Full Access)[0m
[... 280 lines cut ...]

api_endpoint_reachable | ok
-- | --

</details>

<details><summary>SpotifyPlus</summary>

integration_version | v1.0.48
-- | --
clients_configured | 1: Lai Yee Tine (premium)
api_endpoint_reachable | ok

</details>

<details><summary>Xiaomi Miot Auto</summary>

component_version | 0.7.19
-- | --
can_reach_server | ok
can_reach_spec | ok
logged_accounts | 2
total_devices | 10

</details>

sairon, 2024-08-14:
This is a network connectivity issue, nothing that can be fixed here. Make sure that the device is connected to network with working internet connection with access to Github. Eventually retry after a while, could be an intermittent issue.

closed 2024-08-14 by sairon

p-mitana, 2024-08-15:
@sairon, I'm afraid it's not a network issue. I actually can successfully ping github.com from the web terminal addon. I even can manually download the addon via `wget`. Still, both `ha os update` and GUI fail.

[image]

All the previous updates were installed correctly.